PSP-Archive/ARK-4 #116: Infinity Loop in Sony Logo in game

skelletR, 2023-05-11:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.20

### Homebrew/GAME (if applicable)

Jak and Daxter: The Lost Frontier (UCUS98634) Dax format

### Steps to reproduce

open the game

I have a problem with infinity boot in the game Jak and Daxter: The Lost Frontier in dax format until now, when show the logo sony scene infinity

in previous version play in dax format without problem

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-05-11:
In which version was this playable?

JoseAaronLopezGarcia, 2023-05-11:
Let me know if this one fixes it:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1683820176

skelletR, 2023-05-11:
> Let me know if this one fixes it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1683820176

I dont remeber the version unfortunatly, this fix it partly, now show it the sony logo and move on but after the after sony logo show another logo wthi a meteorite fell but black screen and again from the metereon scene infinity

JoseAaronLopezGarcia, 2023-05-11:
Lets try this one: https://github.com/PSP-Archive/ARK-4/releases/tag/v1683823189

skelletR, 2023-05-11:
> Lets try this one: https://github.com/PSP-Archive/ARK-4/releases/tag/v1683823189

Just try it, again the same infinity in the second logo, only this game do it this in dax format, the other i have all play good

JoseAaronLopezGarcia, 2023-05-11:
Have you tried using the original dax.exe to create the file?
https://github.com/PSP-Archive/ARK-4/tree/main/contrib/PC/ciso

In any case, I think the best thing to do is test out older versions until it works and report back.

skelletR, 2023-05-11:
> Have you tried using the original dax.exe to create the file? https://github.com/PSP-Archive/ARK-4/tree/main/contrib/PC/ciso
> 
> In any case, I think the best thing to do is test out older versions until it works and report back.

I have test it from 4.19.2 until the latest the same, after try it more old version, not have try, how can try it with this?

JoseAaronLopezGarcia, 2023-05-11:
> I have test it from 4.19.2 until the latest the same, after try it more old version, not have try, how can try it with this?

Download dax.exe and zlib.dll from that folder and run it with command line:

dax.exe -l9 input.iso output.dax

skelletR, 2023-05-11:
> Download dax.exe and zlib.dll from that folder and run it with command line:
>
> dax.exe -l9 input.iso output.dax

Thanks, it work with this, is a few mb more but its ok and i found 4.16 version it working the dax format, i try it with all after thathis version and inifinity loop again and with more old version just a black screen

JoseAaronLopezGarcia, 2023-05-11:
Hey so I managed to reproduce the issue, which is good as it will let me debug it. In the meantime I have found that enabling Inferno Cache solves it as a workaround.

skelletR, 2023-05-11:
> Hey so I managed to reproduce the issue, which is good as it will let me debug it. In the meantime I have found that enabling Inferno Cache solves it as a workaround.

This is good, yeah i try it with inferno cache enabled and is working

JoseAaronLopezGarcia, 2023-05-11:
I got it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1683845103

closed 2023-05-12 by skelletR