PSP-Archive/ARK-4 #118: Been 10 - Protector of Earth Crashing

KarnFu, 2023-05-13:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.33

### Homebrew/GAME (if applicable)

Ben 10: Protector of Earth (ULUS10307)

### Plugins ( if applicable )

Cxmb = Enable
Category = Enable
Dayviewer = Enable
Hold = Enable
Leda = Enable
No DRM Free = Enable
PsP Powersaver = Enable
Zerovsh Patcher = Enable
CFE = Disable
Cushome = Enable

### Steps to reproduce

1. Ben 10 Protector of Earth loads just fine
2. After selecting new game in the main menu of the game
3. It tries to access Memory Stick and freezes
4. And the PSP just shut down

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Always

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Always

TheSubPlayer, 2023-05-13:
Have you tried starting the game with the plugins disabled? (hold <kbd>START</kbd> while the game starts to disable plugins and/or <kbd>SELECT</kbd> to disable configs.

JoseAaronLopezGarcia, 2023-05-13:
Could this be one of those heavy DRM games?
@KarnFu have you tried the game in PBP (Official Sony) format?

KarnFu, 2023-05-14:
> Have you tried starting the game with the plugins disabled? (hold START while the game starts to disable plugins and/or SELECT to disable configs).

Yes I did try that but could not get it to work.

KarnFu, 2023-05-14:
> Have you tried starting the game with the plugins disabled? (hold START while the game starts to disable plugins and/or SELECT to disable configs).

I did try it with the plugins disabled but but could not get the ISO to work. I tried it with LME and everything is fine.

KarnFu, 2023-05-14:
> Could this be one of those heavy DRM games? @KarnFu have you tried the game in PBP (Official Sony) format?

Maybe a heavy DRM game because I did manage to find the eboot and everything is fine now

JoseAaronLopezGarcia, 2023-05-15:
I just tried an ISO of this game and couldn't reproduce the issue. Perhaps it was a bad download or copy?

KarnFu, 2023-05-15:
> I just tried an ISO of this game and couldn't reproduce the issue. Perhaps it was a bad download or copy?

Understood I bought the official eboot from PSN, it's all great. ARk feels great everything feels great running on ARK.

closed 2023-05-15 by KarnFu