PSP-Archive/ARK-4 #119: PSP USB pairing with PS2/PS3 on Vita (Standalone install)

PookaMustard, 2023-05-13:
### New Feature

Some PSP games offer a feature to pair with a PS2 or PS3 game by USB. Examples of these game pairs include:

- Assassin's Creed Bloodlines + Assassin's Creed II (PS3)
- Daxter + Jak X Combat Racing (PS2)
- Resistance Retribution + Resistance 2 (PS3)
- OutRun 2006 Coast 2 Coast (PSP-PS2)
- Crash Tag Team Racing (PSP-PS2)

On the Vita, stock PSPEMU (through PSN downloads or chovy-signed games) supports this kind of pairing, but ARK-4 standalone doesn't support it. (For the record, Adrenaline doesn't support it either).

I've been told that the stock PSPEMU calls a Vita-side UDCD driver called usbpspcm.skprx which emulates the PSP communication protocol. PSPEMU probably calls it through syscalls to achieve something like this: PSP syscall -> PSPEMU catching it -> calls funcs from usbpspcm. In theory, it should work by disabling the default USB handler that ARK-4 uses, but trying that out with Adrenaline didn't make it work so there's probably something else at play.

isage, 2023-05-13:
Some more info: with disabled usb patches, `sceUsbStart("USBPSPCommunicationDriver"...)` returns 0x80243407 (not available)

isage, 2023-05-13:
some digging through ghidra later:
sceUsbStart from kermit_usb.prx with "USBPSPCommunicationDriver" as argument calls (well, should call)`sceKermit_driver_0x36666181(uVar3,0xc,0x164,2,0,local_30);` 
this is handled in pspemu eboot via
```
      case 0x164:
        uVar29 = DAT_81015de0;
        if (DAT_8216b83c != 0) {
          uVar23 = *(undefined4 *)(local_bc + 0x10);
          uVar6 = FUN_81006364(*(undefined4 *)(local_bc + 0x18),1,0x100);
          uVar26 = 0x8101611d;
          sceUsbPspcmSetTitle(uVar6,uVar23);
          uVar29 = VectorShiftRight(CONCAT44(uVar26,uVar26),0x20);
        }
```

yet, under adrenaline sceUsbPspcmSetTitle never gets called

JoseAaronLopezGarcia, 2023-05-13:
Do you think this can be done via standalone installs?
We don't have the ability to run native vita code there. We can indeed issue Kermit commands, so maybe it is possible to enable this functionality. However my experience with official pspemu is that it usually uses a blacklist to allow/enable certain functionality depending on the game, specially in later firmwares. This might not apply to USB access but it certainly applies to other things such as WLAN and even IO.

isage, 2023-05-13:
Sadly, i know almost zero about how psp/pspemu works, so no idea.
Since I'm testing with same title (which isn't from store, so shouldn't be in whitelist if there's one) in adrenaline and raw pspemu, i can only assume something in adrenaline breaks it. Initially i thought usb patches for xmb "usb mode" were a culprit, but no

JoseAaronLopezGarcia, 2023-05-13:
No it can't be the XMB patches itself, but it can and most likely is the native vita side patches that adrenaline does tk allow the xmb to mount usb: https://github.com/TheOfficialFloW/Adrenaline/blob/master/user/usb.c

isage, 2023-05-13:
Yes, that's what I've meant. And those https://github.com/TheOfficialFloW/Adrenaline/blob/master/cef/vshctrl/main.c#L813-L826

Removing them doesn't help (just, obviously, breaks xmb "usb mode")

isage, 2023-05-13:
You might be right about whitelist, though (title_specific.bin to be specific, since adrenaline messes with that, and usb handler functions seem to check something from there. i'll dig some more)

JoseAaronLopezGarcia, 2023-05-13:
Well like I said, you can always patch it from Vita side via a plugin, allowing it to work even with standalone installs.
Here's an example of a native vita plugin that patches psp emu to extend ARK functionality:
https://github.com/rereprep/ArkRightAnalog

isage, 2023-05-13:
ok, this is indeed title_specific/whitelist stuff.
i'm not sure how that should be handled (and if it should be handled at all) in ARK, but gere's the solution:
https://github.com/TheOfficialFloW/Adrenaline/blob/master/user/titleinfo.c#L41-L46 value at info offset - 4
(e.g. data_addr + 0x115644c / data_addr + 0x115654c) should be set to `0x10`

vsh usb patches seem to cause no troubles with games

JoseAaronLopezGarcia, 2023-05-13:
Yes, this needs to be done outside ARK, either in Adrenaline (for Pentazemin compat) or a plugin (for Standalone).
Either way great job figuring this out.

isage, 2023-05-13:
Yeah, pullrequested fix to adrenaline and made separate plugin https://github.com/isage/adrenaline_usb_enabler which should work with standalone theoretically

JoseAaronLopezGarcia, 2023-05-13:
> Yeah, pullrequested fix to adrenaline and made separate plugin https://github.com/isage/adrenaline_usb_enabler which should work with standalone theoretically

Wow that was fast, awesome work!

PookaMustard, 2023-05-14:
I really never expected a solution to pop up in a week, let alone in around 12 hours :P

isage's plugin works with adrenaline, which is very surprising because I thought we'd need The Flow on it but he hasn't made a new Adrenaline release in a while. With ARK-4 standalone I'm not quite sure I hooked it correctly or not, I applied the plugin to arkfast's Sasuke vs. Commander (NPUZ00146) but it didn't work.

So while adrenaline works for my use case, I'm still interested in making it work with ARK-4.

Either way, thanks everyone.

Pik-a-boo, 2023-05-14:
> Yeah, pullrequested fix to adrenaline and made separate plugin https://github.com/isage/adrenaline_usb_enabler which should work with standalone theoretically

Can you make a plugin/fix that makes WEP/Open only games to work with the Vita as well?
That seems like another limitation that still didnt get resolved.

JoseAaronLopezGarcia, 2023-05-15:
> Can you make a plugin/fix that makes WEP/Open only games to work with the Vita as well? That seems like another limitation that still didnt get resolved.

This one is a bit strange cause I'm sure I've been able to use WPA2 connections on ePSP standalone in the custom launcher. Perhaps depending on how you call the wrapper it might just passthrough to whatever the Vita is connected to.

JoseAaronLopezGarcia, 2023-05-16:
You can always run ARK 4 inside Adrenaline. Though I understand the standalone version has its pros.

Yoti, 2023-06-06:
> Can you make a plugin/fix that makes WEP/Open only games to work with the Vita as well? That seems like another limitation that still didnt get resolved.

May you provide a list of such games?

Pik-a-boo, 2023-06-06:
> > Can you make a plugin/fix that makes WEP/Open only games to work with the Vita as well? That seems like another limitation that still didnt get resolved
> 
> May you provide a list of such games?

Some of the games got posted here: with more info on the situation as well: https://github.com/PSP-Archive/ARK-4/issues/32
One of the region versions of Infected also is WEP only but I don't remember which (I think USA).

JoseAaronLopezGarcia, 2023-07-06:
I totally forgot to mention that it's actually pretty easy to have a Standalone installation with USB support, you just have to install ARK's PBOOT bubble (via FastARK and/or ChovySign) on a PSN game that's known to have USB support.
Same goes for camera.

We are trying to find if there's a game that supports both so we can make a dummy loader with that game's ID.

isage, 2023-07-06:
Camera isn't a problem in games. It doesn't work in (adrenaline) xmb, but that's adrenaline's fault

JoseAaronLopezGarcia, 2023-07-06:
> Camera isn't a problem in games. It doesn't work in (adrenaline) xmb, but that's adrenaline's fault

So we only really need to grab the ID of any of the games compatible with USB and make the dummy loader out of it?
That's pretty easy to do. This very thread has a list of games with USB support.
Is there any other hardware that's only available to some games depending on their ID?

JoseAaronLopezGarcia, 2025-03-18:
@isage made a plugin for this that fixes USB on all ePSP apps. Closing.

closed 2025-03-18 by JoseAaronLopezGarcia