PSP-Archive/ARK-4 #121: Infinity shuts down PSP after starting the installer app POST installation of ARK and Infinity

chasemc215, 2023-05-14:
Infinity just shuts the console down after installing Infinity after ARK was installed, I'm not sure what to do.

chasemc215, 2023-05-14:
Nevermind, I solved it myself.

closed 2023-05-14 by chasemc215

DeadLootJ, 2023-05-15:
How exactly did you solve it? I am having the same issue with my PSP Go, and I have no idea how to fix it.

JoseAaronLopezGarcia, 2023-05-15:
> How exactly did you solve it? I am having the same issue with my PSP Go, and I have no idea how to fix it.

Run Infinity from OFW.

DeadLootJ, 2023-05-15:
I did that with no dice, may be an issue with my Go then. Thanks for the suggestion though.

JoseAaronLopezGarcia, 2023-05-15:
I've heard that having it on another folder makes it work. So maybe copy it elsewhere and launch it with ARK's file manager.

DeadLootJ, 2023-05-16:
Thank you for the advice, I will try it out and see if it works.

chasemc215, 2023-05-16:
I actually just ran ARK before opening Infinity POST installation of Infinity