PSP-Archive/ARK-4 #122: PS1 Games Eboot Only with Custom Launcher

skelletR, 2023-05-15:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.33

### Homebrew/GAME (if applicable)

All PS1 games i have

### Steps to reproduce

The PS1 Games only run from the custom launcher
From the normal menu only a black screen 

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-05-15:
Confirmed on my PSP Go. Strange, not long ago it was working. Maybe try older versions until we find the one that breaks it.

skelletR, 2023-05-15:
> Confirmed on my PSP Go. Strange, not long ago it was working. Maybe try older versions until we find the one that breaks it.

Yeah, im the same guy i try almost many old version to found why jak and daxter in dax format not working in previous version

JoseAaronLopezGarcia, 2023-05-15:
That's too old, I remember it working very recently, on 4.19 and maybe even 4.20, we need to pinpoint the exact release that broke it.

skelletR, 2023-05-15:
> That's too old, I remember it working very recently, on 4.19 and maybe even 4.20, we need to pinpoint the exact release that broke it.

oh understand, i check it again and tell you

JoseAaronLopezGarcia, 2023-05-15:
So 4.20.15 works but 4.20.16 doesn't? I'll check that out.

skelletR, 2023-05-15:
> So 4.20.15 works but 4.20.16 doesn't? I'll check that out. Edit: yes I can confirm that it breaks in 4.20.16 so I'm investigating the issue.

Yeah exacly, 4.20.16 not work

JoseAaronLopezGarcia, 2023-05-15:
Got it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1684172853

skelletR, 2023-05-15:
> Got it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1684172853

Thanks, is working

closed 2023-05-15 by JoseAaronLopezGarcia