PSP-Archive/ARK-4 #128: [feature request] Full CFW functionality even if mem stick isn't inserted

wad11656, 2023-05-17:
### New Feature

I unbrick & install CFW on people's PSP's and then send it back to them. Sometimes they don't include a memory stick with their PSP. I've only just recently started dabbling with ARK. After perma-patching with Infinity, it seems you still need the files on the memory stick (Perhaps the /SAVEDATA/ ones) in order to bring up the VSH menu by pressing select, and perhaps other features. I'm pretty sure PRO/ME Infinity installations ingrain all CFW functionality onto the system's internal memory, to where you don't need to have a memory stick inserted to use certain CFW features. I was hoping a version of ARK could be installed that's still a full version of the CFW (including VSH menu, but maybe minus the custom launcher [due to presumed memory constraints]) without a memory stick inserted

krazynez, 2023-05-17:
You can add the vsh menu to flash for ARK if you wish (In our README how). We prefer not to put things in flash as it is much less of a risk of bricking PSP's (even if you can unbrick most PSP's now). Plus allowing us to have more flexibility if need be and not be contained to the limited space we have in flash. Custom Launcher and Recovery are just EBOOT's that we call.

JoseAaronLopezGarcia, 2023-05-17:
Some of the CFW features require a memory stick, though you can install VSH Menu on flash0. No recovery or launcher though, no way in hell they can fit on the flash.

Anyways, to permanently install VSH Menu in flash0, just rename the file `VSHMENU.PRX` inside `ARK_01234` to `ark_satelite.prx` and copy it to `flash0:/vsh/module/`.

JoseAaronLopezGarcia, 2023-05-17:
Thinking more about this idea, which I actually like, it's clear that we can't have the current recovery menu and launcher.
And the only thing missing from a fully permanent flash installation is the ability to enable/disable settings and plugins.
So I'm thinking maybe add support for the ability to load a recovery menu from flash0 like a traditional CFW when none is available in memory stick. We would have to make a different recovery menu, one that will most likely be much simpler, but with enough features it should be doable.

JoseAaronLopezGarcia, 2023-05-24:
Ok so here's how I've tackled this feature:

- There is now an "ARK Full Installer" which will copy the most important external modules from the `ARK_01234` folder into flash0 so that the CFW can load them from there when there's no memory stick or no standard ARK installation on the memory stick.
- Custom Launcher and Recovery Menu CANNOT be installed since they are too big to fit in flash0. These two however are redundant. The `XMB` has always been the launcher used, and as for recovery options you can use the start/select hotkeys.

Installation is simple, install and run `ARK` as usual and then run the `ARK Full Installer`, then you can delete the `ARK_01234` savedata folder or remove the memory stick.

closed 2023-05-24 by JoseAaronLopezGarcia