PSP-Archive/ARK-4 #129: Burnout Legends black screen booting from custom luncher

jappoman, 2023-05-17:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.35

### Homebrew/GAME (if applicable)

Burnout Legends

### Steps to reproduce

1. Load Burnout Legends iso from custom launcher
2. Black screen
The problem does not appear in normal vsh boot (the iso is good). Also, if you boot the game in vsh and then load the game from custom launcher, it works.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Autoboot Launcher

Always

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Always

krazynez, 2023-05-17:
Yes this is an issue, I can reproduce this as well.

JoseAaronLopezGarcia, 2023-05-18:
Very strange issue.

krazynez, 2023-05-22:
Socom Fireteam Bravo (WORKING)  vs Burnout Legends (Not working)

```diff
UID: 0x0187D21B Attr: 0007 - Name: sceController_Service      |	UID: 0x0187D21D Attr: 0007 - Name: sceController_Service
UID: 0x01879827 Attr: 0007 - Name: sceLED_Service	      |	UID: 0x01879829 Attr: 0007 - Name: sceLED_Service
UID: 0x0208A36B Attr: 0007 - Name: sceCodepage_Service	      |	UID: 0x0208A36D Attr: 0007 - Name: sceCodepage_Service
UID: 0x02070F1D Attr: 0007 - Name: sceWM8750_Driver	      |	UID: 0x02070F1F Attr: 0007 - Name: sceWM8750_Driver
UID: 0x02082D2B Attr: 0007 - Name: sceAudio_Driver	      |	UID: 0x02081D2D Attr: 0007 - Name: sceAudio_Driver
UID: 0x0206EB3F Attr: 0007 - Name: sceHP_Remote_Driver	      |	UID: 0x0206EB41 Attr: 0007 - Name: sceHP_Remote_Driver
UID: 0x02068F4F Attr: 4007 - Name: sceOpenPSID_Service	      |	UID: 0x02069B51 Attr: 4007 - Name: sceOpenPSID_Service
UID: 0x0205C05D Attr: 0007 - Name: sceUSB_Driver	      |	UID: 0x0205D15D Attr: 0007 - Name: sceUSB_Driver
UID: 0x02052C69 Attr: 0007 - Name: sceWlan_Driver	      |	UID: 0x0205376B Attr: 0007 - Name: sceWlan_Driver
UID: 0x02DC210B Attr: 0007 - Name: sceWlanFirmMagpie_driver   |	UID: 0x02DC210D Attr: 0007 - Name: sceWlanFirmMagpie_driver
UID: 0x02DBD919 Attr: 0007 - Name: sceRegistry_Service	      |	UID: 0x02DBD91B Attr: 0007 - Name: sceRegistry_Service
UID: 0x02DB4429 Attr: 0006 - Name: sceMgr_Driver	      |	UID: 0x02DB442B Attr: 0006 - Name: sceMgr_Driver
UID: 0x02DB9151 Attr: 6007 - Name: sceMesgLed		      |	UID: 0x02DB9B41 Attr: 6007 - Name: sceMesgLed
UID: 0x02DA9661 Attr: 4007 - Name: sceAmctrl_driver	      |	UID: 0x02DA3063 Attr: 4007 - Name: sceAmctrl_driver
UID: 0x02DA656F Attr: 4006 - Name: scePspNpDrm_Driver	      |	UID: 0x02DA8471 Attr: 4006 - Name: scePspNpDrm_Driver
UID: 0x02DA0079 Attr: 0007 - Name: sceIoFilemgrDNAS	      |	UID: 0x02DA0C7B Attr: 0007 - Name: sceIoFilemgrDNAS
UID: 0x02D99503 Attr: 0007 - Name: sceIsofs_driver	      |	UID: 0x02D9A105 Attr: 0007 - Name: sceIsofs_driver
UID: 0x02D8F815 Attr: 0007 - Name: sceUtility_Driver	      |	UID: 0x02D90017 Attr: 0007 - Name: sceUtility_Driver
UID: 0x01884A23 Attr: 0007 - Name: Stargate		      |	UID: 0x01884A25 Attr: 0007 - Name: Stargate
UID: 0x018A292D Attr: 0007 - Name: sceMeCodecWrapper	      |	UID: 0x018A292F Attr: 0007 - Name: sceMeCodecWrapper
UID: 0x03D73043 Attr: 0007 - Name: sceImpose_Driver	      |	UID: 0x03D73045 Attr: 0007 - Name: sceImpose_Driver
UID: 0x03D64A7F Attr: 0000 - Name: PSPLINK_USER		      |	UID: 0x03D64A01 Attr: 0000 - Name: PSPLINK_USER
UID: 0x03D70717 Attr: 0000 - Name: PSPLINK		      |	UID: 0x03D70719 Attr: 0000 - Name: PSPLINK
UID: 0x03D6002F Attr: 0007 - Name: sceKernelLibrary	      |	UID: 0x03D60031 Attr: 0007 - Name: sceKernelLibrary
UID: 0x03D5104F Attr: 0000 - Name: APP_APPLICATION_NAME	      |	UID: 0x03D51051 Attr: 0000 - Name: Burnout
UID: 0x03D4EF6B Attr: 0000 - Name: USBHostFS		      |	UID: 0x03D4EF6D Attr: 0000 - Name: USBHostFS
UID: 0x044C0547 Attr: 0006 - Name: sceAudiocodec_Driver	      <
UID: 0x044B9B67 Attr: 0006 - Name: sceSAScore		      <
UID: 0x044B7A11 Attr: 0006 - Name: sceATRAC3plus_Library      <
```

Running Burnout from XMB adds these (some are because of main menu in Burnout probably the networking ones)
```diff
                                  > UID: 0x044BD231 Attr: 0006 - Name: sceVideocodec_Driver
                                  > UID: 0x044B7F51 Attr: 0006 - Name: sceAudiocodec_Driver
                                  > UID: 0x044B5E71 Attr: 0006 - Name: sceSAScore
                                  > UID: 0x044B231B Attr: 0006 - Name: sceATRAC3plus_Library
                                  > UID: 0x044AD245 Attr: 0006 - Name: sceMpegbase_Driver
                                  > UID: 0x044AB16F Attr: 0006 - Name: sceMpeg_library
                                  > UID: 0x044A5F19 Attr: 0006 - Name: sceNet_Service
                                  > UID: 0x044A3E3D Attr: 0006 - Name: sceMemab
                                  > UID: 0x0449AF65 Attr: 0006 - Name: sceNetAdhocAuth_Service
                                  > UID: 0x044A0B0D Attr: 0006 - Name: sceNet_Library
                                  > UID: 0x018A7233 Attr: 0006 - Name: sceNetAdhoc_Library
                                  > UID: 0x018AF759 Attr: 0006 - Name: sceNetAdhocctl_Library
                                  > UID: 0x00CAC07F Attr: 0006 - Name: sceNetAdhocMatching_Librar
                                  > UID: 0x00CA6E25 Attr: 0006 - Name: sceNetAdhocDownload_Librar
```

JoseAaronLopezGarcia, 2023-05-24:
this should fix it: https://github.com/PSP-Archive/ARK-4/releases/tag/v1684932535

JoseAaronLopezGarcia, 2023-05-24:
Should fix it for all of them.

JoseAaronLopezGarcia, 2023-05-24:
Confirmed fixed.

closed 2023-05-24 by JoseAaronLopezGarcia