PSP-Archive/ARK-4 #130: Boot latest iso/homebrew/ps1 game loaded from custom launcher / vsh

jappoman, 2023-05-17:
### New Feature

As the title says i wolud like to have the option to fastboot the latest game i played (iso, homebrew or ps1 title) in the custom launcher or vsh without the need to look for it again in my game list. I would like to have two methods: 

- fastboot with a button press: boot into vsh/custom launcher and then press a button to boot into the latest booted iso/homebrew/ps1
- fastboot without a button press: boot directly intointo the latest booted iso/homebrew/ps1 once the console is booted

Thanks in advance

krazynez, 2023-05-22:
Why not just the ability to sort by access time? So most recent would be always at the top of the list. Bullet point one could be possible, I would be more concerned about bullet point 2 and the implementations/effects that could cause others.

jappoman, 2023-05-22:
Sort by access time seems good. However, bullet point 1 and 2 looks better to me: you can keep your games sorted by title (which is cool to me) and then boot into the latest game you selected with a simple button press. Maybe the button could be pressed while the console is booting, like pressing DEL to access the bios: in this case you have the best of both worlds: a fast method to return to your latest game and a default method to access the custom launcher. What do you think about?

jappoman, 2023-05-24:
L trigger seems good too!

JoseAaronLopezGarcia, 2023-05-24:
Here: https://github.com/PSP-Archive/ARK-4/actions/runs/5069252045

jappoman, 2023-05-24:
This include isos/psx/homebrew?

JoseAaronLopezGarcia, 2023-05-24:
Yes, everything but the Recovery Menu and UMD Drive.

jappoman, 2023-05-24:
Seems perfect to me! I will test once i get home! Thank you very much

JoseAaronLopezGarcia, 2023-05-24:
I'm closing this. The feature is available in latest release. If there's any issues let me know.

closed 2023-05-24 by JoseAaronLopezGarcia