PSP-Archive/ARK-4 #135: More Translations Needed/Wanted

krazynez, 2023-05-27:
### New Feature

Please if you can translate these for you language it would help us tremendously.

```txt
Custom Firmware Settings
Plugins Manager
USB Charge
Overclock
PowerSave
Autoboot Launcher
Disable Pause on PSP Go
Force Extra Memory
Memory Stick Speedup
Inferno Cache
Old Plugin Support on PSP Go
Skip Sony Logos
Hide PIC0 and PIC1
Prevent hibernation deletion on PSP Go
Hide MAC Address
Hide DLC
Turn off LEDs
```

JoseAaronLopezGarcia, 2023-05-27:
Note: we only need Korean and Netherlands. And maybe recheck the Japanese one.

krazynez, 2023-05-27:
> Note: we only need Korean and Netherlands. And maybe recheck the Japanese one.

for reference the Japanese one we currently have is:
```txt
カスタムファームウェアの設定
プラグインマネージャー
USBチャージ
オーバークロック
パワーセーブ
オートブートランチャー
PSP Goの一時停止を無効にする
メモリ増設
メモリースティックの高速化
インフェルノキャッシュ
PSP Goの旧プラグインのサポート
ソニーロゴをスキップする
PIC0とPIC1の非表示
PSP Goのハイバネーション削除を防止する
MAC アドレスを非表示
DLCを隠す
シャットダウン LED
```
If that is off/not correct let us know.

duncanyoyo1, 2023-05-27:
Netherlands (Dutch)

```Aangepaste firmware-instellingen
Plug-insbeheren
USB-opladen
Overklokken
Energiespaarstand
Autostart-lanceren
Pauze uitschakelen op de PSP Go
Extra geheugen afdwingen
Memory Stick-versnelling
Inferno-cache
Ondersteuning voor oude plug-ins op de PSP Go
Sony-logo's overslaan
PIC0 en PIC1 verbergen
Voorkomen dat slaapstand wordt verwijderd op de PSP Go
MAC-adres verbergen
DLC verbergen
LED's uitschakelen

multi-vitamin, 2023-06-02:
Korean

`커스텀 펌웨어 설정
플러그인 관리
USB 충전
오버클럭
절전
자동 실행 런처
PSP Go에서 일시 정지 비활성화
추가 메모리 강제 할당
메모리 스틱 속도 향상
인페르노 캐시
PSP Go에서 오래된 플러그인 지원
소니 로고 건너뛰기
PIC0 및 PIC1 숨기기
PSP Go에서 절전 모드 삭제 방지
MAC 주소 숨기기
DLC 숨기기
LED 끄기`

JoseAaronLopezGarcia, 2023-06-02:
Thanks, added. I'm closing this for now as we have all translations.

closed 2023-06-02 by JoseAaronLopezGarcia