PSP-Archive/ARK-4 #136: [VITA] Permanent ARK-4 Installation on Adrenaline

rhobiusvoid, 2023-05-29:
### New Feature

Judging from the guides I've seen (and the lack of information above all else due to Google being useless) is that ARK-4 must be executed everytime one boots into Adrenaline, and as such, the Adrenaline Bubble Manager can't be used for booting into games (like Dissidia 012) using ARK-4 and its settings (only for the ARK-Loader itself).

Shouldn't come as a surprise that some people just want the convenience of booting up their favorite PSP game without having to go through a rigmarole of just simply booting up Adrenaline, then finding ARK-Loader in the memory stick, and then choosing the game they want to play. This would also help a lot for certain games like Dissidia 012 and Metal Slug Anthology, which have the unfortunate problem of slow, laggy menus (and outright softlocks in rare cases due to infinite loading).

rhobiusvoid, 2023-05-29:
In case you need a scenario demonstration, here you go.
1. Install ARK-4 as normal through the Adrenaline guide listed in the README.
2. Do an optional step that permanently installs it to Adrenaline's ePSP.
This way, booting into Adrenaline will already have ARK-4 loaded, and Bubble Manager will boot into said games with ARK-4's benefits immediately without worrying about loading ARK-4 first.

JoseAaronLopezGarcia, 2023-05-29:
You can use Adrenaline Bubble Manager (https://github.com/ONElua/AdrenalineBubbleManager/releases) to create an autoboot bubble for ARK Loader, essentially making it permanent, but still not overwriting original Adrenaline, so basically you have both.

rhobiusvoid, 2023-05-29:
That is a solution but it requires me to load up that bubble again and then go back into the memory stick. What I meant is for a proper replacement, where I don't need to load anything and it's just as easy as picking the game I want to play from the LiveArea instead of just booting up the XMB.

JoseAaronLopezGarcia, 2023-05-29:
You can use Autoboot Launcher or Standalone installations (these being the fastest).

JoseAaronLopezGarcia, 2023-05-30:
Closing this as we already have alternatives to achieve this feature.

closed 2023-05-30 by JoseAaronLopezGarcia