PSP-Archive/ARK-4 #140: ATV Offroad Fury Pro - Game shows up as overclocked on the Vita despite overclock being disabled

Pik-a-boo, 2023-06-04:
### PSP/Vita Model

PCH-1000/1100 (Vita Fat), PCH-2000 (Vita Slim), VTE-1000 (Vita TV)

### ARK Version

4.20.51

### Homebrew/GAME (if applicable)

ATV Offroad Fury Pro

### Steps to reproduce

1.Make sure overclocking is disabled.
2.turn on the game.
3.The game will notify you on one of the corners that the console is overclocked.

Note: Not being overclocked is needed for multiplayer to work correctly with non-overclocked/stock PSP consoles.
This issue also happens on Adrenaline.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-06-04:
Is this via Adrenaline or Standalone ?

JoseAaronLopezGarcia, 2023-06-04:
Sounds like easy to patch, similar to tekken 6 perhaps.

JoseAaronLopezGarcia, 2023-06-05:
Latest pre-release adds a patch to trick the game into thinking it's not overclocked.
This gets rid of the annoying "overclock" message, but I'm not sure if multiplayer will be fixed. Some games have they logic tied to the clock frequency and can have issues when overclocked.

On ARK 4.20.50+ you can now disable settings on a per-game basis, so you can have overclock enabled globally but then disable it on specific games by adding the disable line at the end of SETTINGS.TXT like so:

// Disable overclock on ATV Offroad Fury Pro (Europe)
UCES00786, overclock, off

The only "downside" is that you can't handle these sort of settings via XMB Control or Recovery Menu, you have to manually edit the SETTINGS.TXT file to handle your custom settings. You can always use the File Manager in the launcher or recovery.

JoseAaronLopezGarcia, 2023-06-05:
Patch is now included in the latest release.

closed 2023-06-05 by JoseAaronLopezGarcia

Pik-a-boo, 2023-06-05:
> Latest pre-release adds a patch to trick the game into thinking it's not overclocked. This gets rid of the annoying "overclock" message, but I'm not sure if multiplayer will be fixed. Some games have their logic tied to the clock frequency and can have issues when overclocked.
> 
> On ARK 4.20.50+ you can now disable settings on a per-game basis, so you can have overclock enabled globally but then disable it on specific games by adding the disable line at the end of SETTINGS.TXT like so:
> 
> // Disable overclock on ATV Offroad Fury Pro (Europe) UCES00786, overclock, off
> 
> The only "downside" is that you can't handle these sort of settings via XMB Control or Recovery Menu, you have to manually edit the SETTINGS.TXT file to handle your custom settings. You can always use the File Manager in the launcher or recovery.

Wait so it's technically "overclocked"?
TBH that was the main issue I got with the game so it's a shame.

Pik-a-boo, 2023-06-06:
OK just to report back the multiplayer seem to work fine now.