PSP-Archive/ARK-4 #144: CSO Files Crash The PSP When Launched From XMB

KarnFu, 2023-06-08:
### PSP/Vita Model

PSP-3000 (Slim & Bright), PSP-N1000 (PSP Go)

### ARK Version

4.20.54

### Steps to reproduce

- CSO files cause the PSP to crash when launched from the XMB.
- The game could not be started (800200D9) is displayed after I launch IL-2 Sturmovik: Birds of Prey(Game ID: ULES01316) from XMB
- The was working prior to update

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

UMD

krazynez, 2023-06-08:
also try to enable "Unlock Extra Memory" and see if that helps.

KarnFu, 2023-06-08:
> which was the last working version?

4.20.52 was the last working version and I tried to enable "force extra memory" right now and still the same issue. The CSO games that are crashing on XMB so far are God of War Ghost of Sparta, FIFA 08, ans Surfs Up.

JoseAaronLopezGarcia, 2023-06-08:
Fixed: https://github.com/PSP-Archive/ARK-4/releases/tag/r42055

KarnFu, 2023-06-09:
Thanks it's fixed now because the CSO games are loading as normal from the XMB. But IL-2 STURMOVIK: BIRDS OF PREY(ULES01316) ISO, still won't load. See in the attached video

https://github.com/PSP-Archive/ARK-4/assets/131934463/c01af375-50a0-434f-87df-0ea772b5d5fc