PSP-Archive/ARK-4 #146: 4.20.56 Custom Launcher Freeze

LMS737, 2023-06-13:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.56

### Plugins ( if applicable )

Nothing special, everything disabled.

### Steps to reproduce

1. Update to ARK 4.20.56
2. Confirm in System information that we are on ARK 4.20.56 cIPL
3. Reboot device
4. Press SELECT
5. From VSH MENU select Custom Launcher
6. It tries to loads but is stuck. The loading animation is also frozen. Green light stays on.
7. It will not turn off, except plugging out battery.

8. Downgrade to ARK 4.20.35, replacing Ark Save data + Run Ark loader.
9. Reboot device.
10. Press SELECT and select Custom Launcher.
11. No freeze, everything works after downgrade.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-06-14:
Can you try the previous release please: https://github.com/PSP-Archive/ARK-4/releases/tag/r42055

4.20.35 is relievley older and a lot has changed since that release.

LMS737, 2023-06-14:
I just tried the version you suggested (4.20.55). And same result, it freezes for me.

krazynez, 2023-06-14:
Do you have  "Scan category entries"  and/or "Scan DLC files" enabled in Menu Settings for Custom Launcher?

You can also view these settings in Recovery Menu under "Menu Settings"

JoseAaronLopezGarcia, 2023-06-18:
Latest pre-release should fix this.

LMS737, 2023-06-18:
I just updated to the latest pre-release (75407c3d0768bc7fa1c4afc3f74a0bf578c07414) and it did not fix it for me. It still freezes for me.

"Scan category entries" and "Scan DLC files" is disabled. Everything is default.

krazynez, 2023-06-18:
May I ask what region are you in by chance?

JoseAaronLopezGarcia, 2023-06-18:
Can you try using psplink plugin to capture debug output or exceptions?

LMS737, 2023-06-18:
It is a PSP 3001 model so I assume the region is USA. I will try to use the psplink plugin.

krazynez, 2023-06-18:
> It is a PSP 3001 model so I assume the region is USA. I will try to use the psplink plugin.

Okay thanks, yeah PSPLINK should output a much of files/folders when loading ArkMenu. Let us know if you need any help or assistance. Very much appreciated for helping us out with this issue. :+1:

JoseAaronLopezGarcia, 2023-06-18:
Also make sure you properly updated. Using the UPDATE folder is most recommended method.

LMS737, 2023-06-18:
Yeah I just used the UPDATE folder method to double check. But it still freezes. 
I will try to use PSPLink and see if that will help you out.

LMS737, 2023-06-18:
I have followed the steps for setting up PSPLINK. But it seems I need to build a homebrew into an unencrypted .prx file, and then run that.

I think in our case it has to be the program that starts up the custom launcher. How do I obtain this .prx file.

krazynez, 2023-06-18:
> 

No we just need you to run the `usbhostfs_pc` and `pspsh` and see what happens

JoseAaronLopezGarcia, 2023-06-18:
> I have followed the steps for setting up PSPLINK. But it seems I need to build a homebrew into an unencrypted .prx file, and then run that.
> 
> I think in our case it has to be the program that starts up the custom launcher. How do I obtain this .prx file.

Psplink is installed as a plugin, for example:
always, ms0:/psplink/psplink.prx, on

LMS737, 2023-06-18:
Okay, I currently have psplink plugin working I think. I have the usbhostfs_pc and pspsh running. 

usbhostfs window says connected to a device and accepted async connection. And pspsh simply has the host0:/> command prompt.

How do I capture the debug. After running custom launcher, I don't see anything appearing in those 2 windows.

JoseAaronLopezGarcia, 2023-06-18:
Can you fully delete the ARK_01234 savedata and  opy it again from latest pre-release? I wanna make sure it's not some config issue.

LMS737, 2023-06-18:
I fully deleted ARK_01234 savedata, downloaded the latest (dc5b4a660eb310d1dc98450a6b49f9470e66c879).

Copied that over to savedata. Copied over ARK_Live. Ran that, and right after I started the launcher and it froze. 

I unplugged battery, started up again. Launched the launcher, this time no freeze. I did a reboot again, started the launcher and this time it froze again...

LMS737, 2023-06-18:
By the way, I used the same pre-release. But now this time I changed to DEBUG. And on this version, the custom launcher starts without freezing.

JoseAaronLopezGarcia, 2023-06-18:
> By the way, I used the same pre-release. But now this time I changed to DEBUG. And on this version, the custom launcher starts without freezing.

That just baffles the hell out of me since the custom launcher itself doesn't have any "debug build", that's mostly for the core CFW.
Anyways, try this release (with psplink active and checking any output from pspsh), regular build (not debug build):
https://github.com/PSP-Archive/ARK-4/releases/tag/v1687123292

LMS737, 2023-06-18:
Okay, I have the "Loading all modules ... Ready".
But the moment I start any game or select custom launcher, I hear the USB disconnecting and reconnecting again. And that seems to disconnect the psplink connection. Maybe this is a Windows related issue? psplink plugin is installed as always.

I think I will try on linux next...

LMS737, 2023-06-18:
Ok I managed to get the output using Linux.

host0:/> Loading all modules ... Ready
host0:/> data loaded
setting up main menu entries
setting up FTP
setting up settings menu

And that is where it freezes. Hope this helps you to find the issue.

JoseAaronLopezGarcia, 2023-06-18:
> Ok I managed to get the output using Linux.
> 
> host0:/> Loading all modules ... Ready host0:/> data loaded setting up main menu entries setting up FTP setting up settings menu
> 
> And that is where it freezes. Hope this helps you to find the issue.

It "helps" but it blows my mind since it's freezing where I least expect it to.

JoseAaronLopezGarcia, 2023-06-18:
Lets try this one: https://github.com/PSP-Archive/ARK-4/releases/tag/v1687128166
It should give more debug output, hopefully.

LMS737, 2023-06-18:
Yes, I get the following:

host0:/> data loaded
setting up main menu entries
setting up FTP
setting up settings menu
setting up exit

And then freeze.

JoseAaronLopezGarcia, 2023-06-18:
ok now try this one: https://github.com/PSP-Archive/ARK-4/releases/tag/v1687129381

LMS737, 2023-06-18:
Nice! The issue is fixed. No freeze anymore :)

closed 2023-06-19 by JoseAaronLopezGarcia