PSP-Archive/ARK-4 #147: Custom Launcher crashed when searching for games

K-K-K-K-K-K-K-K-K, 2023-06-14:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.56

### Plugins ( if applicable )

vsh, ms0:/seplugins/categorieslite/category_lite.prx, on 
vsh, ms0:/seplugins/hold.prx, on psp, ms0:/seplugins/cwcheat/cwcheat.prx, on 
ULES00151 ULUS10041 ULES00502 ULUS10160, ms0:/seplugins/cwcheat/cwcheat.prx, off 
ULES00151 ULUS10041 ULES00502 ULUS10160, ms0:seplugins/cheatdevice_remastered_lite.prx, on

### Steps to reproduce

1. open the custom launcher. waited until it load up.
2. press 'down' to browse games.
3. it will freeze up. the system does not respond to any control.
4. needed to do forced shutdown by holding down the power slider for five seconds and turning the system back on again.

I think the problem is there's not enough memory to load all pictures, as I used PSP 1000 and my sd card contained all of my PSP games library(90GB).

These are solutions that I think can be implemented.

1. use catagory_lite subfolders system. Only shown games in one subfolder at a time.
2. add an option to only show games as a list of .ISO file names. saving memory on loading every game's preview images.
3. add a shortcut to open 'ms0:/ISO' directory in file explorer quickly and launch the game directly from there.
4. Add settings to only load pictures on entries that were currently on screen only. Like in Sony’s XMB.

### USB Charge

Disabled

### Overclock

UMD

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

UMD

JoseAaronLopezGarcia, 2023-06-14:
"I think the problem is there's not enough memory to load all pictures, as I used PSP 1000 and my sd card contained all of my PSP games library(90GB)."

I doubt it, I have memory sticks with over 100 ISO games and they work fine. The custom launcher is also not made to use extra RAM on slim models, so it's always running as if it's a 1K. My best bet is there's maybe a game with an icon that it doesn't like, or maybe the game is compressed and it made the CSO reader crash (not very likely though, the CSO reader is pretty robust as of now, but who knows).

Try moving games away one at a time until it stops crashing, perhaps you can figure out which one it is and what format it's on (ISO/CSO/PBP/etc).

krazynez, 2023-06-14:
Could also be non UTF-8 Chars that are causing it possibly also.

JoseAaronLopezGarcia, 2023-06-14:
yeah I think that is it. They were fixed many releases ago but it seems we broke them again.

K-K-K-K-K-K-K-K-K, 2023-06-14:
The custom still crashed but noticeably better than it was before. It took about 30 seconds instead of 2-3 seconds to crash now.

The folder I removed is the one that contained English-translated Japanese titles.

This is the list of games in that folder.

`
7th Dragon 2020-II   (English Patched v0.91).iso
air (English v0.9).iso
Digimon Adventure (English v1.2).iso
Dot Hack Link (English patched   v1.0.2).iso
Eiyuu Densetsu Ao no Kiseki (English   v15).iso
Ever17 (English Beta 4).iso
GA Slapstick Wonder Land (English   v1.0).iso
Genso Suikoden (Google Translate).iso
Grand_Knights_History_v1.16.iso
Hatsune Miku Project Diva Extend (English   v2.0).iso
Kidou Senshi Gundam AGE Universe Accel   (English v1.0).iso
Kidou Senshi Gundam Gundam vs. Gundam   NEXT PLUS (English Patched).iso
Kingdom Hearts Birth by Sleep Final Mix   (English Patched).iso
Kurohyou_Ryu_ga_Gotoku_Shinshou (English   11-2020).iso
Last Ranker (English v2.3).iso
La_Pucelle_Ragnarok_v1.1.2.iso
Mahou_Shoujo_Lyrical_Nanoha_AS_Portable-English.iso
Mahou_Shoujo_Madoka_Magica_Portable_JPN_PSP-BAHAMUT.iso
Mahou_Shoujo_Nanoha_As_Portable_The_Gears_of_Destiny_English.iso
Mystery Dungeon Shiren The Wanderer 3   (English Patched).iso
Nayuta no Kiseki (Eng v4.15 - EDIT   v0.9.1).iso
Never 7 (English v2).iso
Ore no Imouto (English v1) Disc 1.iso
Ore no Imouto (English v1) Disc 2.iso
Planetarian_Chiisana_Hoshi_no_Yume   (English).iso
Remember11-eng-v2.1.iso
Samurai Dou Portable (English   Patched).iso
Sound of the Sky - Quintet of Maidens   (English Patched v1.0).iso
Steins Gate (English v1.0.1).iso
Suzumiya_Haruhi_no_Yakusoku (English   Patched).iso
Tales of Phantasia - Narikiri Dungeon X   (English Patched).iso
Tales of the World - Radiant Mythology 3   (English v1.02).iso
Tales Of The World Radiant Mythology 2   (English v0.3).iso
Tantei Opera Milky Holmes (English   v0.4).iso
The Idol Master SP - Missing Moon   (English Patched).iso
The Idol Master SP Perfect Sun (English   Patched).iso
The Idolmaster SP - Wandering Star   (English Patched).iso
Toaru_Majutsu_no_Index (English   Patched  03_12_19_2).iso
Yarudora Portable Double Cast (English   v1.1).iso
Zero no Kiseki (English Patched).iso
`

ca1e, 2023-06-16:
#114

JoseAaronLopezGarcia, 2023-06-18:
Closing as a duplicate of https://github.com/PSP-Archive/ARK-4/issues/114

closed 2023-06-18 by JoseAaronLopezGarcia

JoseAaronLopezGarcia, 2023-06-18:
Latest pre-release shoud fix this.