PSP-Archive/ARK-4 #152: Wifi host and webserver

K-K-K-K-K-K-K-K-K, 2023-06-16:
### New Feature

When using fttp function PSP is a pain because it doesn't support WPA2

That mean, to do wireless file transfer, I have to log into my access point to disable security, restart it, then finally got to do file transfer. It very much unpractical.

That's why, isn't it much better to let's PSP be the host instead? and let's other device connect to it. This way, just have to config on PC's wifi setting instead of messing with the access point's configuration.

And why not add a webserver for ability to transfer file without ftp client too? If psp cannot handle hosting full-featured webpage while transfering a file, then make a simple html page with an upload button will do. Save all file to a folder in memory stick and use Ark’s builtin file manager to move file around afterward.

krazynez, 2023-06-16:
the PSP can be a server and a client with ftp.

JoseAaronLopezGarcia, 2023-06-18:
The PSP doesn't have hardware to create an Access Point like Android phones can do.
My recommendation is to just use your Android's hotspot with no password for PSP online stuff. Simply disable it when not needed. That's how I connect my PSPs to the internet.

bucanero, 2023-11-04:
@K-K-K-K-K-K-K-K-K  regarding the webserver request, Apollo save tool (PSP) has a simple web server available that will give access to the whole PSP file system (`ms0:/`, `ef0:/`). If you install Apollo, check "Tools" menu > "Network tools" > "web server".
Note: it's quite limited in resources (no multi-thread, no uploading, or high-speed transfers), but it can help when you need quick access to some small files.

It can also be helpful to other devs as an example implementation.

JoseAaronLopezGarcia, 2024-03-09:
Closing this as there are alternatives as mentioned above.

closed 2024-03-09 by JoseAaronLopezGarcia