PSP-Archive/ARK-4 #153: CWCheat isn't showing any cheat.

Kenezinn, 2023-06-17:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.56

### Plugins ( if applicable )

game, ms0:/PSP/SAVEDATA/ARK_01234/Poggers/remotejoy/RemoteJoyLite.prx, off
vsh, ms0:/PSP/SAVEDATA/ARK_01234/Poggers/remotejoy/RemoteJoyLite.prx, off
pops, ms0:/PSP/SAVEDATA/ARK_01234/Poggers/remotejoy/RemoteJoyLite.prx, off

xmb, ms0:/PSP/SAVEDATA/ARK_01234/Poggers/XMBIH/XMBIH.PRX, on

game, ms0:/PSP/SAVEDATA/ARK_01234/Poggers/CWCHEAT/CWCHEAT.PRX, on

### Steps to reproduce

1. Start a game.
2. Keep Select pressed to show CWCheat.
3. CWCheat menu is shown.
4. No cheat can be selected, it doesn't show anything.

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Homebrew

### Memory Stick Speedup

Always

### Inferno Cache

Game

JoseAaronLopezGarcia, 2023-06-17:
This just means the cheat database doesn't have cheats for that game. This isn't really an ARK issue. Download an updated cheat database.

Kenezinn, 2023-06-17:
it was working properly when using PRO, i also tried to change the cheat database, it just doesn't work

a0a0max, 2023-06-18:
[seplugins.zip](https://github.com/PSP-Archive/ARK-4/files/11780984/seplugins.zip)
Try this package. It works fine.

JoseAaronLopezGarcia, 2023-06-18:
@Kenezinn can you point to us which version of cwcheat you are using?

Kenezinn, 2023-06-18:
> [seplugins.zip](https://github.com/PSP-Archive/ARK-4/files/11780984/seplugins.zip) Try this package. It works fine.
> 
> that's happened to me before #52 It should be investigated why the two same cwchet plugins versions having different signatires works differently

oh my god, thank you so much! the package worked just fine, it seems like cwcheat just doesn't work out of seplugins folder or something idk cause i tried this database before

Kenezinn, 2023-06-18:
> @Kenezinn can you point to us which version of cwcheat you are using?

it's 0.2.3, but it's working fine now with the package @a0a0max mentioned

JoseAaronLopezGarcia, 2023-06-18:
> > @Kenezinn can you point to us which version of cwcheat you are using?
> 
> it's 0.2.3, but it's working fine now with the package @a0a0max mentioned

Do you have a link of the one that wasn't working?

Kenezinn, 2023-06-18:
> Do you have a link of the one that wasn't working?

yeah, here: https://www.mediafire.com/file/pdsncs7l09tt20s/CWCheat+PSP+6.61.zip

JoseAaronLopezGarcia, 2023-06-18:
> > Do you have a link of the one that wasn't working?
> 
> yeah, here: https://www.mediafire.com/file/pdsncs7l09tt20s/CWCheat+PSP+6.61.zip

Thanks, I'll investigate it. I know for a fact that some (quite a few) plugins try to detect the CFW it's running on and do different things/patches. Perhaps something like this is happening.

JoseAaronLopezGarcia, 2023-06-25:
Ok so I've checked the differences between the two and can confirm that the "issue" is the cheat database itself.
@Kenezinn the one you were using is only 2.5MB in size, while the one @a0a0max is 8.5MB in size.
Cwcheat itself wasn't having any issue to detect the game, it simply didn't have any cheats for that game in the database.

I'm gonna keep this in mind for the upcoming wiki page to explain this in the plugins section.

closed 2023-06-25 by JoseAaronLopezGarcia