PSP-Archive/ARK-4 #159: PSP GO kernel exploit error

VisionR1, 2023-06-25:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.58

### Steps to reproduce

i try install 4.20.58 but i get this, in previous 4.20.57 all is good

Loading ARK-4
Reading kxploit file...
Loading Kxploit at ef0:/PSP/GAME/ARK_Loader/K.BIN
Scanning stubs for kxploit...
Doing kernel exploit...
ERROR (1): Exploit failed! 
Exiting in 10 seconds...

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-06-25:
Oh yeah this is a compilation issue which has produced the wrong kernel exploit files. There's supposed to be three different files depending on device but it sometimes only produces multiple copies of the same file.

This happens sometimes at random and it's hard to notice. Just download any of the latest pre-releases, they should have correct kernel exploit files.

VisionR1, 2023-06-25:
> Oh yeah this is a compilation issue which has produced the wrong kernel exploit files. There's supposed to be three different files depending on device but it sometimes only produces multiple copies [...]

Yeah, the K.BIN from ARK_01234 folder and the K.BIN from PSVita folder is the same

closed 2023-06-25 by VisionR1