PSP-Archive/ARK-4 #162: Cant install ark4

Kitki30, 2023-06-27:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

i dont know (latest) 

### Steps to reproduce

Just clicked ark loader on psp and it said that he cant load  k.(something) exploit
and it turned off pso

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

Kitki30, 2023-06-27:
fixed it I installed pre-release

JoseAaronLopezGarcia, 2023-06-27:
https://github.com/PSP-Archive/ARK-4/releases/tag/r42059

closed 2023-06-27 by JoseAaronLopezGarcia

Kitki30, 2023-06-27:
ok