PSP-Archive/ARK-4 #165: ARK Launcher: Files - Startup Folder

svensvond, 2023-06-28:
### New Feature

When selecting default menu: files - it would be great to have the files view be on a default folder other than root.
Since i organize my isos in a subfolder i always have to navigate to that folder.

JoseAaronLopezGarcia, 2023-06-28:
yeah this can be done, give me a few minutes.

svensvond, 2023-06-28:
thats what i call service! really appreciate that.

JoseAaronLopezGarcia, 2023-06-28:
https://github.com/PSP-Archive/ARK-4/releases/tag/v1687943030

Here's how I've done it: whenever you save settings, it'll remember the folder you are currently on in the browser and automatically load it when you start. You can force settings to be saved by pressing X anywhere in the settings menu.

svensvond, 2023-06-28:
very nice - it has a little great side effect:
Since you seem to cache the last file/folder display, it does not need to read it from sd again (if you do not change any directory) which is much faster on larger cards (like 512GB)

closed 2023-06-29 by JoseAaronLopezGarcia