PSP-Archive/ARK-4 #167: ARK Launcher: Hide files / folders with hidden attribute

svensvond, 2023-06-28:
### New Feature

I do not want the user (the kids ;) ) to see different folders / files in the file-browser. I can set the hidden attribute in - for example -windows, but it has no affect.

JoseAaronLopezGarcia, 2023-06-28:
It doesn't seem like the PSP IO layer has support for Windows-style file attributes but rather it uses a more UNIX approach.

This means to implement this I have to implement it the UNIX way: "normalfile.iso" -> ".hiddenfile.iso"
Placing a dot at the start of the file/folder is considered "hidden" in UNIX, and that's pretty easy to implement.

I will also have to implement an option in the settings menu to "Show Hidden".

JoseAaronLopezGarcia, 2023-06-28:
here you go: https://github.com/PSP-Archive/ARK-4/releases/tag/v1687945858

closed 2023-06-29 by JoseAaronLopezGarcia

svensvond, 2023-06-29:
Ok, looks good.
There is no option but to rename the files / folders? ok.
But maybe the apps / games could also be hidden in the game-view, currently it can be stilled accessed there.

JoseAaronLopezGarcia, 2023-06-29:
> Ok, looks good. There is no option but to rename the files / folders? ok. But maybe the apps / games could also be hidden in the game-view, currently it can be stilled accessed there.

If you rename /ISO/ to /.ISO/ you won't see them in the game section, only in the browser if you enable "Show Hidden".
Same for /PSP/GAME/ to /PSP/.GAME/ though instead of doing that to /PSP/GAME/ I would use /PSP/APPS/ instead (which the Launcher is compatible with).

svensvond, 2023-06-29:
i have inside /PSP/GAME folders like LME .lme_installer and want this to be hidden in games overview

JoseAaronLopezGarcia, 2023-06-29:
ok hold on I'll get around to it once I finish making some cool changes to the file browser (you're gonna like them).

svensvond, 2023-06-29:
:)

JoseAaronLopezGarcia, 2023-06-29:
Here you go: https://github.com/PSP-Archive/ARK-4/releases/tag/v1688039004
Game should now also be hidden if they start with a '.', I've also improved the file browser when it comes to viewing and launching games, so you might want to go around your ISO folder and let me know what you think.

JoseAaronLopezGarcia, 2023-06-29:
Ok I've added the option to disable ICON0 in Browser and also made it use the Skip GameBoot feature and the Start button for Current game, you have all those options so you can leave it to your linking.

svensvond, 2023-06-29:
nice.

svensvond, 2023-06-29:
initially there where no ICON0 in Browser showing, i toggled skipgameboot option: Now i cant get rid of the ICON0 image display in browser any more - what happened? I even deleted the settings file: Images in Filebrowser are still beeing loaded. I dont see any option to just turn off ICON0 display. I dont see any explicit option for turning images on and off either.

JoseAaronLopezGarcia, 2023-06-29:
Update to latest pre-release.

svensvond, 2023-06-29:
works like a charm - maybe one tiny little thing. I really like the "game boot preview thing" in file browser! Could you bind that to a key, for example right trigger in any case? So that if you want to quick start press X or RIGHT TRIGGER for game boot.

svensvond, 2023-06-29:
ok, had start button set to "last game" .. but yes, this works.