PSP-Archive/ARK-4 #168: Can't access FTP server

MerGr, 2023-06-29:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.59

### Steps to reproduce

1. Go into custom launcher
2. Press Triangle
3. Try to select any option in between first and last
4. Launcher just skips the in between options and goes between first and last only

### USB Charge

Disabled

### Overclock

Game

### Power Save

VSH

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-07-06:
Any confirmation on this?

JoseAaronLopezGarcia, 2023-07-24:
Closing due to inactivity

closed 2023-07-24 by JoseAaronLopezGarcia