PSP-Archive/ARK-4 #169: Add music/videos option to custom launcher

Kitki30, 2023-06-29:
### New Feature

hi can you add **music** and **videos** option in **Custom Launcher** i think **Custom Launcher** is **better** than Xmb but it doesnt have options like **music, videos** and **browser**(i dont need it but you can ad it)

JoseAaronLopezGarcia, 2023-06-29:
- Music player is already a thing, you can open any mp3 or even more than one mp3 to make a Playlist.
- Video player, while possible, is a bit harder to do and I didn't want to add too much more code due to memory limitations. We might be able to do it if we manage to lower memory consumption, which should be doable but don't expect it any time soon.
- As for Browser, if you mean Web Browser, no thats not possible, too much work and it wouldn't even get close to what the original Browser can do.

Kitki30, 2023-06-30:
ok

Kitki30, 2023-06-30:
ok

Kitki30, 2023-06-30:
ok

JoseAaronLopezGarcia, 2023-06-30:
Don't close this, we will work on the video player.

Kitki30, 2023-06-30:
sorry for spam it was my internet (it was laggy and i clicked send many times)

JoseAaronLopezGarcia, 2025-03-16:
You can view movies in PMF format.

closed 2025-03-16 by JoseAaronLopezGarcia