PSP-Archive/ARK-4 #171: ARK Launcher - File Browser: Loading Times

svensvond, 2023-06-29:
### New Feature

**Tweak loading times to browse directories with a larger amount of isos**

When putting a larger amount of isos in a directory, loading of this directory takes quite some time. I think in some way the iso files are processed (meta data?). Here are some examples:

Entering a directory of isos (270): 40 sec
Entering a directory of (empty) files (270): 38 sec
Entering a directory of (empty) directories (270): 5 sec

Since it should be just a listing of items regardless of its type, shouldnt it be equally fast?

JoseAaronLopezGarcia, 2023-06-29:
I'll look into it.

svensvond, 2023-06-29:
looks to me if its related to getFileSize() which is only executed for files. (unsigned BrowserFile::getFileSize(){ ?)
I dont care about the file size on the psp itself actually, faster loading times would weight much more.

JoseAaronLopezGarcia, 2023-07-06:
Closing as duplicate of https://github.com/PSP-Archive/ARK-4/issues/178

closed 2023-07-06 by JoseAaronLopezGarcia