PSP-Archive/ARK-4 #172: Can’t install ark-4

hozluvice, 2023-06-30:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

latest

### Steps to reproduce

trying to install ark-4 with previous 6.61 PROMOD-C2p infinity followed youtube step by step and will not have the new ark-4

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

svensvond, 2023-06-30:
Please describe the exact technical problem (e.g. error messages) 
Also use "Steps to reproduce" as a detailed workflow description

hozluvice, 2023-06-30:
I did this exact following procedure of this video https://youtu.be/bE8mw7RRqGo

I load up the ark loader, I then install it throught infinity once Install I reboot the psp but ark 4 does not appear on recovery menu nor new menus appear

krazynez, 2023-07-01:
> I did this exact following procedure of this video https://youtu.be/bE8mw7RRqGo
> 
> I load up the ark loader, I then install it throught infinity once Install I reboot the psp but ark 4 does not appear on recovery menu nor new menus appear

Please provide exact steps taken. It will help us out more than linking to a video.

JoseAaronLopezGarcia, 2023-07-06:
Make sure you boot Infinity again to enable ARK.

JoseAaronLopezGarcia, 2023-07-08:
has this been solved?

hozluvice, 2023-07-09:
Yes

closed 2023-07-09 by krazynez