PSP-Archive/ARK-4 #173: No key for update check

Kitki30, 2023-06-30:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.59

### Homebrew/GAME (if applicable)

Launcher

### Steps to reproduce

1. Open custom launcher 
2. Press triangle
3. Go to Network
4. And you see Press [] to check updates and if you start ftp server you can see FTP Server is running @ 192.168.0.17. Press () to stop and also Press [] to check for updates

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Always

Kitki30, 2023-06-30:
My point is that there is no button in place of [] and ().

JoseAaronLopezGarcia, 2023-06-30:
I don't understand what you mean here. It's working as intended.

krazynez, 2023-06-30:
> My point is that there is no button in place of [] and ().

Do you mean have images of the buttons instead of text?

Kitki30, 2023-06-30:
yes image of button instead of []

Kitki30, 2023-06-30:
@krazynez

krazynez, 2023-07-01:
How is this a bug?

Kitki30, 2023-07-02:
no button showed

krazynez, 2023-07-02:
There is no button its just text

JoseAaronLopezGarcia, 2023-07-06:
It would be extremely complicated to do this unless the PSP's font has the PS buttons in it. I'm not sure, might have to look at how the XMB does it.

closed 2023-07-24 by JoseAaronLopezGarcia