PSP-Archive/ARK-4 #174: Plugin Problems

K-K-K-K-K-K-K-K-K, 2023-07-01:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.59

### Plugins ( if applicable )

vsh, ms0:/seplugins/categorieslite/category_lite.prx, on
vsh, ms0:/seplugins/hold.prx, on psp, ms0:/seplugins/cwcheat/cwcheat.prx, on
ULES00151 ULUS10041 ULES00502 ULUS10160, ms0:/seplugins/cwcheat/cwcheat.prx, off
ULES00151 ULUS10041 ULES00502 ULUS10160, ms0:seplugins/cheatdevice_remastered_lite.prx, on

### Steps to reproduce

1.Use this database(normal version)
https://github.com/Saramagrean/CWCheat-Database-Plus-/blob/master/cheat.db

2.Download plugins from this site
https://www.gamebrew.org/wiki/TempAR_PSP
https://www.gamebrew.org/wiki/CWCheat_PSP

3.make 2 versions of plugins.txt with either tempar or cwcheat so they can be easily hot swap and test.

3.copy folder of both plugins to memory stick, copt cheatdb into both of plugins’ folders.

problems:
Cwcheat: do not show list of available cheat entries, only blank screen, [load cheat] do nothing.
Tempar: cheat entries loaded correctly, but system will frozen up if enable tempar_lite like the guide said. Did not test if cheat actually work yet.
vsh, ms0:/seplugins/TempAR/tempar_lite.prx, on

### USB Charge

Disabled

### Overclock

Game

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-07-01:
So let me get this straight, you're using an old cheat database and cwcheat can't find any cheats for your game in that database, and this is somehow an ARK issue?
I'm sorry but this doesn't make any sense at all.
Just use a better cheat database.

K-K-K-K-K-K-K-K-K, 2023-07-01:
> So let me get this straight, you're using an old cheat database and cwcheat can't find any cheats for your game in that database, and this is somehow an ARK issue?
> I'm sorry but this doesn't make any sense at all.
> Just use a better cheat database.

The cheat was updated few months ago and the lasted psp game was released years ago. It’s good enough for most game, and tempAR can opened and see entries shown up.

the cheat for that particular game do existed in that db when I opened it with notepad and search, and tempAR verified that’s it’s working.

Only cwcheat shown blank, empty page, no option to select.

and tempar_lite.prx make system froze up.

I didn’t even know if there’s a better cheat than this out there, every tutorial I found pointed to that repo.

JaegerFox, 2023-07-14:
this is not a problem related to ARK-4 (I think), the database you are using is too large for cwcheat to read, TempAR supports larger lists.
tempar lite its just for PS1 games

xan1242, 2023-08-09:
Hello.

I am also having severe problems loading plugins of my own.

So @davidgfnet and I investigated and we came down to an interesting discovery.

The CFW loads the PRX modules _before_ the game's address space and before the game boots. The issue is that the space between the game and the module is _very small_, so if your plugin's code is big enough (the .text segment) it'll just get discarded by the kernel and not loaded.

(The game in question here is Tag Force, Japanese release)

If I do a modinfo of it via pspsh, this is the result:

```
UID: 0x04029F1B Attr: 0000 - Name: modehsys
Entry: 0x08804000 - GP: 0x088E9410 - TextAddr: 0x08804000
TextSize: 0x0006DE48 - DataSize: 0x0006F5D8 BssSize: 0x00000000
Segment 0: Addr 0x08804000 - Size 0x00619680
```

As we can see, the game's base address is 0x08804000 which is pretty run of the mill.

Now, say I load a module that waits and does not let the game boot, where does it end up?

```
UID: 0x03D4FE77 Attr: 0000 - Name: TFTestIntermediary
Entry: 0x0880008C - GP: 0x0880A830 - TextAddr: 0x08800000
TextSize: 0x00002678 - DataSize: 0x0000008C BssSize: 0x00000000
Segment 0: Addr 0x08800000 - Size 0x00002834
```

It ends up at 0x08800000.

This effectively gives us 0x4000 bytes of space of code to work with. That amount of space isn't even enough to put a printf in, only very minimal code.

The issue could be worked around if the plugin is loaded from a separate thread, which is exactly what psplink does!

https://github.com/pspdev/psplinkusb/blob/master/bootstrap/main.c

So an option to force high memory addresses for the plugins would be a nice one!

As a proof-of-concept, I've bootstrapped my own plugin just like psplink and it worked! I'll continue iterating on this and make a prx plugin loader of my own to see if I can work around these problems more efficiently.

JoseAaronLopezGarcia, 2023-08-09:
> Hello.
>
> I am also having severe problems loading plugins of my own.
>
> So @davidgfnet and I investigated and we came down to an interesting discovery.
>
> The CFW loads the PRX modules _before_ the [...]

ARK has an API function that no other CFW has that can help fix this issue:

- Setting up callback:
https://github.com/PSP-Archive/ARK-4/blob/e076016520ee0d6c7192bbca5b6b50a345e2093a/core/compat/psp/main.c#L74
- Actual callback
https://github.com/PSP-Archive/ARK-4/blob/e076016520ee0d6c7192bbca5b6b50a345e2093a/core/compat/psp/syspatch.c#L234

This allows you to basically handle every plugin that is loaded and do whatever you want to it first or do a custom start.
So you can make a plugin that manages the loading of other plugins.
This will only work on ARK, PRO and ME do not have this functionality.

JoseAaronLopezGarcia, 2023-08-09:
Another way we can potentially fix the issue is to manually allocate the region of RAM that we know is always used by the game's PRX (0x08800000) before loading plugins and freeing it up after loading plugins, this should prevent it from loading plugins in that memory range:

- SceUID memid = sceKernelAllocPartitionMemory(2, "", PSP_SMEM_Addr, SIZE, 0x08800000);
- Load and Start Plugins
- sceKernelFreePartitionMemory(memid);

xan1242, 2023-08-09:
Sounds pretty good!

But I've also noticed that this is jank needed only for plugins that load in userspace. If you compile it as a kernel prx, it's OK.

It would be a nice feature to include because it would be a real nice addition for game mods on CFW.

JoseAaronLopezGarcia, 2023-08-09:
> 
I've made this build with the "fix" using sceKernelAllocPartitionMemory, extract FLASH0.ARK into /PSP/SAVEDATA/ARK_01234/ and run ARK_Loader again to flash the new files, then let me know if it actually solves anything.

[FLASH0.zip](https://github.com/PSP-Archive/ARK-4/files/12302088/FLASH0.zip)

xan1242, 2023-08-09:
```
UID: 0x03D4A871 Attr: 0000 - Name: TFTestIntermediary
Entry: 0x08C00018 - GP: 0x08C0A4D0 - TextAddr: 0x08C00000
TextSize: 0x00002310 - DataSize: 0x0000008C BssSize: 0x00000000
Segment 0: Addr 0x08C00000 - Size 0x000024DC
```

So, the relocation worked but despite this the module isn't even being loaded if it exceeds a certain size... The problem lies probably somewhere else then.

Thank you for the build tho, it really helped to understand some things.

JoseAaronLopezGarcia, 2023-08-09:
So is the build better or not? Should I leave the code?

xan1242, 2023-08-09:
Nah, no need at the moment. That actually wasn't the issue. I actually managed to make it stay resident in memory _above_ the game executable just fine by loading it from a separate thread in a kernel module.

Now I'm trying to start it and it's not starting for some reason... Figuring that out now.

xan1242, 2023-08-10:
OK so I got my module to load consistently.

For some reason, loading libkprintf before I load my own module seems to fix the problem. I don't fully understand why, but loading libkprintf first delays it long enough for the game to load its module first, but not _start_ it first.

So my module loads after the game module (modehsys in this situation) but starts up before it does.

In theory the fix would be for all modules to hook right at module loading (which sctrlHENSetPluginHandler should be, if I'm not mistaken).

I am not sure if there's an easy way to bootstrap all plugins like this without plugins themselves having to do that on their own with a separate plugin.

I'll test out the sctrlHENSetPluginHandler properly now, though.

Here's the my janky code that works (without sctrlHENSetPluginHandler):
```c
#include <pspsdk.h>
#include <pspkernel.h>
#include <stdio.h>
#include <string.h>

#define MODULE_NAME "TFTestIntermediary"
#define MODULE_PATH "ms0:/seplugins/TFTest.prx"
#define KPRINTF_MOD_PATH "ms0:/usr/local/lib/libkprintf/libkprintf.prx"

#ifndef __INTELLISENSE__
PSP_MODULE_INFO(MODULE_NAME, PSP_MODULE_KERNEL, 1, 1);
PSP_MAIN_THREAD_ATTR(0);
#endif

int module_stop(int argc, char* argv[]);

int build_args(char *args, const char *execfile, int argc, char **argv)
{
	int loc = 0;
	int i;

	strcpy(args, execfile);
	loc += strlen(execfile) + 1;
	for(i = 0; i < argc; i++)
	{
		strcpy(&args[loc], argv[i]);
		loc += strlen(argv[i]) + 1;
	}

	return loc;
}

SceUID load_module(const char *path, int flags, int type)
{
	SceKernelLMOption option;
	SceUID mpid;

	/* If the type is 0, then load the module in the kernel partition, otherwise load it
	   in the user partition. */
	if (type == 0) {
		mpid = 1;
	} else {
		mpid = 2;
	}

	memset(&option, 0, sizeof(option));
	option.size = sizeof(option);
	option.mpidtext = mpid;
	option.mpiddata = mpid;
	option.position = 0;
	option.access = 1;

	return sceKernelLoadModule(path, flags, type > 0 ? &option : NULL);
}

int load_start_module(const char *name, int argc, char **argv)
{
	SceUID modid;
	int status;
	char args[1024];
	int len;

	modid = load_module(name, 0, 0);
	if(modid >= 0)
	{
		len = build_args(args, name, argc, argv);
		modid = sceKernelStartModule(modid, len, (void *) args, &status, NULL);
	}
	else
	{
		Kprintf("lsm: Error loading module %s %08X\n", name, modid);
	}

	return modid;
}

int load_start_module2(const char *name, SceSize args, void *argp, int type)
{
	SceUID mod = load_module(name, 0, type);
	if (mod < 0)
	{
		Kprintf("Error 0x%08X loading module.\n", mod);
		return mod;
	}

	mod = sceKernelStartModule(mod, args, argp, NULL, NULL);
	if (mod < 0)
	{
		Kprintf("Error 0x%08X starting module.\n", mod);
		return mod;
	}

	return mod;
}

int _main(SceSize args, void *argp)
{
	SceUID modid;
	modid = load_start_module2(KPRINTF_MOD_PATH, args, argp, 0);
	if(modid < 0)
		Kprintf("Failed to Load/Start module '%s' Error: 0x%08X\n", MODULE_PATH, modid);

	modid = load_start_module2(MODULE_PATH, args, argp, 1);
	if(modid < 0)
		Kprintf("Failed to Load/Start module '%s' Error: 0x%08X\n", MODULE_PATH, modid);
	
	
	sceKernelSelfStopUnloadModule(1, 0, NULL);
	sceKernelDelayThread(2000000);
	sceKernelExitGame();

	return 0;
}

int module_start(SceSize args, void* argp)
{
	SceUID uid;

	uid = sceKernelCreateThread("TFTestBoot", _main, 32, 0x10000, 0, 0);
	if(uid < 0)
	{
		return 1;
	}
	sceKernelStartThread(uid, args, argp);
	

    return 0;
}
```

xan1242, 2023-08-10:
So I've figured out what's happening.

This is the chain of events:

- You select your game and attempt to boot it
- CFW intercepts it and starts loading plugins immediately
- Game executes and resets the userspace memory

My module in particular was a userspace one which didn't manage to execute in time if it was too large.

Plugins are expected to be kernel modules, hence why their memory always survives.

The solution to the problem is simple - wait for the game to load and then load your userspace modules. In my particular instance, instead of loading libkprintf, I simply waited for the game to start like this:

`while(sceKernelFindModuleByName(MODULE_NAME_INTERNAL) == NULL) sceKernelDelayThread(1);`

And ever since then the module is loaded normally.

I propose for a feature to be added that will do this at a CFW level. Like I suggested earlier, hook `sceKernelLoadModule` and load+start plugins right as the game's userspace module loads.

I'd prefer this way because it'd be safer than polling with `sceKernelFindModuleByName`

FULANO93, 2023-08-15:
I'm having the same issue with CwCheat, any game I try to enable some cheat just shows this -
Patapon 2 [US] ID: HB4B12D3BC

The correct game just doesn't appear with its cheats,
I believe it is not loading the database correctly, even when I put it to reload the database, if I active plugin, the prx file changes the name to CWCHEA~2.prx

Something like CheatMaster, it works fine, but when I try to load the database, it just freezes the console and turns off...
The only one I can use normally is TempAR, but this one, even if it works, the cheats for PS1 games don't appear, only the PSP ones, even though I use tempar_lite for PS1 games... with that I am unable to use cheats on PS1 games... I believe it's a problem that Ark-4 handles with plugins, because on my PSP 6.20 Pro-C2 the 3 plugins work normally..

For informational purposes:
PSP 2000 model 2g
Ark-4 6.20.63 cIPL
cheatpops.db 3.60 MB
cheat.db 4.43 MB

JoseAaronLopezGarcia, 2023-09-12:
This was fixed in latest version.

closed 2023-09-12 by JoseAaronLopezGarcia