PSP-Archive/ARK-4 #176: UMD crashed custom launcher

Kitki30, 2023-07-01:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.59

### Homebrew/GAME (if applicable)

idk but i think its UCES-00356

### Steps to reproduce

Go to custom launcher
Insert the UMD
qait 5 seconds
and psp crashes and turns off

### USB Charge

Always

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Always

### Memory Stick Speedup

Disabled

### Inferno Cache

Always

Kitki30, 2023-07-01:
UMD
[image]

JoseAaronLopezGarcia, 2023-07-07:
fixed in latest pre-release

closed 2023-07-07 by JoseAaronLopezGarcia