PSP-Archive/ARK-4 #178: Loadingtimes speedup & bug fix displaying specific folders

svensvond, 2023-07-03:
### New Feature

@JoseAaronLopezGarcia 
Since i cant communicate directly i use this issue to do so.

I took some time and had a look why the loading time was so bad for displaying folders with many iso files in it.
I made some changes and now loading times are blazing fast,  Its 4 Seconds vs 40 Seconds now.

also i speed up root folder loading time, which tool some time because of common::beautifySize

I Also fixed a bug which resulted in ARK displaying specific folders as unreadable files (+ wont display in gamemgr also)
This bug has been introduced with commit https://github.com/PSP-Archive/ARK-4/commit/3d7af53b21d29c0b9b11b4b52b367d45560f0c9a

[image]
[image]

Please have a look at my fork: https://github.com/svensvond/ARK-4/commit/26f7a13203b4f9a533ee48dddde6a841a8289e7b

JoseAaronLopezGarcia, 2023-07-03:
Awesome! Open a PR and we'll review and merge your changes.

JoseAaronLopezGarcia, 2023-07-06:
Check out latest pre-release, it should fix one of the issues with loading times.

JoseAaronLopezGarcia, 2023-07-06:
Latest release now has the improvements.

closed 2023-07-06 by JoseAaronLopezGarcia