PSP-Archive/ARK-4 #183: X and O have same function in VSH menu

Jambul46, 2023-07-06:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.60

### Plugins ( if applicable )

always, ms0:/SEPLUGINS/battWarn/BATTWARN.PRX, on
xmb, ms0:/SEPLUGINS/categorieslite/category_lite.prx, on
xmb, ms0:/SEPLUGINS/CFE/cfe_loader.prx, on
game, ms0:/SEPLUGINS/CFE/cfe_loader.prx, on
game, ms0:/SEPLUGINS/CWCHEAT/CWCHEAT.PRX, on
ps1, ms0:/SEPLUGINS/CWCHEAT/cwcheatpops.prx, on
game, ms0:/SEPLUGINS/MP3PLAY/MP3PLAY.PRX, on
xmb, ms0:/SEPLUGINS/PPS/PPS.PRX, off
always, ms0:/SEPLUGINS/PPS/PPS.PRX, on
xmb, ms0:/SEPLUGINS/WP/WP.PRX, on
xmb, ms0:/SEPLUGINS/XMBIH/XMBIH.PRX, on

### Steps to reproduce

Open VSH menu

X and O have same function

### USB Charge

Always

### Overclock

Homebrew

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-07-06:
This is not a bug, it's intended like that to make it universal (for consoles where circle is accept).
To "go back" you use "select" button instead.

JoseAaronLopezGarcia, 2023-07-06:
Alright we can probably fix it with ease.

JoseAaronLopezGarcia, 2023-07-06:
Done: https://github.com/PSP-Archive/ARK-4/releases/tag/v1688652524

closed 2023-07-06 by JoseAaronLopezGarcia