PSP-Archive/ARK-4 #185: 4.20.60 rev5 custom launcher crash

BjornHo, 2023-07-08:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.60 rev 5

### Steps to reproduce

1. On 4.20.60 rev 3
2. Use the UPDATE option by placing rev5 into PSP/GAME
3. Update
4. Reboot Device
5. Start Custom Launcher
6. Crash/Freeze

See issue #146. For some reason, Custom Launcher is very sensitive and the issue is back.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-07-08:
does this fix it?
https://github.com/PSP-Archive/ARK-4/releases/tag/v1688828050

BjornHo, 2023-07-08:
Yes the issue is fixed!

closed 2023-07-08 by JoseAaronLopezGarcia