PSP-Archive/ARK-4 #186: Some folders are recognized as files, making it impossible to open them

TheSubPlayer, 2023-07-09:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.60 r6

### Homebrew/GAME (if applicable)

Custom Launcher

### Steps to reproduce

Idk how to reproduce, it just happened 

[image]
[image]

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Always

### Memory Stick Speedup

Always

### Inferno Cache

Always

JoseAaronLopezGarcia, 2023-07-10:
Check latest revision, this should be fixed.

TheSubPlayer, 2023-07-10:
Yeah, it's fixed

closed 2023-07-10 by JoseAaronLopezGarcia