PSP-Archive/ARK-4 #187: SUGESTÃO DE MELHORIAS NO ARK4

TheusLG, 2023-07-17:
Dear all, good morning.

I frequently use ARK4, and yesterday, I realized that it would be really interesting if ARK4 had a feature where games could read TEXTURE folders or read game (ini files).

For example, in PPSSPP, there is a MODIFIED game called Efootball that integrates with the TEXTURE folder (a folder located at the root of the PSP), where the game reads new outfits, cleats, stadiums, players, etc.

I believe it would be very cool to bring this innovation to the PSP console.

krazynez, 2023-07-17:
If the game is modified to do that wouldn't/shouldn't it already work that way?

TheusLG, 2023-07-17:
Understood. In the way I'm suggesting, it would only read the texture folder, disregarding the game modification.

JoseAaronLopezGarcia, 2023-07-17:
I have no idea what this feature does or even if it's compatible with real PSP but I will investigat it.

TheusLG, 2023-07-17:
Here is a video containing an example of what I would like for the PSP console.

Note: The configurations and readings are done in the PPSSPP emulator for Android. But the folders used are the same.

JoseAaronLopezGarcia, 2023-07-17:
Aren't texture packs HD replacements? I doubt those would work on real PSP due to hardware limitations.

TheusLG, 2023-07-17:
Não! I did two tests, placing the TEXTURES folder inside the PSP folder, and another test placing the TEXTURES folder at the root of the PSP, but the game didn't read it.

However, inside the TEXTURES folder, there is an .ini file that contains the codes and configurations of the mods. If somehow we could make the game understand that the .ini file is related to the TEXTURES folder, it would already be a huge progress.

If you want, I can share the .ini file.

Regards,

JoseAaronLopezGarcia, 2025-03-16:
Closing as duplicate of: https://github.com/PSP-Archive/ARK-4/issues/496