PSP-Archive/ARK-4 #188: ARK_Full_Installer Issue

FULANO93, 2023-07-17:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.60 Rev 7

### Steps to reproduce

1 - Move ARK_01234 folder into PSP/SAVEDATA folder.
2 - Move ARK_Loader folder into PSP/GAME folder.
3 - Launch ARK Loader.
4 - Run Classic cIPL.
5 - Reboot
6 - Run ARK_Full_Installer

I tried here install twice the ARK_Full_Installer from original 6.61 firmware, it installed, but I think I'm doing something wrong, because in theory it was supposed to install the files in flash0 and then I wouldn't need the Ark_01234 save folder anymore, right? But for me when I remove the save folder it stops having access to what was in it, as if I hadn't actually done the complete installation... I've already read the tutorial on github 5 times and in the text file that comes with it and it doesn't explain well what needs to be done (or I'm doing this wrong) or when to do it install the full version, it just says "Copy `ARK_Full_Installer` to `/PSP/GAME/` and run it from `ARK`." I've tested both, install from normal xmb and from Ark's custom launcher, it always does the same result, it always needs the save Ark_01234 to work some functions... it never stays 100% when I remove the save folde... I'm using the classic cIPL as a permanent patch, since my PSP 2000 is a 2g model... the 'Live' name on version disappears...

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-07-17:
You will still need ARK_01234 for Custom Launcher and Recovery.

FULANO93, 2023-07-17:
Have some way to implement this to run without need the save folder ?

krazynez, 2023-07-17:
No the EBOOT's are far to big to fit in flash.

JoseAaronLopezGarcia, 2023-07-21:
Custom Launcher and Recovery require a memory stick.
They are however `redundant` as you can do the same thing with the `XMB` and using the `start` and `select` keys to load the system in `vanilla` mode (settings and plugins disabled).

closed 2023-07-21 by JoseAaronLopezGarcia