PSP-Archive/ARK-4 #189: DTP-T1000A Fails to launch all ISOs

ProtoFgt, 2023-07-24:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

4.20.60

### Steps to reproduce

I've gone ahead and updated the DTP-T1000A to 6.60 (Kernel 1.00, Userland 5.00 according to the CP Panel), as well as verified that my memory stick is properly configured manually (Copy ARK_Loader to /PSP/GAME, ARK_01234 to /PSP/SAVEDATA).
Launching the Loader definitely works successfully, but launching ISOs after this point does not.
I have verified this exact configuration works correctly with a DTP-H1500A and a PSP-3000 (for functional parity).
No extra plugins have been added or configured, I am using only what is available from the Loader and Save data portions of the ARK4 package.
Went ahead and also verified UMDs work as expected in both OFW and post-ARK Loader scenarios so this is exclusively ISO loading that is not working as expected.

If any additional information is needed let me know.
Thanks.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-07-24:
Added (Testing and Development) them to the bug tracker thanks for noticing that those were missing :+1:

ProtoFgt, 2023-07-24:
Thanks for the response so far.

> Damn, this one is going to be hard to figure out since I don't own any of the devkit models.
> 
> When you say that they fail, do you mean they exit back to XMB with an error code? Or do you mean that they crash the console?
> 
> Have you tried earlier versions of ARK? Support for the Terminator units was added on version 4.18.5 which can be found here: https://github.com/PSP-Archive/ARK-4/releases/tag/r185
> 
> Every other version can be found in the changelog here: https://github.com/PSP-Archive/ARK-4/blob/main/CHANGELOG.md

When I say they fail I mean the PSP attempts to launch them and you get stuck at the gameboot screen (Plays a chime, White screen, PSP logo). Rather than fade to black and launch the game as expected it will just stay there.

I have not checked older versions (or for that matter, newer) but will give a few others a shot and report back. If you have any other ideas to try let me know and I'll give it a swing.

> Added (Testing and Development) them to the bug tracker thanks for noticing that those were missing 👍

In your defense, an issue about these models is definitely an edge case and very unlikely to be present.

ProtoFgt, 2023-07-24:
Reporting back after a quick test.

Tested the same basic setup on 4.18.5, and it did load the ISO this time, so I guess something between 4.18.5 and 4.20.60 is the culprit.

Any additional information you'd need on this?

JoseAaronLopezGarcia, 2023-07-24:
Try a version that isn't so old, 4.19 or 4.20.XX, etc
I can't do any testing myself so I rely on you to more or less pinpoint where it started failing.

ProtoFgt, 2023-07-25:
No worries, going through them now until it doesn't work. I'll try to get the exact revision that broke it. Will have an answer soon.

ProtoFgt, 2023-07-25:
Alright, narrowed it down.
https://github.com/PSP-Archive/ARK-4/releases/tag/r1815 Fails
https://github.com/PSP-Archive/ARK-4/releases/tag/r1814 Works
https://github.com/PSP-Archive/ARK-4/compare/r1814...r1815 For comparison between the commits

Watching the flashing process once starting the loader I noticed a few things that stood out, specifically line quantity so I grabbed video to compare the outputs and the findings are below.

- First line that's different is "Loading ARK-4" at the top on all revisions equal or newer than 4.18.15. Not a problem, just mentioning it.
- "Repairing kernel" is the second line I see on ones equal or newer than 4.18.15. This one is probably more important. Wasn't sure if it's the culprit though.

Since there's 11 builds between those, I went ahead and figured out which the last working one was.
https://github.com/PSP-Archive/ARK-4/releases/tag/v1661815305 Fails (exhibits same behaviors)
https://github.com/PSP-Archive/ARK-4/releases/tag/v1661815202 Works
https://github.com/PSP-Archive/ARK-4/compare/v1661815202...v1661815305 For comparison

There's only 1 change in there between those so that's probably what broke it, but why that would break ISO loading for only a devkit seems unusual.

ProtoFgt, 2023-07-25:
Sorry for the repeated edits, got a little lost along the way, but those last 2 builds I linked are definitely the last working/first failing builds on the devkit.

If you'd like to have me test anything let me know.

JoseAaronLopezGarcia, 2023-07-25:
Fixed.

closed 2023-07-25 by JoseAaronLopezGarcia