PSP-Archive/ARK-4 #190: CFW updater failing due to invalid mountpoint `sf0:/`

jeffangelion, 2023-07-30:
### PSP/Vita Model

PSP-2000 (Slim)

### ARK Version

4.20.61 Rev 2

### Steps to reproduce

1. Run ARK-4 Updater

### USB Charge

Always

### Overclock

Always

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Always

### Inferno Cache

Always

krazynez, 2023-07-30:
"Run ARK-4 Updater" which way ? 
* ARK Menu (Custom Launcher)
* XMB
* UPDATE from zip file
* UPDATE from web server

jeffangelion, 2023-07-30:
Yes, UPDATE from zip file

krazynez, 2023-07-30:
You are already on the latest stable: 4.20.61 Rev 2

are you trying the testing branch?

jeffangelion, 2023-07-30:
I'm at **4.20.60 Rev 7**, provided version is one that has described issue

krazynez, 2023-07-30:
Did you launch it via xmb? If so can you try using Custom Launcher to launch the UPDATE to rule out any oddities.

krazynez, 2023-07-30:
Okay so I downgraded to 4.20.60 r7 then went to 4.20.61 r2 with no issue, not 100% sure why that happened to you.

jeffangelion, 2023-07-30:
Let me be more clear and attach photo
If you cannot reproduce this - please ask more details
BTW it was launched via XMB (Updater extracted from release zip)
[image]

krazynez, 2023-07-30:
Can you try the UPDATE in this one please: https://github.com/PSP-Archive/ARK-4/releases/download/v1690728035/ARK-4-v1690728035.zip

krazynez, 2023-07-30:
Its a test version, but if it works... great, the next release you can still use the stable UPDATE after that and it will work like a normal release.

jeffangelion, 2023-07-30:
Unfortunately, there is the same error

krazynez, 2023-07-30:
Very strange. Is it an actual Memory Stick, or a mircosd card adapter to memory stick?

jeffangelion, 2023-07-30:
Yes, it is microSD->Memory Stick adapter, however I installed all ARK-4 releases from 4.20.59 via this adapter just fine

krazynez, 2023-07-30:
> Yes, it is microSD->Memory Stick adapter, however I installed all ARK-4 releases from 4.20.59 via this adapter just fine

Yeah I assumed so, but was curious. I also have the same thing in mine. very strange its looking for sf0, I have search for everything not seeing any type-o's that we could have done. 

Not really a fix, but you could just extract ARK_01234 to `ms0:/PSP/SAVEDATA` then run `ARK Loader` if you wish to get on the latest version. But this indeed is a very strange bug. I will have to check the updater for a possible overflow which could be happening.

krazynez, 2023-07-30:
Oh! Think I found the bug.

krazynez, 2023-07-30:
try this one: https://github.com/PSP-Archive/ARK-4/releases/download/v1690729991/ARK-4-v1690729991.zip

jeffangelion, 2023-07-30:
Yes, this one correctly uses `ms0:/` however it still fails to extract files
Should I provide ARK_01234 separately as a savedata?

krazynez, 2023-07-30:
> Yes, this one correctly uses `ms0:/` however it still fails to extract files Should I provide ARK_01234 separately as a savedata?

May I please have a picture of the error. 

Do you remove the ARK_01234 every time you update? (You don't have to do that if so)

jeffangelion, 2023-07-30:
> Do you remove the ARK_01234 every time you update? (You don't have to do that if so)
My bad, copied it
Now updater works just fine

krazynez, 2023-07-30:
Yep I just removed mine and got the same error. Well you did find another bug for me so much appreciated anyways  :+1:

jeffangelion, 2023-07-30:
Thanks for a quick solution!

jeffangelion, 2023-07-30:
TL;DR
Regression from 68a7036 (main.c:L60) was fixed

closed 2023-07-30 by jeffangelion