PSP-Archive/ARK-4 #192: Games would not boot and show error 80010016

Phantom1224, 2023-08-02:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.61 Rev 3

### Steps to reproduce

1. Copy and paste any ISO formats or EBOOT games on the corresponding folders.
2. Run the game.
3. The screen will remain black with no buttons interacting with it from the meantime
4. The device will go back to the main menu with  a message "The game could not be started. (80010016)"

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-08-02:
is the ISO's located on `ef0` or `ms0`?

JoseAaronLopezGarcia, 2023-08-02:
Is this only a PSP Go thing? How are you running the ISOs? From XMB or Custom Launcher?

Phantom1224, 2023-08-02:
> is the ISO's located on `ef0` or `ms0`?

It is located on ef0. I forgot to include it on my report.

Phantom1224, 2023-08-02:
> Is this only a PSP Go thing? How are you running the ISOs? From XMB or Custom Launcher?
I tested every ISO formats that you can use (ISO, CSO, CSOv2, JSO, ZSO, and DAX) through both XMB and the custom launcher, none of the games worked and only EBOOT files do.

JoseAaronLopezGarcia, 2023-08-02:
What plugins do you have enabled?

Phantom1224, 2023-08-03:
> What plugins do you have enabled?

I have a few ones but I never thought of checking if it causes the error...damn,I should have put it, too. Editing now

JoseAaronLopezGarcia, 2023-08-03:
Try booting the games while holding start button.

Phantom1224, 2023-08-03:
> Try booting the games while holding start button.

Still gives the same error for me.

4stringsteve, 2023-08-21:
@Phantom1224 were you able to solve the problem? I'm in the same boat as you, fresh install of infinity and ARK and getting hit with the same error.

Phantom1224, 2023-08-21:
> @Phantom1224 were you able to solve the problem? I'm in the same boat as you, fresh install of infinity and ARK and getting hit with the same error. 

Sadly, I never found a solution. I rolled back versions for now.

4stringsteve, 2023-08-21:
@Phantom1224 gotcha, is rolling back as simple as using an older update? Or do I need to start from scratch using Chronoswitch?

Phantom1224, 2023-08-21:
> @Phantom1224 gotcha, is rolling back as simple as using an older update? Or do I need to start from scratch using Chronoswitch? 

You can install an older version like an update install and since you use inifinity then nothing much to worry about using Chronoswitch.

krazynez, 2023-08-21:
@Phantom1224 which one was the last version that worked ?  4.20.60 Rev 7 ?

Phantom1224, 2023-08-21:
> @Phantom1224 which one was the last version that worked ?  4.20.60 Rev 7 ?

Yep

krazynez, 2023-08-24:
Thanks to @4stringsteve for testing last working "test" version is:

https://github.com/PSP-Archive/ARK-4/commit/d9370f339e9f8267dbf1c2ceb46278ecc167d84c

One of the Rev's broke PSP games in : https://github.com/PSP-Archive/ARK-4/releases/tag/r42061

We do not have a git history of commits for the rev's @JoseAaronLopezGarcia

JoseAaronLopezGarcia, 2023-08-28:
Fixed. https://github.com/PSP-Archive/ARK-4/releases/tag/v1693209261

closed 2023-08-28 by JoseAaronLopezGarcia

MateusBMP, 2023-09-24:
In the version 4.20.64 the same error has returned, @JoseAaronLopezGarcia

krazynez, 2023-09-24:
> In the version 4.20.64 the same error has returned, @JoseAaronLopezGarcia

Try this one: https://github.com/PSP-Archive/ARK-4/releases/download/v1695515761/ARK-4-v1695515761.zip

MateusBMP, 2023-09-24:
Yes! I'm literally found this commit now and will be returned here to say and test kkkk
I'll test this early as possible

krazynez, 2023-09-25:
> Yes! I'm literally found this commit now and will be returned here to say and test kkkk I'll test this a soon as possible

please do once you verify it does (because it should :wink: ) we can make a rev. Thanks for the follow up on this as well :+1:

MateusBMP, 2023-09-25:
I test the "Option 2: Copy the UPDATE folder to /PSP/GAME/ and run it." and it's works! You can make a rev. Thanks!