PSP-Archive/ARK-4 #199: Disable psx shortcut menu

fusccaa, 2023-08-10:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.61 r3

### Homebrew/GAME (if applicable)

All

### Plugins ( if applicable )

always, ms0:/SEPLUGINS/PspPowerSaver.prx, on

always, ms0:/SEPLUGINS/CUSHOME/CUSHOME.PRX, on

always, ms0:/SEPLUGINS/660_plugins_on_661.prx, on

always, ms0:/SEPLUGINS/dayviewer.prx, on

### Steps to reproduce

1. Open any game
2. Press R, L, X, down
3. and then a menu with ps1 games will open
4. the problem is that these commands are used a lot mainly in PES 2013, so whenever I press these buttons while playing, this menu opens with PS1 games making me lose my progress in the game

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

fusccaa, 2023-08-10:
[image]

JoseAaronLopezGarcia, 2023-08-10:
Sorry it seems that a debug code managed to slip into final release. We'll get rid of it.

fusccaa, 2023-08-10:
> Sorry it seems that a debug code managed to slip into final release. We'll get rid of it.

that's bad, because I can't play football or any other game that I accidentally press that combination of buttons... maybe I have to go back to version 6.61 of the system.

JoseAaronLopezGarcia, 2023-08-10:
> > Sorry it seems that a debug code managed to slip into final release. We'll get rid of it.
> 
> that's bad, because I can't play football or any other game that I accidentally press that combination of buttons... I'm waiting for the update... Tanks

[FLASH0.zip](https://github.com/PSP-Archive/ARK-4/files/12312543/FLASH0.zip)
Use this build, extract FLASH0.ARK to /PSP/SAVEDATA/ARK_01234/ and run ARK Loader again to flash new files.
Let me know if that fixes the issue.

krazynez, 2023-08-10:
Yeah we can get rid of it and do a rev, that is my fault. I did not see any game that had that combo for buttons so thought it was safe. Thanks for making the issue to get our attention and sorry for you impact.

fusccaa, 2023-08-10:
> [FLASH0.zip](https://github.com/PSP-Archive/ARK-4/files/12312543/FLASH0.zip) Use this build, extract FLASH0.ARK to /PSP/SAVEDATA/ARK_01234/ and run ARK Loader again to flash new files. Let me know [...]

thank you so much for getting back to me so quickly, this solved my problem.  And by the way this cfw is amazing

krazynez, 2023-08-10:
Yeah he just removed the code that would trigger it. But if your having problems others might too so just doing a rev people can just upgrade to is more user friendly then them replacing the FLASH0.ARK and rerunning ARK Loader.

fusccaa, 2023-08-10:
> Yeah he just removed the code that would trigger it. But if your having problems others might too so just doing a rev people can just upgrade to is more user friendly then them replacing the FLASH0.ARK and rerunning ARK Loader.

Yes, probably more users have the same problem, a new REV is the right thing to do, thank you very much.

JoseAaronLopezGarcia, 2023-08-11:
Fixed on latest release.

closed 2023-08-11 by JoseAaronLopezGarcia