PSP-Archive/ARK-4 #202: Crash logs

Jambul46, 2023-08-11:
### New Feature

Would it be possible to add toggleable crash logs?
Or any kind of debug? My PSP sometimes randomly crashes and it's been doing that for long (even with PRO-C) i tried to completly reinstall it but it didnt help.

In short: Recovery Menu > Log Toggle on/off > log.txt

Something like kernel log in normal linux.

Jambul46, 2023-08-11:
Also it doesnt crash in games only when doing random stuff in xmb (and now also launcher)

krazynez, 2023-08-12:
If you use the debug versions it has very verbose output on screen.Might be hard to getbresults for random crashes though.

Jambul46, 2023-08-12:
Thats why i said that it would be nice to just go into recovery/settings and be able to enable some sort of logging

JoseAaronLopezGarcia, 2023-08-12:
We have debug builds and then there's also PSPLink debugger out there.

Jambul46, 2023-08-12:
PSPLink is no go because i would need to be always connected to pc and like i said it crashes randomly (sometimes just once in a month) and for the debug builds do they log it somewhere? Because when it crashes it completly freezes and then turns off after 10-15 second

JoseAaronLopezGarcia, 2023-08-12:
> PSPLink is no go because i would need to be always connected to pc and like i said it crashes randomly (sometimes just once in a month) and for the debug builds do they log it somewhere? Because when it crashes it completly freezes and then turns off after 10-15 second

Debug builds have a built-in Blue Screen Of Death that shows important information on the screen that you can then take pictures of. it can be useful.

Jambul46, 2023-08-12:
I'll try that then. I thought it just had like an overlay that would show "kernel" log

Jambul46, 2023-08-12:
Do i just run the update pbp from the debug release?

JoseAaronLopezGarcia, 2023-08-12:
> Do i just run the update pbp from the debug release?

Yes.

Jambul46, 2023-08-13:
So after installing the debug version i tried to play P3P it showed it was loading some modules got stuck at sceSAScore and then poof it turned off no BSOD. Any idea why that happend?

Jambul46, 2023-08-14:
Works fine with normal version

krazynez, 2023-08-21:
DEBUG builds do use more ram so it can cause different issues its more for seeing what it is getting stuck on. Usually you want to pair the DEBUG builds with PSPLink to poke around and see. If an execution happens the BSoD will show, however if the system just halts that does not get triggered before hand sometimes sadly.  

Being random would make it very hard for any logging to happen as we would not know what to exactly look for. Having a file that is constantly being written to would not work on the PSP or you would just "lock" up because its just writing to a file the whole time. Threading is a thing yes, but its not really that great to do with io calls from my experience.

We can leave this open if @JoseAaronLopezGarcia can think of something or not. I will also try to brainstorm something, but off the top of my head I can not think of something that would fill the void.

closed 2023-09-09 by JoseAaronLopezGarcia