PSP-Archive/ARK-4 #203: Cannot use micro SD card to completely load ARK Loader

antNguyen1, 2023-08-12:
### PSP/Vita Model

PSP-1000 (Fat)

### ARK Version

ARK 4.20.63

### Steps to reproduce

First, I just want to say thank you for your hard work for providing this custom firmware. It is amazing how people revitalize a system released more than 10 years ago. So, I have a PSP 1000 that had 6.61 ME Infinity on my Sandisk 64GB micro-SD card. It was working fine until I noticed that a new custom firmware that could replace PRO C/ME. I was really interested in the features it provided and I decided to attempt to install it. 

I put my micro-SD card into my PC installed the files while reading the README and launched the ARK Loader on my PSP. Everything looked like it was installing the modules correctly after launching the software. When the PSP booted back up, I could not access the VHS menu and did not see the extra options in the settings of the XMB. 

I then decided to go back to the original firmware using Chronoswitch. Like an idiot I was, I renamed the EBOOT file of Chronoswitch to 661.PBP (I was reading Bing answers because I forgot how to use Chronoswitch. I don' think renaming the EBOOT file should've messed with the micro-SD card). I then put the micro-SD card back into the PSP. But then, the PSP wanted to format the card. My PC also wanted to format the card because it could not read it for some reason. I ran the error checking software provided by Windows and it cannot access the card without formatting it.

Since I kind of already had a backup of my micro-SD card, I decided to format the card on the PSP. I repeated the steps to install ARK-4 and launched ARK Loader again. Still, I could not access the VHS menu and did not see the extra options in the settings of the XMB. I had a 1GB Sony memory card and again and repeated the steps to install ARK-4 and launched ARK Loader. This time it actually worked! I could access the VHS menu and see the extra options in the settings. 

I tried repeating the same steps on another micro-SD card and that also still produced the same problem. So, the only way I can get ARK-4 on my PSP is to use the 1GB Sony memory card. If there is any way to solve this, I would really appreciate it. Thank you.

Step-by-step on how to reproduce the bug/issue:
1. Get a micro-SD card with the memory stick adaptor and move or copy "ARK_01234" folder into `/PSP/SAVEDATA/` folder. 
2. Move or copy "ARK_Loader" folder into `/PSP/GAME/` folder.
3. Launch "ARK Loader". It will "install" ARK modules on PSP Flash and boot the CFW.
4. Press the "SELECT" button to try to access the VSH menu. It does not appear.
5. Go to the "Settings" in the XMB to look for the extra configuration options provided by ARK-4. It does not appear.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

antNguyen1, 2023-08-14:
I have tried using the Full Installer on the micro SD card. Unfortunately, that doesn’t work either. But, I think you are right about the adapter being the source of the problem since both micro SD cards that I am using are functional. I’ll see if a can get another adapter later on during the week. Thank you for the suggestion.

JoseAaronLopezGarcia, 2023-08-14:
Avoid using those dual SD card adapters, they are known to malfunction. Use single SD card adapters.

antNguyen1, 2023-08-15:
I got a new adapter and the micro Sd card works! I have access to the VHS menu and the custom firmware options in the settings through the XMB. For anyone wondering what adaptor did not work for me, it is the Funturbo adapter from Amazon. The new adapter I got is the adapter from WQDMKE on Amazon. Thank you for making Ark-4 and thank you for the assistance!

closed 2023-08-15 by antNguyen1