PSP-Archive/ARK-4 #205: Unable to run standalone PS Vita ARK using both FastARK and NoPspEmuDrm methods

RandomGuy2023, 2023-08-15:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

ARK 4.20.63

### Steps to reproduce

Right after running ARK after installing it with the FastARK method, it often crashes with the C1-2858-3 error. According to ARK's latest changelogs, this issue had been fixed but I still have it. Also, ARK's GitHub's main page mentions if ARK crashes, to just close it and reopen it so has this issue really been fixed or not?

Right after running ARK after installing it with the NoPspEmuDrm method on the PS TV, it often freezes showing an image saying "#ChovyProject". Shouldn't this image only appear if running the standalone PS Vita ARK installed via ChovySign method? When this ARK freezes, I have to take the PS TV's power cable since it doesn't let me close the app and if pressing the PS TV's power button nothing happens. 

Also, both ARK have a bug showing savedata since some don't even show up and the ones that do, show up called "ARK_01234", and when I run them, I'm directed to the Recovery Menu settings.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

JoseAaronLopezGarcia, 2023-08-15:
The eboot used for NoPspEmuDrm was created using ChovySign (which was made by Silika, the same dev that made NoPspEmuDrm) so it's natural that the #ChovyProject splash screen shows up.

I haven't used the FastARK method for a while, and the latest version works just fine for me, but maybe there's something else, perhaps there's an ISO or EBOOT that the custom launcher doesn't like and causes it to crash (this happens to me on my 3K and it didn't happen before so it's gotta be a new bug). Try using the recovery menu's file browser to boot stuff and see if that works.

JoseAaronLopezGarcia, 2023-08-15:
I have no idea why NoPspEmuDrm would have that behavior. Which Vita firmware version are you using? ARK is technically compatible with 3.60, 3.65 and all the way up to 3.74, however I only have 3.60 on my vita so I can't be 100% sure how well it works on later firmwares.

JoseAaronLopezGarcia, 2023-08-15:
What do you mean by "Savedata issue"? If you mean the "Scan savedata entries", that feature has nothing to do with viewing savedata files, it scans for ISOs and eboots contained inside savedata folders, which was a classic method of injecting PSP homebrew on the Vita via CMA back in the old days when we didn't have native hacks and VitaShell that let us access the Vita's memory stick directly. It should be left disabled as it's largely useless today, but I didn't want to bother removing the code so I just left it as a feature that is disabled by default.

RandomGuy2023, 2023-08-15:
> What do you mean by "Savedata issue"? If you mean the "Scan savedata entries", that feature has nothing to do with viewing savedata files, it scans for ISOs and eboots contained inside savedata folders, which was a classic method of injecting PSP games and homebrew on the Vita via CMA back in the old days when we didn't have native hacks and VitaShell that let us access the Vita's memory stick directly. It should be left disabled as it's largely useless today, but I didn't want to bother removing the code so I just left it as a feature that is disabled by default.

In that case, it's  ot possible for ARK to show actual savedata from games? I made a confusion with them. So, the savedata which shows up is actually DLC inside savedata from games in the form of EBOOT?

RandomGuy2023, 2023-08-15:
> No there's no way to view the savedata other than the File Browser, the Vita already has its own savedata manager so I considered this feature unnecessary.

So, the savedata which shows up is actually DLC in the form of EBOOT inside savedata folders from games?

RandomGuy2023, 2023-08-15:
> No there's no way to view the savedata other than the File Browser, the Vita already has its own savedata manager so I considered this feature unnecessary.

So, the savedata which shows up after enabling the "Scan savedata entries" setting is actually DLC in the form of EBOOT inside savedata folders from games?

RandomGuy2023, 2023-08-15:
> No there's no way to view the savedata other than the File Browser, the Vita already has its own savedata manager so I considered this feature unnecessary.

ArkFast has an icon with a picture from ARK-2, not ARK-4, and is the program called ArkFast or FastARK like mentioned on ARK's GitHub main page?

Can PS Vita's ARK run PS1 games or not? On ARK's GitHub's main it's mentioned it can't but on ARK's latest version GitHub page it's mentioned it can.

Regarding the standalone PS Vita ARK version install via FastARK method, what does ARk's GitHub's main page mean by "if out of date with latest release, use the updater built into the launcher"? Is it talking about the update icon that appears at the top of screen after clicking a LiveArea app before running the app itself (ArkFast, in this case)? I clicked it but nothing happened and so it didn't install an update to match the latest ARK version.

JoseAaronLopezGarcia, 2023-08-15:
> 
> ArkFast has an icon with a picture from ARK-2, not ARK-4, and is the program called ArkFast or FastARK like mentioned on ARK's GitHub main page?
> 

That's because ArkFast was originally made for ARK-2 but it was easy to repack it with ARK-4, you must be using an older version though since we made a fork named FastARK-4 that has a new icon. As for the name, I'm not sure if it's really called ArkFast or FastArk, I've seen both and I've used both, they both refer to the same program anyways.

> 
> Can PS Vita's ARK run PS1 games or not? On ARK's GitHub's main it's mentioned it can't but on ARK's latest version GitHub page it's mentioned it can.
> 

Support for PS1 games in Standalone installations was added back just two releases ago (in 4.20.62), which is why some parts of the readme might be obsolete (I'll check them out).
There is a catch though, the only way to play PS1 games in Standalone installations is by using the original PS1 emulator for PSP and a custom port of the Peops SPU plugin, which means that audio compatibility is a hit and miss, it works for the most part and some games are perfect (i.e Tekken 3 and Bushido Blade) but some games have audio issues (i.e. Sheep Raider when near water), slowdows (i.e. GTA 2 when there's too much traffic on screen), no audio at all (i.e. Metal Gear Solid) or even crashes (this is rare so I can't give an example).

A proper solution for Standalone PS1 using the Official PS1 Emulator by Sony is in the works but we haven't fully figured it out yet so I can't give an ETA.

When using Adrenaline-ARK, PS1 games work fine.

> Regarding the standalone PS Vita ARK version install via FastARK method, what does ARk's GitHub's main page mean by "if out of date with latest release, use the updater built into the launcher"? Is [...]

In ARK's Custom Launcher (the main menu for Standalone installs), press triangle and enter "Network", you will be able to check for updates there and update ARK. You can't update ARK using the Vita's own updater as ARK isn't in any Vita Homebrew Database.

> 
> EDIT: I tried the standalone PS Vita ARK install via FastARK method again and ARK didn't crash until I deleted ArkFast and restarted the PS Vita so does that mean I have to keep ArkFast? Also, may I remove Ape Quest after installing ARK? I guess it doesn't even need to be installed unlike the Sasuke vs. Commanded minigame installed by FastArk.
> 

It might be possible that FastARK is still needed, I'm not sure since I haven't deleted it in my setup.

RandomGuy2023, 2023-08-15:
> That's because ArkFast was originally made for ARK-2 but it was easy to repack it with ARK-4, you must be using an older version though since we made a fork named FastARK-4 that has a new icon. As [...]

I just followed ARK' GitHub's main page's men [mentioned](https://github.com/Yoti/ArkFast-4/releases/tag/42056) link. It doesn't have a ARK-4 icon but an ARK-2 one. I think it should only have one name to not confuse users.

I thought the launcher was related to FastArk but after all it's ARK's Custom Launcher the GitHub page is talking about.

And what about Ape Quest, is it needed?

JoseAaronLopezGarcia, 2023-08-15:
You only need Ape Quest if you want the resulting ARK bubble to be bootable in Official Firmware, if you only run Henkaku then it's not needed.

RandomGuy2023, 2023-08-15:
> You only need Ape Quest if you want the resulting ARK bubble to be bootable in Official Firmware, if you only run Henkaku then it's not needed.

Ok, but I previously read the user shouldn't install a minigame but a game from PSN because the former wouldn't make ARK work and now after all one should install ArkFast's Sasuke vs. Commander minigame?

JoseAaronLopezGarcia, 2023-08-15:
Information changes when new evidence arrives.

RandomGuy2023, 2023-08-15:
> Information changes when new evidence arrives.

What if the user has HENkaku but already a PSN game installed? May it use it instesd of Ape Quest or the Sasuke vs. Commander minigame?

So, am I using ArkFast's right version or not (the one that has ARK-2's icon)? I still don't get the "If out of date with latest release, use the updater built into the launcher.". Why would the user need to update ARK providing it already downloaded and installed the latest ARK version or is that for users which have an older ARK version? Also, the latest ArkFast version (4.20.56) doesn't match the latest ARK version (4.20.63 Rev 1) so is it ok to use it?

JoseAaronLopezGarcia, 2023-08-15:
When you install the Sasuke vs Commander minigame, FastArk-4 (the latest version from Yoti's repo) will attempt to copy a valid PSP license if you have a PSN game installed, this will make the bubble bootable in Official Firmware, otherwise it will use a Fake license that is only bootable with Henkaku.

RandomGuy2023, 2023-08-15:
> When you install the Sasuke vs Commander minigame, FastArk-4 (the latest version from Yoti's repo) will attempt to copy a valid PSP license if you have a PSN game installed, this will make the [...]

I use that version and it has a ARK-2 icon, not a ARK-4 one. Is that fake license related to the Sasuke vs. Commander game?

It's confusing to have 3 methods to install standalone PS Vita ARK (instead of only two, one for hacked PS Vita and one for unhacked PS Vita) as well as three methods to update PSP ARK on the Readme (instead of only two).

Why is a new version called "Rev" often released instead of an entirely new one?

Can one run OFW on the PSP after doing a full flash install or only after doing a permanent install like mentioned in ARK's GitHub's main page?

If pressing the L, R, triangle, square, cross and circle buttons and all the D-pad buttons at once when playing PSP ISOs, the PSP screen goes black stating "Loading..." and the PSP returns to XMB. This is very annoying since I hold the PSP like that for a while when playing a game.

How can one do a full flash PSP ARK install? Does the user have to do a permanent ARK install with one of the two cIPL or Infinity like I've already done (I did the latter), copy "ARK_Full_Installer" to "/PSP/GAME/" and run it from ARK or can one do the latter two steps right after doing a manual install? May the ARK_01234 folder be deleted regardless of the used method to install ARK on the PSP and if so will the user lose XMB Custom Firmware settings as well as Custom Launcher's and so will have to set them again?

What's cIPL and pandora and how can I know which model is my PSP 3000 and if it's compatible with pandora? If I have a PSP 1000/2000 compatible with both new and old cIPL, which method should I follow and if I have a PSP compatible with both cIPL and Infinity, which method show I follow? Why are there three different permanent CFW methods on the PSP?

Why is there this new Classic Recovery Menu on the PSPs with full flash install which is used when there is no "Recovery app" available? Does this mean if one has a Memory Stick inserted but no ARK_01234 savedata folder will it still be able to access this Classic Recovery Menu on PSPs with full flash install? And is it possible to have the ARK_01234 savedata folder while having no Memory Stick inserted and thus having this Classic Recovery Menu on PSPs with full flash install?

JoseAaronLopezGarcia, 2023-08-15:
- I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
- We will eventually simplify installation on PS Vita.
- Revisions are small changes that don't deserve their own release.
- Are you joking or trolling? How the hell do you press all Dpads at once? And why the hell are you holding all buttons down? Be serious or I will be closing this thread very soon.

The rest of the questions are already answered on the readme or online.

RandomGuy2023, 2023-08-15:
> * I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
> * We will eventually simplify installation on PS Vita. [...]

I always delete FastArk before installing the new one.

I'm not joking otherwise I wouldn't mention this problem, don't you think? That problem didn't happen with the 4.20.60 version and I do that only in a certain game by using both thumbs and index fingers (that way I can press all D-pad buttons and the L, R, triangle, square, cross and circle buttons at once).

No, they're not, hence I'm asking. You even admit there's contradictory information because some of it is updated on the changelog but not in the main page.

RandomGuy2023, 2023-08-16:
> * I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
> * We will eventually simplify installation on PS Vita. [...]

I always delete FastArk before installing the new one.

I'm not joking otherwise I wouldn't mention this problem, don't you think? That problem didn't happen with the 4.20.60 version and I do that only in a certain game by using both thumbs and index fingers (that way I can press all D-pad buttons and the L, R, triangle, square, cross and circle buttons at once).

No, they're not (at least in the readme), hence I'm asking. You even admit there's contradictory information because some of it is present in the changelog and not in the GitHub's main page because you forgot to update it in one place instead of updating it in both and it regards issues previous ARK versions had. What do you mean by online information?

I can't update ARK to a Rev version on the PSP using the "Update via Internet" XMB setting since it doesn't detect the latest version so I always have to use the method regarding copying "UPDATE" folder and running ARK UPDATE because I also enabled Custom Launcher's Force update setting but Custom Launcher also doesn't update ARK. What do you mean by clicking "Network" in ARK's Custom Launcher? That setting has to do with the FTP server, not with updating ARK.

RandomGuy2023, 2023-08-16:
> * I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
> * We will eventually simplify installation on PS Vita. [...]

Open
Unable to run standalone PS Vita ARK using both FastARK and NoPspEmuDrm methods
#205
RandomGuy2023 opened this issue 8 hours ago · 19 comments
Labels
bug
Comments
RandomGuy2023 commented 8 hours ago • 
PSP/Vita Model
PCH-2000 (Vita Slim)

ARK Version
ARK 4.20.63

Homebrew/GAME (if applicable)
No response

Plugins ( if applicable )
No response

Steps to reproduce
Right after running ARK after installing it with the FastARK method, it often crashes with the C1-2858-3 error. According to ARK's latest changelogs, this issue had been fixed but I still have it. Also, ARK's GitHub's main page mentions if ARK crashes, to just close it and reopen it so has this issue really been fixed or not?

Right after running ARK after installing it with the NoPspEmuDrm method on the PS TV, it often freezes showing an image saying "#ChovyProject". Shouldn't this image only appear if running the standalone PS Vita ARK installed via ChovySign method? When this ARK freezes, I have to take the PS TV's power cable since it doesn't let me close the app and if pressing the PS TV's power button nothing happens.

Also, ARK has a bug showing savedata since some don't even show up, and the ones that do, show up called "ARK_01234", and when I run them, I'm directed to the Recovery Menu settings.

EDIT: I tried the standalone PS Vita ARK install via FastARK method again and ARK didn't crash until I deleted ArkFast and restarted the PS Vita so does that mean I have to keep ArkFast? Also, may I remove Ape Quest after installing ARK? I guess it doesn't even need to be installed unlike the Sasuke vs. Commander minigame installed by FastArk.

USB Charge
Disabled

Overclock
Disabled

Power Save
Disabled

Autoboot Launcher
Disabled

Disable PSP Go Pause
Disabled

Unlock Extra Memory
Disabled

Memory Stick Speedup
Disabled

Inferno Cache
Disabled

@RandomGuy2023 RandomGuy2023 added the bug label 8 hours ago
JoseAaronLopezGarcia commented 8 hours ago
The eboot used for NoPspEmuDrm was created using ChovySign (which was made by Silika, the same dev that made NoPspEmuDrm) so it's natural that the #ChovyProject splash screen shows up.

I haven't used the FastARK method for a while, and the latest version works just fine for me, but maybe there's something else, perhaps there's an ISO or EBOOT that the custom launcher doesn't like and causes it to crash (this happens to me on my 3K and it didn't happen before so it's gotta be a new bug). Try using the recovery menu's file browser to boot stuff and see if that works.

RandomGuy2023 commented 7 hours ago • 
The eboot used for NoPspEmuDrm was created using ChovySign (which was made by Silika, the same dev that made NoPspEmuDrm) so it's natural that the #ChovyProject splash screen shows up.

I haven't used the FastARK method for a while, and the latest version works just fine for me, but maybe there's something else, perhaps there's an ISO or EBOOT that the custom launcher doesn't like and causes it to crash (this happens to me on my 3K and it didn't happen before so it's gotta be a new bug). Try using the recovery menu's file browser to boot stuff and see if that works.

I have the C1-2858-3 issue right after running the standalone PS Vita ARK installed with the FastARK method, i.e., before even running a PSP ISO or EBOOT, regardless from the Recovery menu's file browser or not.

What about the standalone PS Vita ARK installed with the NoPspEmuDrm method freezing on the PS TV showing the mentioned image and ARK's savedata bug?
[... 125 lines cut ...]
What's cIPL and pandora and how can I know which model is my PSP 3000 and if it's compatible with pandora? If I have a PSP 1000/2000 compatible with both new and old cIPL, which method should I follow and if I have a PSP compatible with both cIPL and Infinity, which method show I follow? Why are there three different permanent CFW methods on the PSP?

Why is there this new Classic Recovery Menu on the PSPs with full flash install which is used when there is no "Recovery app" available? Does this mean if the user has a Memory Stick inserted but no ARK_01234 savedata folder will it still be able to access this Classic Recovery Menu on PSPs with full flash install? And is it possible to have the ARK_01234 savedata folder while having no Memory Stick inserted and thus having this Classic Recovery Menu on PSPs with full flash install?

JoseAaronLopezGarcia commented 2 hours ago
I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
We will eventually simplify installation on PS Vita.
Revisions are small changes that don't deserve their own release.
Are you joking or trolling? How the hell do you press all Dpads at once? And why the hell are you holding all buttons down? Be serious or I will be closing this thread very soon.
The rest of the questions are already answered on the readme or online.

RandomGuy2023 commented 32 minutes ago • 
I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
We will eventually simplify installation on PS Vita.
Revisions are small changes that don't deserve their own release.
[... cut ...] wered online? Where can I find them?

I can't update ARK to the latest version (4.20.63 Rev 1) on the PSP using the "Update via Internet" XMB setting since it doesn't detect it so I have to use the method regarding copying "UPDATE" folder and running ARK UPDATE because I also enabled Custom Launcher's Force update setting but Custom Launcher also doesn't update ARK. What do you mean by clicking "Network" in ARK's Custom Launcher? That setting has to do with the FTP server, not with updating ARK.

RandomGuy2023, 2023-08-16:
> * I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
> * We will eventually simplify installation on PS Vita. [...]

I always delete FastArk before installing the new one.

I'm not joking otherwise I wouldn't mention this problem, don't you think? That problem didn't happen with the 4.20.60 version and I do that only in a certain game by using both thumbs and index fingers (that way I can press all D-pad buttons and the L, R, triangle, square, cross and circle buttons at once).

No, they're not (at least in the readme), hence I'm asking. You even admit there's contradictory information because some of it is present in the changelog and not in the GitHub's main page because you forgot to update it in one place instead of updating it in both and it regards issues previous ARK versions had. What do you mean by questions already answered online? Where can I find them?

I can't update ARK to the latest version (4.20.63 Rev 1) on the PSP using the "Update via Internet" XMB setting since it doesn't detect it so I have to use the method regarding copying "UPDATE" folder and running ARK UPDATE because I also enabled Custom Launcher's Force update setting but Custom Launcher also doesn't update ARK. What do you mean by clicking "Network" in ARK's Custom Launcher? That setting has to do with the FTP server, not with updating ARK.

RandomGuy2023, 2023-08-16:
> * I'm 100% sure the FastARK-4 from Yoti's repo does NOT have an ARK-2 Icon. Make sure you delete the old FastARK before installing the new one.
> * We will eventually simplify installation on PS Vita. [...]

I always delete FastArk before installing the new one.

I'm not joking otherwise I wouldn't mention this problem, don't you think? That problem didn't happen with the 4.20.60 version and I do that only in a certain game by using both thumbs and index fingers (that way I can press all D-pad buttons and the L, R, triangle, square, cross and circle buttons at once).

No, they're not (at least in the readme), hence I'm asking. You even admit there's contradictory information because some of it is present in the changelog and not in the GitHub's main page because you forgot to update it in one place instead of updating it in both and it regards issues previous ARK versions had. What do you mean by questions already answered online? Where can I find the answers?

I can't update ARK to the latest version (4.20.63 Rev 1) on the PSP using the "Update via Internet" XMB setting since it doesn't detect it so I have to use the method regarding copying "UPDATE" folder and running ARK UPDATE because I also enabled Custom Launcher's Force update setting but Custom Launcher also doesn't update ARK. What do you mean by clicking "Network" in ARK's Custom Launcher? That setting has to do with the FTP server, not with updating ARK.

ARK's GitHub's main page regarding the PSP forgot to state to run ARK Loader after running Infinity the second time.

On the PSP, ARK Loader folder may be deleted as well as Infinity and ARK_Full_Installer but if the ARK_01234 folder is deleted and one clicks the Custom Launcher, the PSP restarts, so it's better to keep the ARK_01234 folder. Also, if deleting it anyway and clicking on Recovery Menu, a different menu called Classic Recovery Menu appears instead of the normal Recovery Menu.

JoseAaronLopezGarcia, 2023-08-16:
- I don't know if deleting FastArk is needed or not.
- But why do this? This could break your PSP's buttons. You also need to understand that the CFW and some plugins have button combinations so you might be triggering them. For example, running the game while holding Start will disable plugins, holding Select disables settings, holding L+R+Start+Down will exit from the game back into Custom Launcher. Just don't do this, you risk physically damaging the console!
- Information about pandora batteries is all over the internet.
- There was an issue with the updater that would make it fail in the XMB, this was fixed a few days ago.
- Ok I'll check the readme and add missing details.
- Normal Recovery Menu and Custom Launcher can't be installed to flash0 because they are too big, so instead it will install Classic Recovery Menu, which contains PRO Shell that can be used as a Launcher. There is no way around this, there simply is no free space in flash0 for such big files.

RandomGuy2023, 2023-08-16:
> * I don't know if deleting FastArk is needed or not.
> * But why do this? This could break your PSP's buttons. You also need to understand that the CFW and some plugins have button combinations so you [...]

How can it break the PSP's buttons? But this issue wasn't happening until recently.

But I still have that issue and I have to manually update ARK.

What about the information regarding cIPL? 

What do you mean by clicking "Network" in ARK's Custom Launcher? That setting has to do with the FTP server, not with updating ARK.

It's confusing to have 3 methods to install standalone PS Vita ARK (instead of only two, one for hacked PS Vita and one for unhacked PS Vita) as well as three methods to update PSP ARK on the Readme (instead of only two).

Why are there three different permanent CFW methods on the PSP?

Could you add the information the full flash ARK install step on the PSP must be done after doing one of the three permanent ARK install methods as well as the Live CFW step must be done after the manual installation one (one of the permanent ARK install methods must be done after doing the Live CFW one, then) to make things less confusing for users when installing ARK? 

Regarding standalone PS Vita ARK via FastARK, could you refer to it by a name and not two (FastARK and ArkFast) and change the information that says it requires Henkaku/h-encore or any native hack because afterwards you say it will work on OFW or so you think users will remove HENkaku and install OFW? Also, the way it's written now, it seems users need both a PSP game from PSN and the Sasuke vs. Commander minigame. Could you say the former is needed for OFW users and the latter for HENkaku ones to not confuse users?

Also, regarding PS Vita ARK via ChovySign, you say it doesn't require Henkaku/h-encore or any native hack but then you say one might need HENkaku/h-encore to facilitate the installation process but that it won't be needed anymore after installing. This part also confuses users.

JoseAaronLopezGarcia, 2023-08-16:
- Things break from usage, specially abusive usage, what you describe is extremely harmful, specially for the dpad.
- If you have already updated you won't see updates cause we haven't released any.
- "Network" has FTP and Updater options.
- I already answered about the 3 different standalone instructions.
- Different features and compatibility.
- Ok I'll revise the readme.
- I think this part is already well explained.
- What's confusing about this? The method doesn't require henkaku, but using it facilitates the procedure. What's not to understand here?

RandomGuy2023, 2023-08-16:
> * Things break from usage, specially abusive usage, what you describe is extremely harmful, specially for the dpad.
> * If you have already updated you won't see updates cause we haven't released any. [...]

But I was on 4.20.63 and it didn't detect the 4.20.63 Rev 1 version.

What about the information regarding cIPL? Is it found on the internet? If a PSP supports all 3 permanent installs, which one should I use?

I can't find the updater setting in "Network". Which button should I press to use it?

The latest ArkFast version (4.20.56) doesn't match the latest ARK version (4.20.63 Rev 1) so is it OK to use it?

So, ARK's Custom Launcher's "Scan savedata entries" setting reveals DLC in the form of EBOOT inside savedata folders from games?

JoseAaronLopezGarcia, 2023-08-16:
- The server is probably not updated, we're on vacation you know.
- It's been over 15 years since cIPL became a thing on PSP and you can't find any information about it online? Seriously? Do you know how to use Google? I'm starting to think you're trolling me and this discussion is gonna end NOW.
- Can't you read? It clearly says in really big text "Press [] to check for updates".
- Can't you read? The readme clearly states "if ArkFast's version of ARK is out of date, use ARK's built-in update function".
- DO NOT ENABLE THAT FEATURE IF YOU DO NOT UNDERSTAND IT. I HAVE ALREADY VERY CLEARLY STATED THAT THE FEATURE IS NOT NEEDED ANYMORE, PLEASE READ WHAT I WRITE.

JoseAaronLopezGarcia, 2023-08-16:
I'm also pretty certain by the way you talk that you're a dupe account of Windows10User, the biggest noob I've ever had the displeasure of having to reply to, completely incapable of understanding the most basic things.

closed 2023-08-16 by JoseAaronLopezGarcia

RandomGuy2023, 2023-08-16:
> I'm also pretty certain by the way you talk that you're a dupe account of Windows10User, the biggest noob I've ever had the displeasure of having to reply to, completely incapable of understanding the most basic things.

And I'm pretty sure you're an idiot who doesn't even know how to write in english and then it' the other user's fault for not understanding. What you wrote in GitHub is so confusing and you even admitted you have contradictory informarion and then you blame me.

RandomGuy2023, 2023-08-24:
> * The server is probably not updated, we're on vacation you know.
> * It's been over 15 years since cIPL became a thing on PSP and you can't find any information about it online? Seriously? Do you [...]

If it depends on users being in vacation, I don't understand why it exists and why did you say the issue has been fixed. I guess I'll always update it manually, then. 

Ok, I found that setting on Custom Launcher's Network setting, but what does that you're talking about have to do with ArkFast since the Custom Launcher updater (supposedly) installs the latest ARK version and not the latest ArkFast version, and even if it did install the latest ArkFast version and not the latest ARK one, how would it work since I delete ArkFast right after using it to install ARK, and even if I didn't, ArkFast is used to install ARK and since the latter is already installed, why would I need to install ArkFast's latest version and run it? Also, you didn't write that way in GitHub since it isn't understandable if you're talking about FastArk or ARK and what launcher you're talking about, hence why I'm confused.

Who said I was going to leave it enabled? I'm only asking about it since I'm curious. Anyway, If it's not needed why is there in the first place?

EDIT: I tried PSP's ARK's Custom Launcher's updater and it actually downgraded the ARK 4.20.63 Rev 1 version to the previous one (4.20.63).

EDIT 2: Now I'm having the C1-2858-3 error right after installing the standalone ARK with the FastARK method so it doesn't matter if ArkFast is kept or not.

EDIT 3: I ran the standalone PS Vita ARK installed via NoPspEmuDrm method again on the PS TV, it froze after running a PSP ISO and after some time ARK crashed with the C1-2858-3 error.

EDIT 4: Apparently, I no longer have the issue in which if pressing the L, R, triangle, square, cross and circle buttons and all the D-pad buttons at once when playing PSP ISOs the PSP screen went black stating "Loading..." and the PSP returned to XMB.

RandomGuy2023, 2023-08-24:
> I'm also pretty certain by the way you talk that you're a dupe account of Windows10User, the biggest noob I've ever had the displeasure of having to reply to, completely incapable of understanding the most basic things.

And I'm pretty sure you're an idiot who doesn't even know how to write in english and then it's my fault for not understanding. What you wrote in GitHub is so confusing that other users also asked what should they do in GBATemp regarding installing ARK as well as the differences regarding cIPL and Infinity and you didn't say them to search in Google, to read ARK's GitHub page or accused them of being blind so I'm not the only one which is confused. You even admitted you have contradictory information and you'll revise the Readme and then you blame me. Also, other users asked you in GBATemp as well if they should install a PSP game from the PSN when using the FastArk method or the latter's minigame.

Is it my fault you don't update GitHub's main page but GitHub's ARK changelog regarding playing PS1 games on standalone PS Vita ARK and standalone PS Vita ARK crashes? 

Is it my fault I'm having the XMB return issue when until recently I wasn't and I never had it with Pro-C? 

Is it my fault you told the update issue was fixed and that another user was insulted (maybe that one you accuse me of being) for installing a minigame with FastArk and then you latter say on GitHub that the minigame actually needs to be installed? 

Is it my fault you refer to ArkFast as FastArk and ArkFast as if they were different programs? 

Is it my fault for PSP's ARK to crash when you don't even know if the user must keep ArkFast? 

Is it my fault ArkFast has an icon with a picture from ARK-2 and not ARK-4 (which you didn't even knew) and that on ARK's GitHub's main page regarding the PSP you forgot to say to run ARK Loader after running Infinity the second time? 

Is it my fault you release standalone PS Vita ARK versions that crash after being run?

You even admitted you didn't use the FastArk for some time.