PSP-Archive/ARK-4 #207: Dissidia 012 crash

JaySomyl, 2023-08-16:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.63 r1

### Homebrew/GAME (if applicable)

Dissidia 012 (US Version)

### Steps to reproduce

1. start game from xmb
2. white screen appears and power indicator led stays on
3. after a few seconds the psp shuts down completely

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-08-16:
Eboot or ISO?

JaySomyl, 2023-08-16:
tried both, but the issue happens either way for me

JaySomyl, 2023-08-21:
md5 is 64d7b19cd3c627bd81e60a2e6fb46549 which should be the correct one as far as i can tell

JoseAaronLopezGarcia, 2023-08-28:
Try using npdrm_free.

krazynez, 2023-09-01:
> Try using npdrm_free.

Tested with same issue (just putting here for documentation purpose as we already discussed this on Discord).

JoseAaronLopezGarcia, 2023-09-12:
Is this still a thing in latest version?

krazynez, 2023-09-13:
Okay so I just tested again with `nploader` and am not having the issues anymore @JaySomyl. Try that and let us know if it resolves your issues as well :+1:

JaySomyl, 2023-09-13:
issue is resolved now thank you very much!

krazynez, 2023-09-13:
> issue is resolved now thank you very much!

Awesome to hear! Thank you for the update.

closed 2023-09-13 by krazynez