PSP-Archive/ARK-4 #210: Soft/hard brick

sticktail, 2023-08-22:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.63 Rev 1

### Homebrew/GAME (if applicable)

N/A

### Steps to reproduce

1. Follow the steps for installation on psp
2. Follow non permanent CFW install
3. Follow steps in video to run or install files on psp to run. https://www.youtube.com/watch?v=bE8mw7RRqGo&t=41s
4. Delete files put in the psp (I didn't get the program to run or didn't know how). 
5. Deleted the EBOOT.PBP from PSP > GAME > UPDATE file 
6. Reinstall/COPY the EBOOT.PBP file form ARK4 > UPDATE file folder
7. Now the psp will no longer do a normal quick boot up sequence. It bricks up or I have the press or hold the home button while it is turning on now for the psp to load up on it's own.  

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

sticktail, 2023-08-22:
I never got to boot up the program besides loading the infinity portion on the psp and that's about it for me. Now I have a psp that I need to hold down the home button just so it can run and turn on normally now.

krazynez, 2023-08-22:
Sounds like you are trying to run Infinity to get permanent CFW. If so try this: https://github.com/PSP-Archive/ARK-4/wiki/Infinity

sticktail, 2023-09-01:
Just went through the process and it helped but what I did notice is that I had the system still in ark4 on infinity and I clicked off on that and the * on the ark 4 went away and the psp booted back up to normal which was awesome so I have the infinity program installed to the latest version and don't have any of the other programs selected on the infinity app so no the psp runs perfectly fine now!

sticktail, 2023-09-01:
Just FYI to everyone. I got cold feet and worried when I went through the process that I did on the youtube video I watched and it bricked my psp. My psp was only able to load up because I was holding down and pressing randomly on the home button. I uninstalled everything and only loaded back up the infinity portion. I deselected the Ark4 on infinity because that's what I forgot to do when "resetting" my psp back to normal. The only difference is that I now I have infinity on there but that's it. I enjoy having a fast-loading psp every time I turn it on.

Also: out of fear I bought a region 3 psp 3006 which showed up over in one of my game stores in missouri and as soon as I booted it up it had everything I needed for a modded psp to change region to region which was super cool. Now my only issue is I can't seem to get that psp to play the movies and TV shows I owned when sony had a on-demand selling tv shows and movies. The system says that I need to "activate the system to run the program".

JoseAaronLopezGarcia, 2023-09-09:
Is this issue solved?

JoseAaronLopezGarcia, 2023-09-12:
Closing does to lack of activity.

closed 2023-09-12 by JoseAaronLopezGarcia