PSP-Archive/ARK-4 #211: PSP Go Settings Icons

nallwolf, 2023-08-23:
### PSP/Vita Model

PSP-N1000 (PSP Go)

### ARK Version

4.20.63 r1

### Plugins ( if applicable )

xmb, ef0:/seplugins/660_plugins_on_661.prx, off
xmb, ef0:/seplugins/category_lite.prx, off
xmb, ef0:/seplugins/dayviewer.prx, off
ps1, ef0:/seplugins/POPBIOS.PRX, off
ps1, ef0:/seplugins/popsloader.prx, off
xmb, ef0:/seplugins/UVC.PRX, off
game, ef0:/seplugins/UVC.PRX, off

### Steps to reproduce

1. Activate ARK-4
2. Last 2 XMB icons in Settings category on PSP Go are now incorrect
[image]
(Picture shows plugins active, but it happens with them disabled as well)

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Disabled

krazynez, 2023-08-24:
Bluetooth Device Settings is causing the one off I believe, as the other models do not have that setting. This probably also is and issue on the e1000 as well because network settings.

nallwolf, 2023-08-24:
The E1000 does not have any incorrect icons, I've only seen this on the Go and it's more like it's repeating the top 2 icons once it hits RSS Settings rather than being one off

krazynez, 2023-10-06:
New setting to fix the icons: https://github.com/PSP-Archive/ARK-4/releases/tag/v1696627609

JoseAaronLopezGarcia, 2023-10-08:
I'm gonna close this one for now, but we should continue to research using the Extras section instead of Game, that would open up a whole new world of opportunities.

closed 2023-10-08 by JoseAaronLopezGarcia