PSP-Archive/ARK-4 #216: cwcheat, cheat not showing up

McFly75, 2023-08-27:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.63 Rev 1

### Steps to reproduce

Have cwcheat installed and enabled
Start a PSP game (I used God of War -Chains of Olympus)
Hold [select] to bring up the cwcheat menu
Press select cheats

Now for me this is blank (only black boxes) when using Ark-4
but when I switch to Pro-C2 the cheats shows up just fine.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Autoboot Launcher

Disabled

### Disable PSP Go Pause

Disabled

### Unlock Extra Memory

Homebrew

### Memory Stick Speedup

Always

### Inferno Cache

Always

krazynez, 2023-08-27:
This could be your db not being updated. I just tried it and it was working fine on my 1k. I don't have the latest available  cheatdb to my knowledge (nor do I know where it is located). I tested out "one hit kills" and "Infinite Health" they worked fine, but when I tried "Big Heads" it locked up on my PSP.

McFly75, 2023-08-27:
I use the db from:
https://github.com/Saramagrean/CWCheat-Database-Plus-

And it works fine on Pro-C2 CFW on both my PSP 3000 & PSP Go.

But on Ark-4 the menu open fine, so the plugin is working, but it does not show the cheats like on Pro-C2.

krazynez, 2023-08-27:
I see I get the same result with that database.

krazynez, 2023-08-27:
If I switch to TempAR it does populate the database correctly however. Interesting...

McFly75, 2023-08-27:
OK, I haven´s tried the TempAR yet.

Normally I don´t use cheats, but I just found about the 60 fps cheats.

I just played Need for Speed: Most Wanted which normally run in 30 fps, but with clock set to 333/166 and the 60 fps cheat, it ran very smooth on my PSP Go.

McFly75, 2023-08-27:
Yes, I can now confirm that it seems that with cwcheat, it does not populate the database correctly.

And it works fine when using TempAR. (on my PSP 3000)

However I am a big fan of cwcheat, especially now when testing 60 fps cheats, because cwcheat has a fps counter that can be enabled.

krazynez, 2023-08-29:
Ahhhh okay so that database is massive and for some reason there seems to me not enough space for it, however if you remove all the cheats you are not using and replace it with the games you desire they do show up properly. I just tested it with GoW: CoO.

krazynez, 2023-09-08:
refer to : https://github.com/PSP-Archive/ARK-4/issues/153

JoseAaronLopezGarcia, 2023-09-08:
> OK, but it still seems to be a problem with Ark-4, because Pro-C2 does not have this issue.

Can you send me a link of the cwcheat version you're using? I want to reproduce the issue myself.

McFly75, 2023-09-08:
https://github.com/Saramagrean/CWCheat-Database-Plus-

JoseAaronLopezGarcia, 2023-09-08:
I can confirm this issue exists, we're looking into it.

JoseAaronLopezGarcia, 2023-09-08:
Latest pre-release should fix this.

JoseAaronLopezGarcia, 2023-09-08:
Fixed in latest release.

closed 2023-09-08 by JoseAaronLopezGarcia