PSP-Archive/ARK-4 #217: ARK doesn't appear in LiveArea after using FastARK to install the Sasuke vs. Commander minigame

Goldtuga, 2023-08-30:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.63 Rev 2

### Steps to reproduce

Run FastARK-4.
Install the Sasuke vs. Commander minigame.
Return to LiveArea.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

krazynez, 2023-08-30:
Reboot once it finishes the install

krazynez, 2023-08-30:
You can get a PSP demo for a licenese but okay I will have to take a look again with a fake license thanks for the bug report 👍

Goldtuga, 2023-08-30:
> You can get a PSP demo for a licenese but okay I will have to take a look again with a fake license thanks for the bug report 👍

Isn't that for non-HENkaku or other hack users?

krazynez, 2023-08-30:
work with both henkaku and non-henkaku, not sure about the folders sounds like a de-sync.

Goldtuga, 2023-08-30:
> work with both henkaku and non-henkaku, not sure about the folders sounds like a de-sync.

I thought it was only intended for non-HENkaku PS Vita so is it recommended to install a PSP demo or FastARK's minigame? 

I already de-synced and re-synced and the folders still don't appear in FileZilla.

ARK's GitHub instructions are confusing, hence I don't know if installing a PSP demo before running FastARK or installing the latter's minigame is required for HENkaku and non-HENkaku users, not to mention regarding the PSP it states to run "ARK" after running Infinity the second time and not "ARK Loader" and it doesn't have the information the full flash ARK install step on the PSP must be done after doing one of the three permanent ARK install methods as well as the Live CFW step must be done after the manual installation one. Could you addres this misinformation? 

What do you mean by "limitations with PS1 games"?

Also, I ran the standalone PS Vita ARK installed via NoPspEmuDrm method on the PS TV, updated ARK via the Custom Launcher and ARK often freezes after running a PSP ISO and after some time it crashes with the C1-2858-3 error.

krazynez, 2023-08-30:
> I thought it was only intended for non-HENkaku PS Vita so is it recommended to install a PSP demo or FastARK's minigame?
>
> I already de-synced and re-synced and the folders still don't appear in [...]

If you use a real license you can use FastARK-4 with or without henkaku.

It is recommended to use the Wiki for installation instructions now.

PS1 games with standalone's audio is hit or miss with games.

NoPspEmuDrm right now is mainly stable on 3.60 people have installed it on higher firmware but have had issues as well. I have let the devs know some issues we are having with it so hopefully soon we can have better support on later firmware.

Goldtuga, 2023-08-30:
> If you use a real license you can use FastARK-4 with or without henkaku.
>
> It is recommended to use the Wiki for installation instructions now.
>
> PS1 games with standalone's audio is hit or miss with [...]

The information is confusing because it looks HENkaku users must install a PSP demo and then FastARK's minigame. Where is the Wiki?

What is standalone audio?

But I'm using 3.60.

krazynez, 2023-08-30:
> The information is confusing because it looks HENkaku users must install a PSP demo and then FastARK's minigame. Where is the Wiki?
>
> What is standalone audio?
>
> But I'm using 3.60.

Top of readme also tab in github. Its just recommended to do not required.

sorry forgot a comma. ARK's Standalone versions have PSX audio issues depending on games.

That error from my limited research is by not having the nonpdrm plugin could be causing that.

Goldtuga, 2023-08-30:
> Top of readme also tab in github. Its just recommended to do not required.
>
> sorry forgot a comma. ARK's Standalone versions have PSX audio issues depending on games.
>
> That error from my limited [...]

Could you update the GitHub's main page, then? So, according to the Wiki, a PSN PSP demo or game is needed for OFW users and Sasuke vs. Commander is needed for HENkaku ones? If HENkaku users already have a PSN PSP demo or game installed, is Sasuke vs. Commander still needed?

So, users which don't have the nonpdrm plugin installed have issues with PS1 games or it's actually that plugin that causes the issues?

krazynez, 2023-08-30:
> Could you update the GitHub's main page, then? So, according to the Wiki, a PSN PSP demo or game is needed for OFW users and Sasuke vs. Commander is needed for HENkaku ones? If HENkaku users already [...]

Sasuke vs. Commander is still needed (FastARK installs it) how ARK gets injected. PSP Game/Demo is only used for the RIF file. I will most likely be removing all install stuff from the Readme as we are focused on the Wiki.

No nonpdrm is for PSP/PSN games not PSX.

Goldtuga, 2023-08-30:
> Sasuke vs. Commander is still needed (FastARK installs it) how ARK gets injected. PSP Game/Demo is only used for the RIF file. I will most likely be removing all install stuff from the Readme as we [...]

ARK's GitHub's main page WIKI doesn't state one can get a kernel error after running the standalone PS Vita ARK, states ARK Loader must be run after running Infinity the second time on the PSP, doesn't state the FLASH0.ARK file must be deleted on the PSP, doesn't even mention the full flash installation step on the PSP and doesn't mention the PS1 games compatibility so could you at least have the same information in both places?

krazynez, 2023-08-30:
That is not a kernel error. FLASH0.ARK? We never delete that. The Wiki is not fully finilzed I am still adding things but all installation are there for users  to get started which is where I wanted to get first. This is starting to move off topic. If you would like to discuss more general areas of concerns or questions please join the[ PSP Homebrew Discord](https://discord.gg/bePrj9W) in the `#ark-cfw` channel

Goldtuga, 2023-08-30:
> That is not a kernel error. FLASH0.ARK? We never delete that. The Wiki is not fully finilzed I am still adding things but all installation are there for users to get started which is where I wanted [...]

It doesn't matter if it's a kernel error or not. What matters is that the main page states the user running PS Vita's standalone ARK might have a kernel error (like you can see [here](https://github.com/PSP-Archive/ARK-4#standalone-via-fastark, [here](https://github.com/PSP-Archive/ARK-4#standalone-via-nopspemudrm and [here](https://github.com/PSP-Archive/ARK-4#standalone-via-chovysign))) and the WIKI doesn't. Then why is it stated [here](https://github.com/PSP-Archive/ARK-4#live-cfw) to delete that file? I thought the WIKI was more correct than the main page. I don't have Discord nor plan to.

krazynez, 2023-08-30:
That is not the C1-2858-3 error that is the K.BIN sometimes fails to trigger.

Goldtuga, 2023-08-30:
> That is not the C1-2858-3 error that is the K.BIN sometimes fails to trigger.
> 
> Also all these post are spamming the discord server webhook is why I stated its better to just post your concerns/comments there that way people don't have to mute the projects channel to watch updates for all the projects.

OK, but the mainpage states one can have a kernel error and the WIKI doesn't and what about what I stated regarding FLASH0.ARK and the informaton in the WIKI (and the main page) being confusing?

Sorry, I don't understand the second paragraph since I don't know how Discord works. Every time one makes a post here, the users gets notifications in ARK's Discord channel? If so, every time one makes a post in the Discord channel won't the users get a notification to? What projects are you talking about?

krazynez, 2023-08-30:
I will take your opinions into consideration this issue has gone off the rails.

closed 2023-08-30 by krazynez