PSP-Archive/ARK-4 #218: PS Vita standaline ARK crashes with kernel error

Goldtuga, 2023-08-31:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.63 Rev 2

### Steps to reproduce

After running ARK installed via FastARK or NoPspEmuDrm, it often crashes in the boot or afterwards when running PSP ISOs.

I'm on HENkaku 3.60.

[image]

[image]

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

krazynez, 2023-08-31:
Is your vita activated?

Goldtuga, 2023-08-31:
> Is your vita activated?

Unlike what you stated in the other issue, it is indeed a kernel error like one can see in the pictures. What do you mean? I tried running ARK after having installed a PSP PSN demo and FastARK's Sasuke vs. Commander minigame. In my case, unlike what you stated in the other issue, FastARK doesn't install the Sasuke vs. Commander minigame if a PSN PSP demo/game is already installed.

krazynez, 2023-08-31:
FastARK-4 uses Sasuke vs. Commander as its base game a PSP demo/Game is only used to copy its RIF (license file) if wanted. 

You can try copying the K.BIN from PSVita/Standaline from the latest stable ARK release to `ux0:pspemu/PSP/SAVEDATA/ARK_01234/` possibly its a bad K.BIN if so I can reupload it.

Goldtuga, 2023-08-31:
> FastARK-4 uses Sasuke vs. Commander as its base game a PSP demo/Game is only used to copy its RIF (license file) if wanted.
> 
> You can try copying the K.BIN from PSVita/Standalone from the latest stable ARK release to `ux0:pspemu/PSP/SAVEDATA/ARK_01234/` possibly its a bad K.BIN if so I can reupload it.

What do you mean by "if wanted"? I already did that and it still crashed with the same error.

krazynez, 2023-08-31:
its failing on K.BIN. You don't have to worry about the RIF part its already installed as you are getting tobthe exoloit sections. So something with your K.BIN is wrong. If your using the NoPspEmuDrm method make sure you copied the PSVita/Standalone/K.BIN to the path mentioned in my last message. FastARK-4 already prepacks the correct K.BIN file

Goldtuga, 2023-08-31:
> its failing on K.BIN. You don't have to worry about the RIF part its already installed as you are getting to the exploit sections. So something with your K.BIN is wrong. If your using the NoPspEmuDrm method make sure you copied the PSVita/Standalone/K.BIN to the path mentioned in my last message. FastARK-4 already prepacks the correct K.BIN file

So, installing a PSN PSP demo/game isn't needed if installing the Sasuke va. Commander minigame? 

Already did and it still crashes with the mentioned issue.

krazynez, 2023-08-31:
> So, installing a PSN PSP demo/game isn't needed if installing the Sasuke va. Commander minigame?

Sasuke va. Commander comes with FastARK-4 its the base game that FastARK-4 installs but injects ARK-4 to run. 

> 
> Already did and it still crashes with the mentioned issue.

Then I am not 100% sure what the issue is.

Goldtuga, 2023-08-31:
> > > its failing on K.BIN. You don't have to worry about the RIF part its already installed as you are getting to the exploit sections. So something with your K.BIN is wrong. If your using the NoPspEmuDrm method make sure you copied the PSVita/Standalone/K.BIN to the path mentioned in my last message. FastARK-4 already prepacks the correct K.BIN file
> > 
> > 
> > So, installing a PSN PSP demo/game isn't needed if installing the Sasuke va. Commander minigame?
> 
> Sasuke va. Commander comes with FastARK-4 its the base game that FastARK-4 installs but injects ARK-4 to run.
> 
> > Already did and it still crashes with the mentioned issue.
> 
> Then I am not 100% sure what the issue is. I just redid FastARK-4 a few days ago with no issues. I can't use NoPspEmuDrm because I am on a higher firmware for testing purposes.

I know, but is a PSN PSP demo/game needed besides the Sasuke vs. Commanded minigame that comes with FastARK to make ARK work? Yes or no? If yes, why?

And so what? Both the GitHub main page and the WIKI state it runs from 3.60 OFW up to 3.74.

krazynez, 2023-08-31:
> I know, but is a PSN PSP demo/game needed besides the Sasuke vs. Commanded minigame that comes with FastARK to make ARK work? Yes or no? If yes, why?

No only if your using OFW you would have to.

> 
> And so what? Both the GitHub main page and the WIKI state it runs from 3.60 OFW up to 3.74.

Yes, we are aware its hit or miss it does work but not always we are fine tuning it NoPspEmuDrm is the newest method for Standaline.

Goldtuga, 2023-08-31:
> > > > > its failing on K.BIN. You don't have to worry about the RIF part its already installed as you are getting to the exploit sections. So something with your K.BIN is wrong. If your using the NoPspEmuDrm method make sure you copied the PSVita/Standalone/K.BIN to the path mentioned in my last message. FastARK-4 already prepacks the correct K.BIN file
> > > > 
> > > > 
> > > > So, installing a PSN PSP demo/game isn't needed if installing the Sasuke va. Commander minigame?
> > > 
> > > 
> > > Sasuke va. Commander comes with FastARK-4 its the base game that FastARK-4 installs but injects ARK-4 to run.
> > > > Already did and it still crashes with the mentioned issue.
> > > 
> > > 
> > > Then I am not 100% sure what the issue is. I just redid FastARK-4 a few days ago with no issues. I can't use NoPspEmuDrm because I am on a higher firmware for testing purposes.
> > 
> > 
> > I know, but is a PSN PSP demo/game needed besides the Sasuke vs. Commanded minigame that comes with FastARK to make ARK work? Yes or no? If yes, why?
> 
> No only if your using OFW you would have to.
> 
> > And so what? Both the GitHub main page and the WIKI state it runs from 3.60 OFW up to 3.74.
> 
> Yes, we are aware its hit or miss it does work but not always we are fine tuning it NoPspEmuDrm is the newest method for Standalone.

But even so, it still doesn't show up in LiveArea.

Ok, so should HENkaku users follow the NoPspEmuDrm method or the FastARK one? Should OFW users follow the ChovySign method?

krazynez, 2023-08-31:
> But even so, it still doesn't show up in LiveArea.
You can try recovery mode and rebuild the db.
> 
> Ok, so should HENkaku users follow the NoPspEmuDrm method or the FastARK one? Should OFW users follow the ChovySign method?

HENkaku users can use all 3 methods unless your vita is not activated however you will have to use the NoPspEmuDrm or ChovySign as FastARK-4 will not work. Was just informed about that earlier. For OFW users with no henkaku/h-encore your best bet is FastARK-4 or ChovySign.

Goldtuga, 2023-08-31:
> No only if your using OFW you would have to.
>
>
> Yes, we are aware its hit or miss it does work but not always we are fine tuning it NoPspEmuDrm is the newest method for Standalone.

Then don't state you can't use NoPspEmuDrm because you're on a higher firmware than 3.60 or instead state on ARK's GitHub main page and ARK's GitHub WIKI that it doesn't work for users on a higher firmware then 3.60.

> HENkaku users can use all 3 methods unless your vita is not activated however you will have to use the NoPspEmuDrm or ChovySign as FastARK-4 will not work. Was just informed about that earlier. For [...]

What do you mean by activated? Are you talking about PSN activation which can be manually made by the user or automatically by downloading and installing a game/demo from the PSN? What if the user didn't do one of these things, should he/she install a PSN PSP game/demo before running FastARK or it isn't required because if installing FastARK's Sasuke vs. Commander minigame it automatically activates the PS Vita even if not signed in to a PSN account?

Who informed you, me or other users? But like I stated, after running the standalone PS Vita ARK installed via NoPspEmuDrm on the PS Vita, ARK often freezes after running a PSP ISO and after some time it crashes with the C1-2858-3 error. But I thought ChovySign was for OFW users.

krazynez, 2023-08-31:
> > > > > > its failing on K.BIN. You don't have to worry about the RIF part its already installed as you are getting to the exploit sections. So something with your K.BIN is wrong. If your using the NoPspEmuDrm method make sure you copied the PSVita/Standalone/K.BIN to the path mentioned in my last message. FastARK-4 already prepacks the correct K.BIN file
> > > > > 
> > > > > 
> > > > > So, installing a PSN PSP demo/game isn't needed if installing the Sasuke va. Commander minigame?
> > > > 
> > > > 
> > > > Sasuke va. Commander comes with FastARK-4 its the base game that FastARK-4 installs but injects ARK-4 to run.
> > > > > Already did and it still crashes with the mentioned issue.
> > > > 
> > > > 
> > > > Then I am not 100% sure what the issue is. I just redid FastARK-4 a few days ago with no issues. I can't use NoPspEmuDrm because I am on a higher firmware for testing purposes.
> > > 
> > > 
> > > I know, but is a PSN PSP demo/game needed besides the Sasuke vs. Commanded minigame that comes with FastARK to make ARK work? Yes or no? If yes, why?
> > 
> > 
> > No only if your using OFW you would have to.
> > > And so what? Both the GitHub main page and the WIKI state it runs from 3.60 OFW up to 3.74.
> > 
> > 
> > Yes, we are aware its hit or miss it does work but not always we are fine tuning it NoPspEmuDrm is the newest method for Standalone.
> 
> Then don't state you can't use NoPspEmuDrm because you're on a higher firmware than 3.60 or instead state on ARK's GitHub main page and ARK's GitHub WIKI that it doesn't work for users on a higher firmware then 3.60.
> 
> > > > > > > > its failing on K.BIN. You don't have to worry about the RIF part its already installed as you are getting to the exploit sections. So something with your K.BIN is wrong. If your using the NoPspEmuDrm method make sure you copied the PSVita/Standalone/K.BIN to the path mentioned in my last message. FastARK-4 already prepacks the correct K.BIN file
> > > > > > > 
> > > > > > > 
> > > > > > > So, installing a PSN PSP demo/game isn't needed if installing the Sasuke va. Commander minigame?
> > > > > > 
> > > > > > 
> > > > > > Sasuke va. Commander comes with FastARK-4 its the base game that FastARK-4 installs but injects ARK-4 to run.
> > > > > > > Already did and it still crashes with the mentioned issue.
> > > > > > 
> > > > > > 
> > > > > > Then I am not 100% sure what the issue is. I just redid FastARK-4 a few days ago with no issues. I can't use NoPspEmuDrm because I am on a higher firmware for testing purposes.
> > > > > 
> > > > > 
> > > > > I know, but is a PSN PSP demo/game needed besides the Sasuke vs. Commanded minigame that comes with FastARK to make ARK work? Yes or no? If yes, why?
> > > > 
> > > > 
> > > > No only if your using OFW you would have to.
> > > > > And so what? Both the GitHub main page and the WIKI state it runs from 3.60 OFW up to 3.74.
> > > > 
> > > > 
> > > > Yes, we are aware its hit or miss it does work but not always we are fine tuning it NoPspEmuDrm is the newest method for Standalone.
> > > 
> > > 
> > > But even so, it still doesn't show up in LiveArea.
> > > You can try recovery mode and rebuild the db.
> > > Ok, so should HENkaku users follow the NoPspEmuDrm method or the FastARK one? Should OFW users follow the ChovySign method?
> > 
> > 
> > HENkaku users can use all 3 methods unless your vita is not activated however you will have to use the NoPspEmuDrm or ChovySign as FastARK-4 will not work. Was just informed about that earlier. For OFW users with no henkaku/h-encore your best bet is FastARK-4 or ChovySign.
> 
> What do you mean by activated? Are you talking about PSN activation which can be manually made by the user or automatically by downloading and installing a game/demo from the PSN? What if the user didn't do one of these things, should he/she install a PSN PSP game/demo before running FastARK or it isn't required because if installing FastARK's Sasuke vs. Commander minigame it automatically activates the PS Vita even if not signed in to a PSN account? Also, FastARK's minigame is called Sasuke vs. Commander and not Sasuke Commander like ARK's GitHub main page and ARK's GitHub's WIKI state.
> 

Yes PSN activation so if you download and start a PSP Demo it will just run, if your Vita is not activated it will throw an error saying you need to activate you vita first. **FastARK-4 needs your vita to be activated** it does not work if it is not, NoPspEmuDrm will work however if your Vita is not activated. 

> Who informed you, me or other users? But like I stated, after running the standalone PS Vita ARK installed via NoPspEmuDrm on the PS Vita, ARK often freezes after running a PSP ISO and after some time it crashes with the C1-2858-3 error. But I thought ChovySign was for OFW users. So, FastARK works for OFW users but not for HENkaku ones?

I don't see how/why I was informed is relavant, but was just another dev and I were discussing it. FastARK WORKS ON BOTH HENkaku/h-encore and OFW.

Goldtuga, 2023-08-31:
> Yes PSN activation so if you download and start a PSP Demo it will just run, if your Vita is not activated it will throw an error saying you need to activate you vita first. **FastARK-4 needs your [...]

I asked who informed you because I wanted to know if I was the only one having this problem.

krazynez, 2023-09-01:
Okay just a update I updated the VitaShell link as I thought the fork that added the PSM and PSP to "refresh livearea" was pushed upstream. So that now is fine. I just tested on 3.65 and its working fine on my end so other users with higher firmware should be fine also. 

As for your issue can you please do these steps:
* Delete the ARK bubble on the LiveArea and the `NoPspEmuDrm*.prx` files in `ur0:tai` 
* open the `ur0:tai/config.txt` and remove the  `NoPspEmuDrm*.prx` as well if they are there to act as a clean install
*  redo [these](https://github.com/PSP-Archive/ARK-4/wiki/NoPspEmuDrm) steps for NoPspDrmEmu.

Goldtuga, 2023-09-01:
> Okay just a update I updated the VitaShell link as I thought the fork that added the PSM and PSP to "refresh livearea" was pushed upstream. So that now is fine. I just tested on 3.65 and its working fine on my end so other users with higher firmware should be fine also.
> 
> As for your issue can you please do these steps:
> 
> * Delete the ARK bubble on the LiveArea and the `NoPspEmuDrm.*prx` files in `ur0:tai`
> * open the `ur0:tai/config.txt` and remove the  `NoPspEmuDrm.*prx` as well if they are there to act as a clean install
> * save `config.txt` and reboot
> * redo [these](https://github.com/PSP-Archive/ARK-4/wiki/NoPspEmuDrm) steps for NoPspDrmEmu.

I uninstalled ARK some days ago so I'll have to follow the NoPspEmuDrm method from the beginning again. I'm on 3.60 HENkaku. Did you have the C1-2858-3 error when running ARK and/or when running a PSP ISO?

krazynez, 2023-09-01:
> I uninstalled ARK some days ago so I'll have to follow the NoPspEmuDrm method from the beginning again. I'm on 3.60 HENkaku. Did you have the C1-2858-3 error when running ARK and/or when running a [...]

originally I had that error when booting the ARK bubble. I have tested a lot of games with it and still no error and boots fine.

Goldtuga, 2023-09-01:
> originally I had that error when booting the ARK bubble. I have tested a lot of games with it and still no error and boots fine.

Will you fix the issue regarding FastARK not appearing in LiveArea after being installed?

krazynez, 2023-09-01:
> Will you fix the issue regarding FastARK not appearing in LiveArea after being installed?

That is out of scope for this issue. But I did indeed test that today with both a real and fake license and it worked fine.

Goldtuga, 2023-09-01:
> That is out of scope for this issue. But I did indeed test that today with both a real and fake license and it worked fine.

Since I still have it should I create an issue regarding it?

krazynez, 2023-09-01:
> Since I still have it should I create an issue regarding it?

please do also screenshots are welcome (very useful for us) apperiate adding them to this one.

Goldtuga, 2023-09-01:
> please do also screenshots are welcome (very useful for us) apperiate adding them to this one.

I don't think it's needed because it's as simple as I told, i.e., after running FastARK to install ARK, the latter's icon doesn't appear righf after FastARK's..

krazynez, 2023-09-01:
> > > > > > > > Okay just a update I updated the VitaShell link as I thought the fork that added the PSM and PSP to "refresh livearea" was pushed upstream. So that now is fine. I just tested on 3.65 and its working fine on my end so other users with higher firmware should be fine also.
> > > > > > > > As for your issue can you please do these steps:
> > > > > > > > 
> > > > > > > > * Delete the ARK bubble on the LiveArea and the `NoPspEmuDrm.*prx` files in `ur0:tai`
> > > > > > > > * open the `ur0:tai/config.txt` and remove the  `NoPspEmuDrm.*prx` as well if they are there to act as a clean install
> > > > > > > > * save `config.txt` and reboot
> > > > > > > > * redo [these](https://github.com/PSP-Archive/ARK-4/wiki/NoPspEmuDrm) steps for NoPspDrmEmu.
> > > > > > > 
> > > > > > > 
> > > > > > > I uninstalled ARK some days ago so I'll have to follow the NoPspEmuDrm method from the beginning again. I'm on 3.60 HENkaku. Did you have the C1-2858-3 error when running ARK and/or when running a PSP ISO?
> > > > > > 
> > > > > > 
> > > > > > originally I had that error when booting the ARK bubble. I have tested a lot of games with it and still no error and boots fine.
> > > > > 
> > > > > 
> > > > > Will you fix the issue regarding FastARK not appearing in LiveArea after being installed?
> > > > 
> > > > 
> > > > That is out of scope for this issue. But I did indeed test that today with both a real and fake license and it worked fine.
> > > 
> > > 
> > > Since I still have it should I create an issue regarding it?
> > 
> > 
> > please do also screenshots are welcome (very useful for us) apperiate adding them to this one.
> 
> I don't think it's needed because it's as simple as I told, i.e., after running FastARK to install ARK, the latter's icon doesn't appear right after FastARK's..

Yes please do so its not mixing Standalone versions. I already tagged this one for NoPspEmuDrm, there is another label for FastARK and they do not work the same way with install instructions.

Goldtuga, 2023-09-03:
I reinstalled the standalone PS Vita ARK via NoPspEmuDrm and ARK didn't froze after running a PSP ISO and didn't crash with the C1-2858-3 error.

closed 2023-09-03 by Goldtuga