PSP-Archive/ARK-4 #219: CXMB themes not working

BlackPyxl1, 2023-08-31:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.63 Rev 2

### Plugins ( if applicable )

xmb, ms0:/seplugins/CXMB/CXMB.PRX, on

### Steps to reproduce

1. install cxmb normally
2. put a ctf theme in themes folder
3. load it

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

VSH

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

On

### Hide DLC

Off

### Turn off LEDs

Disabled

krazynez, 2023-08-31:
Can you provide the name of the Custom Theme please so we can test and see if it is just the theme. For instance, I am using the "PS3 Theme" on my PSP (and Vita) with no issues. I have also replicated your CFW settings and CXMB is still working.

BlackPyxl1, 2023-08-31:
> Can you provide the name of the Custom Theme please so we can test and see if it is just the theme. For instance, I am using the "PS3 Theme" on my PSP (and Vita) with no issues. I have also replicated your CFW settings and CXMB is still working.

https://www.pspunk.com/files/theme/ctf/loli_t.660.ctf

krazynez, 2023-08-31:
Just tested on my 1K and it works fine.

krazynez, 2023-09-03:
Where did you get CXMB from.

JoseAaronLopezGarcia, 2023-09-08:
Any news on this? I too have no issues with CXMB. Sometimes I have to apply the theme twice for it to register correctly though.

JoseAaronLopezGarcia, 2023-09-09:
Closing due to lack of activity and reports from other users.

closed 2023-09-09 by JoseAaronLopezGarcia