PSP-Archive/ARK-4 #220: PS Vita standalone ARK doesn't appear in LiveArea after being installed

Goldtuga, 2023-09-01:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.63 Rev 2

### Steps to reproduce

Every time I use FastARK to install the standalone PS Vita ARK and it finishes installing it, ARK doesn't appear in LiveArea.

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

krazynez, 2023-09-01:
Okay onward to this one now :smile: Did the vita reboot after you installed ARK with FastARK-4 by itself?

Goldtuga, 2023-09-01:
> Okay onward to this one now 😄 Did the vita reboot after you installed ARK with FastARK-4 by itself?

Yes.

krazynez, 2023-09-01:
Okay try this.

Delete these files/folders:
* `ux0:pspemu/bgdl/*`
* `ux0:bgdl/t`
* `ux0:pspemu/PSP/GAME/NPUZ00146`

Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please

krazynez, 2023-09-01:
[image]

Goldtuga, 2023-09-02:
> Okay try this.
> 
> Delete these files/folders:
> 
> * `ux0:pspemu/bgdl/*`
> * `ux0:bgdl/t`
> * `ux0:pspemu/PSP/GAME/NPUZ00146`
> 
> Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please

I deleted those folders and followed the instructions from the WIKI and ARK is now appearing after running FastARK but it only does so after restarting the PS Vita so could you add the information one has to do it after using FastARK? Also, after installing the Sasuke vs. Commander minigame, a new menu appears, I exited FastARK and then restarted the PS Vita (and the minigame was installed, which is weird, because I clicked "X" to install it and ARK and the former is only installed after restarting the PS Vita and the latter only appears after installing the minigame).

Another thing, I'm not having the C1-2858-3 kernel error after running ARK or a PSP ISO.

> 

Now I'm not having this issue since the new folders are appearing in VitaShell.

Goldtuga, 2023-09-02:
> [image]

Now I'm not having this issue since the folders created by FastARK are noe appearing in VitaShell.

Goldtuga, 2023-09-02:
ARK's GitHub doesn't state to run ARK Loader under "Permanent CFW via Infinity" on the last step but "ARK CFW" so I guess it's referring to ARK Loader. Previously, it mentions "ARK Loader" instead of "ARK CFW" under "Manual installation" and "Live CFW" so could you change "ARK CFW" to "ARK Loader" under "Permanent CFW via Infinity" to not confuse users?

Could you also add ARK's GitHub the information under "Full Flash Installation" that this step can only be done after doing one of the three permanent ARK install methods as well as the information under "Live CFW" that this step can only be done on the PSP after doing the "Manual instalation" step to make things less confusing for users when installing ARK on the PSP?

Could you explain in ARK's GitHub what do you mean by "limitations with PS1 games"?

Also, FastARK's minigame is called Sasuke vs. Commander and not Sasuke Commander like ARK's GitHub states.

ARK's GitHub's WIKI doesn't state one can get a kernel error after running the standalone PS Vita ARK unlike ARK's GitHub, doesn't state ARK Loader must be run after running Infinity the second time on the PSP when ARK's GitHub states "ARL CFW" (instead of "ARK Loader") must be run, doesn't state the FLASH0.ARK file must be deleted on the PSP unlike ARK's GitHub under "Live CFW", doesn't mention the "Full Flash Installation" step on the PSP unlike ARK's GitHub and doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub so could you at least have the same information in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?

krazynez, 2023-09-02:
> > Okay try this.
> > Delete these files/folders:
> > 
> > * `ux0:pspemu/bgdl/*`
> > * `ux0:bgdl/t`
> > * `ux0:pspemu/PSP/GAME/NPUZ00146`
> > 
> > Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please
> 
> I deleted those folders, reinstalled FastARK-4, ran it, clicked the "X" button to install the Sasuke vs. Commander minigame with it, a new menu appeared, I closed the app, restarted the PS Vita and it's stated the Sasuke vs Commander minigame is being installed, which should already have occurred. If I don't restart the PS Vita, ARK doesn't appear in LiveArea, since it only appears after the Sasuke vs. Commander minigame is installed, which for some reasons only happens after restarting the PS Vita when I clicked the "X" button to install both this minigame and ARK and the PS Vita didn't restart by itself so could you add the information one must restart the PS Vita after running FastARK and clicking the "X" button or make the PS Vita restart after clicking it? Why does this new menu appear after clicking the "X" button?
> 

Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how it works. 

> Why do you state in ARK's GitHub Wiki to refresh LiveArea in VitaShell if FastARK always appears in the former after installing it?
> 

In case FastARK-4 does not show up that is how it would refresh the database so it would for VPK's.

> I updated to the latest ARK version from the Custom Launcher and I'm not having the C1-2858-3 error after running this ARK or a PSP ISO with it but after running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game) so maybe something in ARK is causing this?

Not sure can try to convert to and EBOOT and see if you get the same results.

krazynez, 2023-09-02:
> Could you state in ARK's GitHub under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods as well as state under "Live CFW" this step can only be done on the PSP after doing the "Manual instalation" step to make things less confusing for users when installing ARK on the PSP?
> 

Umm that is for the PSP only.... I already added it to the Wiki and we have discussed this already that I am moving on with the Wiki as first presidence as the README is going to be revised and have no installation instructions soon™. 

> Could you explain in ARK's GitHub what do you mean by "limitations with PS1 games"?
> 

I will add it to the Wiki yes.

> Also, ARK's GitHub's Wiki doesn't state one can get a kernel error after running the standalone PS Vita ARK unlike ARK's GitHub, doesn't state ARK Loader must be run after running Infinity the second time on the PSP when ARK's GitHub states "ARK CFW" (instead of "ARK Loader") must be run, doesn't state the FLASH0.ARK file must be deleted on the PSP unlike ARK's GitHub under "Live CFW", doesn't mention the "Full Flash Installation" step on the PSP unlike ARK's GitHub and doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub so could you at least state the same in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?

Infinity steps are loose, but the Wiki is more recent and tested. Again FLASH0.ARK does not **have** to be deleted its optional. Also I have not added the "Full Installation" yet that was next to be added. Again I am workig on the Wiki I am not updating the readme anymore with Installation steps. We except PR's if you want to do that.

Goldtuga, 2023-09-02:
> Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how [...]

But in my case it doesn't auto reboot, I have to manually reboot it for an alert to appear at the start that states Sasuke vs. Commander minigame is being installed and afterwards ARK appears in LiveArea.

But it always appears. It's ARK which doesn't appear after running FastARK if the PS Vuta is not manually restarted. How do I convert it to EBOOT?

krazynez, 2023-09-02:
> > > > Okay try this.
> > > > Delete these files/folders:
> > > > 
> > > > * `ux0:pspemu/bgdl/*`
> > > > * `ux0:bgdl/t`
> > > > * `ux0:pspemu/PSP/GAME/NPUZ00146`
> > > > 
> > > > Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please
> > > 
> > > 
> > > I deleted those folders, reinstalled FastARK-4, ran it, clicked the "X" button to install the Sasuke vs. Commander minigame with it, a new menu appeared, I closed the app, restarted the PS Vita and it's stated the Sasuke vs Commander minigame is being installed, which should already have occurred. If I don't restart the PS Vita, ARK doesn't appear in LiveArea, since it only appears after the Sasuke vs. Commander minigame is installed, which for some reasons only happens after restarting the PS Vita when I clicked the "X" button to install both this minigame and ARK and the PS Vita didn't restart by itself so could you add the information one must restart the PS Vita after running FastARK and clicking the "X" button or make the PS Vita restart after clicking it? Why does this new menu appear after clicking the "X" button?
> > 
> > 
> > Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how it works.
> > > Why do you state in ARK's GitHub Wiki to refresh LiveArea in VitaShell if FastARK always appears in the former after installing it?
> > 
> > 
> > In case FastARK-4 does not show up that is how it would refresh the database so it would for VPK's.
> > > I updated to the latest ARK version from the Custom Launcher and I'm not having the C1-2858-3 error after running this ARK or a PSP ISO with it but after running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game) so maybe something in ARK is causing this?
> > 
> > 
> > Not sure can try to convert to and EBOOT and see if you get the same results.
> 
> But in my case it doesn't auto reboot, I have to manually reboot it for an alert to appear at the start that states Sasuke vs. Commander minigame is being installed and afterwards ARK appears in LiveArea.
> 
> But it always appears. It's ARK which doesn't appear after running FastARK if the PS Vita is not manually restarted. How do I convert it to EBOOT?

Then you did something wrong, it auto reboots with a Fake Licence and and Real (copy) License when the steps are done. I tested both ways to verify that already.

> How do I convert it to EBOOT?

ISO-PBP Converter PSP

Goldtuga, 2023-09-02:
> Umm that is for the PSP only.... I already added it to the Wiki and we have discussed this already that I am moving on with the Wiki as first presidence as the README is going to be revised and have [...]

I know it is. That's why I complained about it because that step doesn't appear in the Wiki yet.

You should add it to the other page as well or just remove many information from it (like the install steps one) nd add it to the Wiki.

But the Wiki is still incomplete. Then you should state it's optional in the other page. First you state you added that step and now you state the opposite. What are PR's? Should I do that instead of creating another issue regarding inconsistencies between the two versions?

> Then you did something wrong, it auto reboots with a Fake Licence and and Real (copy) License when the steps are done. I tested both ways to verify that already.
>
>
> ISO-PBP Converter PSP

I did the same thing again and I still have to manually reboot it. Do I have to install a PSN PSP demo/game before installing the Sasuke vs. Commander minigame if no PSN PSP game/demo is installed? If so, shouldn't only one of the two be needed?

krazynez, 2023-09-02:
> I know it is. That's why I complained about it because that step doesn't appear in the Wiki yet.
>
> You should add it to the other page as well or just remove many information from it (like the [...]

Straight from the Wiki "**Recommended**: Download a PSP demo game from PSN (Ape Quest, Locoroco, etc) to have a valid PSP license available during installation."

Do you get an error with no license ?

Goldtuga, 2023-09-02:
> Straight from the Wiki "**Recommended**: Download a PSP demo game from PSN (Ape Quest, Locoroco, etc) to have a valid PSP license available during installation."
>
> Do you get an error with no license ?

Recommended is different from mandatory.

No, I don't. Is a menu supposed to appear after clicking the "X" button after running FastARK when installing the Sasuke vs. Commander minigame?

krazynez, 2023-09-02:
> Recommended is different from mandatory.

Yes exactly... I would have said it was required if so :wink: 

> 
> No, I don't. Is a menu supposed to appear after clicking the "X" button after running FastARK when installing the Sasuke vs. Commander minigame?

Try this one: [FastARK-4.zip](https://github.com/PSP-Archive/ARK-4/files/12504828/FastARK-4.zip) **RENAME TO FastARK-4.vpk** Github does not allow uploading vpks vita issues.

> 
> I still have the same problem after converting the game to EBOOT with ISO-PBP Converter PSP.

Goldtuga, 2023-09-02:
> > > > > > Could you state in ARK's GitHub under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods as well as state under "Live CFW" this step can only be done on the PSP after doing the "Manual instalation" step to make things less confusing for users when installing ARK on the PSP?
> > > > > 
> > > > > 
> > > > > Umm that is for the PSP only.... I already added it to the Wiki and we have discussed this already that I am moving on with the Wiki as first presidence as the README is going to be revised and have no installation instructions soon™.
> > > > > > Could you explain in ARK's GitHub what do you mean by "limitations with PS1 games"?
> > > > > 
> > > > > 
> > > > > I will add it to the Wiki yes.
> > > > > > Also, ARK's GitHub's Wiki doesn't state one can get a kernel error after running the standalone PS Vita ARK unlike ARK's GitHub, doesn't state ARK Loader must be run after running Infinity the second time on the PSP when ARK's GitHub states "ARK CFW" (instead of "ARK Loader") must be run, doesn't state the FLASH0.ARK file must be deleted on the PSP unlike ARK's GitHub under "Live CFW", doesn't mention the "Full Flash Installation" step on the PSP unlike ARK's GitHub and doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub so could you at least state the same in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?
> > > > > 
> > > > > 
> > > > > Infinity steps are loose, but the Wiki is more recent and tested. Again FLASH0.ARK does not **have** to be deleted its optional. Also I have not added the "Full Installation" yet that was next to be added. Again I am workig on the Wiki I am not updating the readme anymore with Installation steps. We except PR's if you want to do that.
> > > > 
> > > > 
> > > > I know it is. That's why I complained about it because that step doesn't appear in the Wiki yet.
> > > > You should add it to the other page as well or just remove many information from it (like the install steps one) nd add it to the Wiki.
> > > > But the Wiki is still incomplete. Then you should state it's optional in the other page. First you state you added that step and now you state the opposite. What are PR's? Should I do that instead of creating another issue regarding inconsistencies between the two versions?
> > > > > > > > > Okay try this.
> > > > > > > > > Delete these files/folders:
> > > > > > > > > 
> > > > > > > > > * `ux0:pspemu/bgdl/*`
> > > > > > > > > * `ux0:bgdl/t`
> > > > > > > > > * `ux0:pspemu/PSP/GAME/NPUZ00146`
> > > > > > > > > 
> > > > > > > > > Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please
> > > > > > > > 
> > > > > > > > 
> > > > > > > > I deleted those folders, reinstalled FastARK-4, ran it, clicked the "X" button to install the Sasuke vs. Commander minigame with it, a new menu appeared, I closed the app, restarted the PS Vita and it's stated the Sasuke vs Commander minigame is being installed, which should already have occurred. If I don't restart the PS Vita, ARK doesn't appear in LiveArea, since it only appears after the Sasuke vs. Commander minigame is installed, which for some reasons only happens after restarting the PS Vita when I clicked the "X" button to install both this minigame and ARK and the PS Vita didn't restart by itself so could you add the information one must restart the PS Vita after running FastARK and clicking the "X" button or make the PS Vita restart after clicking it? Why does this new menu appear after clicking the "X" button?
> > > > > > > 
> > > > > > > 
> > > > > > > Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how it works.
> > > > > > > > Why do you state in ARK's GitHub Wiki to refresh LiveArea in VitaShell if FastARK always appears in the former after installing it?
> > > > > > > 
> > > > > > > 
> > > > > > > In case FastARK-4 does not show up that is how it would refresh the database so it would for VPK's.
> > > > > > > > I updated to the latest ARK version from the Custom Launcher and I'm not having the C1-2858-3 error after running this ARK or a PSP ISO with it but after running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game) so maybe something in ARK is causing this?
> > > > > > > 
> > > > > > > 
> > > > > > > Not sure can try to convert to and EBOOT and see if you get the same results.
> > > > > > 
> > > > > > 
> > > > > > But in my case it doesn't auto reboot, I have to manually reboot it for an alert to appear at the start that states Sasuke vs. Commander minigame is being installed and afterwards ARK appears in LiveArea.
> > > > > > But it always appears. It's ARK which doesn't appear after running FastARK if the PS Vita is not manually restarted. How do I convert it to EBOOT?
> > > > > 
> > > > > 
> > > > > Then you did something wrong, it auto reboots with a Fake Licence and and Real (copy) License when the steps are done. I tested both ways to verify that already.
> > > > > > How do I convert it to EBOOT?
> > > > > 
> > > > > 
> > > > > ISO-PBP Converter PSP
> > > > 
> > > > 
[... cut ...]  PSP demo/game no longer appears in LiveArea. If installing a PSN PSP demo/game, should I click "START" or "X" and may FastARK (and the PSN PSP demo/game if still present on LiveArea) be removed after installing ARK? Anyway, I think it's better if you add a new point before "1." on ARK's GitHub Wiki stating to install the PSN PSP demo/game and remove the "Recommended" part not to confuse users.

I already tried disabling the "Memory Stick Speedup" setting and I still have this issue in the game.

krazynez, 2023-09-02:
> This time I installed a PSN PSP demo before installing ARK, I didn't have the menu and the PS Vita restarted by itself but when clicking "START" it installs the Sasuke vs. Commander minigame, [...]

The PSN PSP DEMO/GAME is **ONLY** for a license file nothing else! It does not affect the PSP DEMO/GAME.

The UI changes if it detect a Game for FastARK-4, bottom left will change.

IT IS NOT REQUIRED! Just Recommended. If you have henkau you don't need a PSP Demo/Game but you can still use it if you do as well.

Goldtuga, 2023-09-02:
> The PSN PSP DEMO/GAME is **ONLY** for a license file nothing else! It does not affect the PSP DEMO/GAME.
>
> The UI changes if it detect a Game for FastARK-4, bottom left will change.
>
> IT IS NOT [...]

Then is it normal for the PS Vita not to restart itself if no PSN PSP demo/game is installed on HENkaku? I always have to manually restart if not installing one since it doesn't restart by itself and so doesn't start installing the Sasuke vs. Commander minigame and because of that it doesn't install ARK.

krazynez, 2023-09-02:
> a not to restart itself if no PSN PSP demo

Use the one I provided to you above. Just rename it.

Goldtuga, 2023-09-02:
> Then is it normal for the PS Vita not to restart itself if no PSN PSP demo/game is installed on HENkaku? I always have to manually restart if not installing one since it doesn't restart by itself [...]

I tried that version and now it restarted the PS Vita but after running ARK, it froze when clicking the D-pad "down" button to run a PSP ISO so I closed the app, rerun it and then it didn't freeze when clicking the D-pad "down" button but after clicking the "X" button to run a PSP ISO, it froze on a black screen. I restarted the PS Vita, ran ARK and didn't have the previous freezing problems. Will you update ARK with that version?

krazynez, 2023-09-03:
> > > > > > > > > > Could you state in ARK's GitHub under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods as well as state under "Live CFW" this step can only be done on the PSP after doing the "Manual instalation" step to make things less confusing for users when installing ARK on the PSP?
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > Umm that is for the PSP only.... I already added it to the Wiki and we have discussed this already that I am moving on with the Wiki as first presidence as the README is going to be revised and have no installation instructions soon™.
> > > > > > > > > > Could you explain in ARK's GitHub what do you mean by "limitations with PS1 games"?
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > I will add it to the Wiki yes.
> > > > > > > > > > Also, ARK's GitHub's Wiki doesn't state one can get a kernel error after running the standalone PS Vita ARK unlike ARK's GitHub, doesn't state ARK Loader must be run after running Infinity the second time on the PSP when ARK's GitHub states "ARK CFW" (instead of "ARK Loader") must be run, doesn't state the FLASH0.ARK file must be deleted on the PSP unlike ARK's GitHub under "Live CFW", doesn't mention the "Full Flash Installation" step on the PSP unlike ARK's GitHub and doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub so could you at least state the same in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > Infinity steps are loose, but the Wiki is more recent and tested. Again FLASH0.ARK does not **have** to be deleted its optional. Also I have not added the "Full Installation" yet that was next to be added. Again I am workig on the Wiki I am not updating the readme anymore with Installation steps. We except PR's if you want to do that.
> > > > > > > > 
> > > > > > > > 
> > > > > > > > I know it is. That's why I complained about it because that step doesn't appear in the Wiki yet.
> > > > > > > > You should add it to the other page as well or just remove many information from it (like the install steps one) nd add it to the Wiki.
> > > > > > > > But the Wiki is still incomplete. Then you should state it's optional in the other page. First you state you added that step and now you state the opposite. What are PR's? Should I do that instead of creating another issue regarding inconsistencies between the two versions?
> > > > > > > > > > > > > Okay try this.
> > > > > > > > > > > > > Delete these files/folders:
> > > > > > > > > > > > > 
> > > > > > > > > > > > > * `ux0:pspemu/bgdl/*`
> > > > > > > > > > > > > * `ux0:bgdl/t`
> > > > > > > > > > > > > * `ux0:pspemu/PSP/GAME/NPUZ00146`
> > > > > > > > > > > > > 
> > > > > > > > > > > > > Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > I deleted those folders, reinstalled FastARK-4, ran it, clicked the "X" button to install the Sasuke vs. Commander minigame with it, a new menu appeared, I closed the app, restarted the PS Vita and it's stated the Sasuke vs Commander minigame is being installed, which should already have occurred. If I don't restart the PS Vita, ARK doesn't appear in LiveArea, since it only appears after the Sasuke vs. Commander minigame is installed, which for some reasons only happens after restarting the PS Vita when I clicked the "X" button to install both this minigame and ARK and the PS Vita didn't restart by itself so could you add the information one must restart the PS Vita after running FastARK and clicking the "X" button or make the PS Vita restart after clicking it? Why does this new menu appear after clicking the "X" button?
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how it works.
> > > > > > > > > > > > Why do you state in ARK's GitHub Wiki to refresh LiveArea in VitaShell if FastARK always appears in the former after installing it?
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > In case FastARK-4 does not show up that is how it would refresh the database so it would for VPK's.
> > > > > > > > > > > > I updated to the latest ARK version from the Custom Launcher and I'm not having the C1-2858-3 error after running this ARK or a PSP ISO with it but after running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game) so maybe something in ARK is causing this?
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > Not sure can try to convert to and EBOOT and see if you get the same results.
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > But in my case it doesn't auto reboot, I have to manually reboot it for an alert to appear at the start that states Sasuke vs. Commander minigame is being installed and afterwards ARK appears in LiveArea.
> > > > > > > > > > But it always appears. It's ARK which doesn't appear after running FastARK if the PS Vita is not manually restarted. How do I convert it to EBOOT?
> > > > > > > > > 
> > > > > > > > > 
[... cut ...]
> I tried that version and now it restarted the PS Vita but after running ARK, it froze when clicking the D-pad "down" button to run a PSP ISO so I closed the app, re-ran it and then it didn't freeze when clicking the D-pad "down" button, but after clicking the "X" button on a PSP ISO, it froze with a black screen. I restarted the PS Vita, ran ARK and didn't have the previous freezing problems. Will you update ARK with this version?

Already did.

This is what it should look like (even caught the K.BIN error not on purpose)
https://github.com/PSP-Archive/ARK-4/assets/39999447/6a5989fa-8647-403e-9ccb-758e5adb2dba

Goldtuga, 2023-09-03:
> Already did.
>
> This is what it should look like (even caught the K.BIN error not on purpose) https://github.com/PSP-Archive/ARK-4/assets/39999447/6a5989fa-8647-403e-9ccb-758e5adb2dba

But the latest version is still 4.20.63 Rev 2.

It looked the same in my PS Vita except the part where it didn't find a license and had an enploit error. If it supposedly still crashes, will you fix it?

Goldtuga, 2023-09-03:
> Already did.
>
> This is what it should look like (even caught the K.BIN error not on purpose) https://github.com/PSP-Archive/ARK-4/assets/39999447/6a5989fa-8647-403e-9ccb-758e5adb2dba

So, I reinstalled that version you mentioned and now it didn't freeze when clicking the D-pad "down" button to run a PSP ISO and after clicking the "X" button on a PSP ISO.

krazynez, 2023-09-03:
> > > > > > > > > > > > Could you state in ARK's GitHub under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods as well as state under "Live CFW" this step can only be done on the PSP after doing the "Manual instalation" step to make things less confusing for users when installing ARK on the PSP?
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > Umm that is for the PSP only.... I already added it to the Wiki and we have discussed this already that I am moving on with the Wiki as first presidence as the README is going to be revised and have no installation instructions soon™.
> > > > > > > > > > > > Could you explain in ARK's GitHub what do you mean by "limitations with PS1 games"?
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > I will add it to the Wiki yes.
> > > > > > > > > > > > Also, ARK's GitHub's Wiki doesn't state one can get a kernel error after running the standalone PS Vita ARK unlike ARK's GitHub, doesn't state ARK Loader must be run after running Infinity the second time on the PSP when ARK's GitHub states "ARK CFW" (instead of "ARK Loader") must be run, doesn't state the FLASH0.ARK file must be deleted on the PSP unlike ARK's GitHub under "Live CFW", doesn't mention the "Full Flash Installation" step on the PSP unlike ARK's GitHub and doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub so could you at least state the same in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > Infinity steps are loose, but the Wiki is more recent and tested. Again FLASH0.ARK does not **have** to be deleted its optional. Also I have not added the "Full Installation" yet that was next to be added. Again I am workig on the Wiki I am not updating the readme anymore with Installation steps. We except PR's if you want to do that.
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > I know it is. That's why I complained about it because that step doesn't appear in the Wiki yet.
> > > > > > > > > > You should add it to the other page as well or just remove many information from it (like the install steps one) nd add it to the Wiki.
> > > > > > > > > > But the Wiki is still incomplete. Then you should state it's optional in the other page. First you state you added that step and now you state the opposite. What are PR's? Should I do that instead of creating another issue regarding inconsistencies between the two versions?
> > > > > > > > > > > > > > > Okay try this.
> > > > > > > > > > > > > > > Delete these files/folders:
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > * `ux0:pspemu/bgdl/*`
> > > > > > > > > > > > > > > * `ux0:bgdl/t`
> > > > > > > > > > > > > > > * `ux0:pspemu/PSP/GAME/NPUZ00146`
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > I deleted those folders, reinstalled FastARK-4, ran it, clicked the "X" button to install the Sasuke vs. Commander minigame with it, a new menu appeared, I closed the app, restarted the PS Vita and it's stated the Sasuke vs Commander minigame is being installed, which should already have occurred. If I don't restart the PS Vita, ARK doesn't appear in LiveArea, since it only appears after the Sasuke vs. Commander minigame is installed, which for some reasons only happens after restarting the PS Vita when I clicked the "X" button to install both this minigame and ARK and the PS Vita didn't restart by itself so could you add the information one must restart the PS Vita after running FastARK and clicking the "X" button or make the PS Vita restart after clicking it? Why does this new menu appear after clicking the "X" button?
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how it works.
> > > > > > > > > > > > > > Why do you state in ARK's GitHub Wiki to refresh LiveArea in VitaShell if FastARK always appears in the former after installing it?
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > In case FastARK-4 does not show up that is how it would refresh the database so it would for VPK's.
> > > > > > > > > > > > > > I updated to the latest ARK version from the Custom Launcher and I'm not having the C1-2858-3 error after running this ARK or a PSP ISO with it but after running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game) so maybe something in ARK is causing this?
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > Not sure can try to convert to and EBOOT and see if you get the same results.
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > But in my case it doesn't auto reboot, I have to manually reboot it for an alert to appear at the start that states Sasuke vs. Commander minigame is being installed and afterwards ARK appears in LiveArea.
[... cut ...]
> So, I reinstalled that version you mentioned (the one which has to be renamed) and now it didn't freeze when clicking the D-pad "down" button to run a PSP ISO and after clicking the "X" button on a PSP ISO.
> 
> Anyway, will you upload that version which needs to be renamed and that after clicking the "X" button installs the Sasuke vs. Commander minigame and restarts the PS Vita instead of opening a new menu and only installing ARK after the PS Vita is manually restarted?

it uses Sasuke vs. Commander for the exploit that is what FastARK-4 installed and injects ARK-4 (hense the bubble)

Goldtuga, 2023-09-03:
> it uses Sasuke vs. Commander for the exploit that is what FastARK-4 installed and injects ARK-4 (hense the bubble)

Should one click the "X" or the "START" button after installing a PSN PSP demo/game and after running FastARK? If doing the former, my PS Vita looks the same as your video except the part where FastARK finds a license and one has to click the "X" button and the part where ARK after being run has an exploit error, but if doing the latter, my PS Vita looks the same as your video (this time FastARK even finds a license and one has to click the "X" button, like in your video) but ARK doesn't have an exploit error after being run, unlike in your video.

krazynez, 2023-09-03:
> > > > > > > > > > > > > > Could you state in ARK's GitHub under "Full Flash Installation" this step can only be done after doing one of the three permanent ARK install methods as well as state under "Live CFW" this step can only be done on the PSP after doing the "Manual instalation" step to make things less confusing for users when installing ARK on the PSP?
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > Umm that is for the PSP only.... I already added it to the Wiki and we have discussed this already that I am moving on with the Wiki as first presidence as the README is going to be revised and have no installation instructions soon™.
> > > > > > > > > > > > > > Could you explain in ARK's GitHub what do you mean by "limitations with PS1 games"?
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > I will add it to the Wiki yes.
> > > > > > > > > > > > > > Also, ARK's GitHub's Wiki doesn't state one can get a kernel error after running the standalone PS Vita ARK unlike ARK's GitHub, doesn't state ARK Loader must be run after running Infinity the second time on the PSP when ARK's GitHub states "ARK CFW" (instead of "ARK Loader") must be run, doesn't state the FLASH0.ARK file must be deleted on the PSP unlike ARK's GitHub under "Live CFW", doesn't mention the "Full Flash Installation" step on the PSP unlike ARK's GitHub and doesn't mention ARK has limitations with PS1 games unlike ARK's GitHub so could you at least state the same in both places and replace "ARK CFW" by "ARK Loader" under ARK's Github "Permanent CFW via Infinity" step?
> > > > > > > > > > > > > 
> > > > > > > > > > > > > 
> > > > > > > > > > > > > Infinity steps are loose, but the Wiki is more recent and tested. Again FLASH0.ARK does not **have** to be deleted its optional. Also I have not added the "Full Installation" yet that was next to be added. Again I am workig on the Wiki I am not updating the readme anymore with Installation steps. We except PR's if you want to do that.
> > > > > > > > > > > > 
> > > > > > > > > > > > 
> > > > > > > > > > > > I know it is. That's why I complained about it because that step doesn't appear in the Wiki yet.
> > > > > > > > > > > > You should add it to the other page as well or just remove many information from it (like the install steps one) nd add it to the Wiki.
> > > > > > > > > > > > But the Wiki is still incomplete. Then you should state it's optional in the other page. First you state you added that step and now you state the opposite. What are PR's? Should I do that instead of creating another issue regarding inconsistencies between the two versions?
> > > > > > > > > > > > > > > > > Okay try this.
> > > > > > > > > > > > > > > > > Delete these files/folders:
> > > > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > > > * `ux0:pspemu/bgdl/*`
> > > > > > > > > > > > > > > > > * `ux0:bgdl/t`
> > > > > > > > > > > > > > > > > * `ux0:pspemu/PSP/GAME/NPUZ00146`
> > > > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > > > Then redo [these](https://github.com/PSP-Archive/ARK-4/wiki/FastARK) steps please
> > > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > > I deleted those folders, reinstalled FastARK-4, ran it, clicked the "X" button to install the Sasuke vs. Commander minigame with it, a new menu appeared, I closed the app, restarted the PS Vita and it's stated the Sasuke vs Commander minigame is being installed, which should already have occurred. If I don't restart the PS Vita, ARK doesn't appear in LiveArea, since it only appears after the Sasuke vs. Commander minigame is installed, which for some reasons only happens after restarting the PS Vita when I clicked the "X" button to install both this minigame and ARK and the PS Vita didn't restart by itself so could you add the information one must restart the PS Vita after running FastARK and clicking the "X" button or make the PS Vita restart after clicking it? Why does this new menu appear after clicking the "X" button?
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > Why did you close the app? FastARK-4 auto reboots the vita and finishes the install. Please follow the Wiki steps to a T. FastARK-4 is OneLua based you can look into that for more explanation on how it works.
> > > > > > > > > > > > > > > > Why do you state in ARK's GitHub Wiki to refresh LiveArea in VitaShell if FastARK always appears in the former after installing it?
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > In case FastARK-4 does not show up that is how it would refresh the database so it would for VPK's.
> > > > > > > > > > > > > > > > I updated to the latest ARK version from the Custom Launcher and I'm not having the C1-2858-3 error after running this ARK or a PSP ISO with it but after running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game) so maybe something in ARK is causing this?
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > 
> > > > > > > > > > > > > > > Not sure can try to convert to and EBOOT and see if you get the same results.
> > > > > > > > > > > > > > 
> > > > > > > > > > > > > > 
[... cut ...]  being error, but if doing the latter, my PS Vita looks the same as your video (this time FastARK even finds a license and one has to click the "X" button, like in your video) but ARK doesn't have an exploit error after being run, unlike in your video. May Ape Quest and FastARK be deleted after installing ARK?
> 
> Also, some folders aren't appearing in FileZilla again.

What ever the button on the bottom left says to do. Do that.

Yes you can delete both afterwards and just keep the ARK bubble.

Goldtuga, 2023-09-03:
> What ever the button on the bottom left says to do. Do that.
>
> Yes you can delete both afterwards and just keep the ARK bubble.

But it never mentions the "START" button, only the "X" one, and ARK's GitHub Wiki mentions both buttons and apparently you also clicked the "START" button according to the video you showed.

But why doesn't it use the PSN PSP demo/game installed right before FastARK instead of using a new game/demo that is installed right by it?

krazynez, 2023-09-03:
After ARK is installed go back into FastARK the button will change because a PSP game is detected

Goldtuga, 2023-09-03:
> After ARK is installed go back into FastARK the button will change because a PSP game is detected. If you have PSP games already installed and FastARK detects it. It will be a diffrent menu selection.

Re-read my previous reply since I edited it. Will you upload the version which needs to be renamed or state in ARK's GitHub Wiki one has to manually restart the PS Vita if installing the Sasuke vs. Commander minigame if not having installed a PSN PSP demo/game before running FastARK?

krazynez, 2023-09-03:
> Re-read my previous reply since I edited it. Will you upload the version which needs to be renamed or state in ARK's GitHub Wiki one has to manually restart the PS Vita if installing the Sasuke vs. [...]

I already pushed the changes and no you don't manually reboot both ways are the same.

Goldtuga, 2023-09-03:
> I already pushed the changes and no you don't manually reboot both ways are the same.

But I reinstalled ARK and I still had to manually reboot it.

krazynez, 2023-09-03:
> > > > After ARK is installed go back into FastARK the button will change because a PSP game is detected. If you have PSP games already installed and FastARK detects it. It will be a diffrent menu selection.
> > > 
> > > 
> > > Re-read my previous reply since I edited it. Will you upload the version which needs to be renamed or state in ARK's GitHub Wiki one has to manually restart the PS Vita if installing the Sasuke vs. Commander minigame if not having installed a PSN PSP demo/game before running FastARK?
> > 
> > 
> > I already pushed the changes and no you don't manually reboot both ways are the same.
> 
> But I reinstalled ARK and I still had to manually reboot thr PS Vita since FastARK doesn't do it.

did you download the latest release from the Wiki from like within the past few hours?

Goldtuga, 2023-09-03:
> did you download the latest release from the Wiki from like within the past few hours?

I downloaded it 5 minutes ago.

krazynez, 2023-09-03:
No License way
https://github.com/PSP-Archive/ARK-4/assets/39999447/5baf857b-719a-43bc-8426-4f74f163d8fb

Goldtuga, 2023-09-03:
> No License way https://github.com/PSP-Archive/ARK-4/assets/39999447/5baf857b-719a-43bc-8426-4f74f163d8fb

That's what I do but for some reason FastARK doesn't state "No licenses were found, using a fake license instead. Your bubble will only work with henkaku.", "Your PSVita will restart. Remember to activate henkaku again." and "PS Vita will restart and finish installing ARK-4".

Goldtuga, 2023-09-03:
No License way https://github.com/PSP-Archive/ARK-4/assets/39999447/5baf857b-719a-43bc-8426-4f74f163d8fb

That's what I do but for some reason FastARK doesn't state "No licenses were found, using a fake license instead. Your bubble will only work with henkaku.", "Your PSVita will restart. Remember to activate henkaku again." and "PS Vita will restart and finish installing ARK-4". It only states "Successfully installed ARK" and afterwards open a menu called One debugger which mentions "functions.lua", "56", "attempt to index field '?' (a nil value)", "Server ftp:/XXXX.XXX.X.XX:1337", "Total access (Unsafe mode)", "Press the button X: Restart the Script", "Press the button O: Exit to LiveArea" and "Press the button SELECT: USB Mode".

krazynez, 2023-09-03:
That is an error for lua

Goldtuga, 2023-09-03:
What is lua and why is it only happening to me?

krazynez, 2023-09-03:
Oh nope my fault pushed the code forgot to make the release one sec

krazynez, 2023-09-03:
Okay now its correct on the wiki for the proper release.

Goldtuga, 2023-09-03:
> Okay now its correct on the wiki for the proper release.

I installed it and now it happened the same as in your videos but it froze the PS Vita to the point I had to click its power button to turn it off, this time when exiting ARK (the other two times were when clicking the D-pad "down" button to run a PSP ISO and after clicking the "X" button on a PSP ISO and I could turn off the PS TV using the controller). I removed it, reinstalled it and it didn't freeze the PS Vita.

Goldtuga, 2023-09-03:
> Okay now its correct on the wiki for the proper release.

I installed it and now it happened the same as in your videos but it froze the PS Vita to the point I had to click its power button to turn it off, this time when exiting ARK (the other two times were when clicking the D-pad "down" button to run a PSP ISO and after clicking the "X" button on a PSP ISO and I could turn off the PS TV using the controller). I removed it, reinstalled it and it didn't freeze the PS Vita.

Also, the "Force update" setting never worked and I always had to manually run the updater from the "Network" setting,

krazynez, 2023-09-03:
Force Update is for ARK to always "find" and update using the updater. If that is disabled and your on the latest stable it will say no need to update.

Goldtuga, 2023-09-03:
> Force Update is for ARK to always "find" and update using the updater. If that is disabled and your on the latest stable it will say no need to update.

What about the freezing issues? 

OK, but there many times I had it enabled, used the updater and it still didn't update. Recently, it actually downgraded ARK.

krazynez, 2023-09-03:
> What about the freezing issues?
>
> OK, but there many times I had it enabled, used the updater and it still didn't update. Recently, it actually downgraded ARK.

Freezing can happen if a thread locks up or some random issue not to much we can do about that.

Not sure it should just update. Downgrade only happens if you install simething later like our test server then swap back to stable.

Goldtuga, 2023-09-03:
> Freezing can happen if a thread locks up or some random issue not to much we can do about that.
>
> Not sure it should just update. Downgrade only happens if you install simething later like our test [...]

It only happens with this program, and besides, you apparently fixed the kernel error (although it seems you still had it in one of your videos).

A while ago, I tried PSP's ARK's Custom Launcher's updater and it actually downgraded the ARK 4.20.63 Rev 1 version to the previous one (4.20.63) instead of upgrading it to the 4.20.63 Rev 2 version.

Goldtuga, 2023-09-03:
> Freezing can happen if a thread locks up or some random issue not to much we can do about that.
>
> Not sure it should just update. Downgrade only happens if you install simething later like our test [...]

It only happens with this program, and besides, you apparently fixed the kernel error (although it seems you still had it in one of your videos). Also, I didn't have this kind of freezes after installing ARK via the NoPspEmuDrm.

A while ago, I tried PSP's ARK's Custom Launcher's updater and it actually downgraded ARK from the 4.20.63 Rev 1 version to the previous one (4.20.63) instead of upgrading it to the 4.20.63 Rev 2 version.

Goldtuga, 2023-09-03:
> Freezing can happen if a thread locks up or some random issue not to much we can do about that.
>
> Not sure it should just update. Downgrade only happens if you install simething later like our test [...]

If it always finds an update as long as the "Force update" setting is enabled, does that mean if ARK's latest version is already installed will the installer install the same version again?

It only happens with this program, and besides, you apparently fixed the kernel error (although it seems you still had it in one of your videos). Also, I didn't have this kind of freezes after installing ARK via NoPspEmuDrm.

A while ago, I tried PSP's ARK's Custom Launcher's updater and it actually downgraded ARK from the 4.20.63 Rev 1 version to the previous one (4.20.63) instead of upgrading it to the 4.20.63 Rev 2 version.

Goldtuga, 2023-09-03:
> Freezing can happen if a thread locks up or some random issue not to much we can do about that.
>
> Not sure it should just update. Downgrade only happens if you install simething later like our test [...]

It only happens with this program, and besides, you apparently fixed the kernel error (although it seems you still had it in one of your videos). Also, I didn't have this kind of freezes after installing ARK via NoPspEmuDrm.

If it always finds an update as long as the "Force update" setting is enabled, does that mean if ARK's latest version is already installed will the installer install the same version again? Maybe I should disable this settung and only run the updater when a new version is released.

A while ago, I tried PSP's ARK's Custom Launcher's updater and it actually downgraded ARK from the 4.20.63 Rev 1 version to the previous one (4.20.63) instead of upgrading it to the 4.20.63 Rev 2 version.

krazynez, 2023-09-03:
> > > > Force Update is for ARK to always "find" and update using the updater. If that is disabled and your on the latest stable it will say no need to update.
> > > 
> > > 
> > > What about the freezing issues?
> > > OK, but there many times I had it enabled, used the updater and it still didn't update. Recently, it actually downgraded ARK.
> > 
> > 
> > Freezing can happen if a thread locks up or some random issue not to much we can do about that.
> > Not sure it should just update. Downgrade only happens if you install simething later like our test server then swap back to stable.
> 
> It only happens with this program, and besides, you apparently fixed the kernel error (although it seems you still had it in one of your videos). Also, I didn't have this kind of freezes after installing ARK via NoPspEmuDrm.
> 
> If it always finds an update as long as the "Force update" setting is enabled, does that mean if ARK's latest version is already installed will the installer install the same version again? Maybe I should disable this setting and only run the updater when a new version is released.
> 
> A while ago, I tried PSP's ARK's Custom Launcher's updater and it actually downgraded ARK from the 4.20.63 Rev 1 version to the previous one (4.20.63) instead of upgrading it to the 4.20.63 Rev 2 version.

Yes I would disable that for that exact reason

krazynez, 2023-09-03:
Okay FastARK-4 is working as it should. I am closing this.

closed 2023-09-03 by krazynez

Goldtuga, 2023-09-03:
> Okay FastARK-4 is working as it should. I am closing this.

But it sometimes freezes the PS Vita.