PSP-Archive/ARK-4 #221: Custom Launcher Navigation Delay

TheSubPlayer, 2023-09-02:
### PSP/Vita Model

PSP-3000 (Slim & Bright)

### ARK Version

4.20.63 r1

### Homebrew/GAME (if applicable)

Custom Launcher

### Steps to reproduce

1. Go to file manager
2. Navigate between ISO files

https://github.com/PSP-Archive/ARK-4/assets/110436924/9f8259ab-9bd2-41e0-9065-c93b4e7db3e1

This delay is probably due to ICON0 loading

### USB Charge

Always

### Overclock

Always

### Power Save

XMB

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Always

### Memory Stick Speedup

Alway

### Inferno Cache

Off

### Skip Sony Logos

On

### Hide PIC0 and PIC1

On

### Hide MAC Address

Off

### Hide DLC

On

### Turn off LEDs

Disabled

krazynez, 2023-09-02:
You can disable ICON0 in the settings see if that indeed speeds it up, I am guessing so as well.

krazynez, 2023-09-03:
Verified, yes if you disable the Icons it fast scrolls not much we can do if enabled as it parses on selection so it does some IO calls to get the ICON0 then displays it.

JoseAaronLopezGarcia, 2023-09-05:
indeed it's because of icon0 loading, probably can be improved but it's not too critical for now

JoseAaronLopezGarcia, 2024-03-09:
Closing as there's an option to disable icon0 in browser and have better scrolling.

closed 2024-03-09 by JoseAaronLopezGarcia