PSP-Archive/ARK-4 #222: Audio bug in european Dragon Ball Z: Tenkaichi Tag Team ISO

Goldtuga, 2023-09-04:
### PSP/Vita Model

PCH-2000 (Vita Slim)

### ARK Version

4.20.63 Rev 2

### Homebrew/GAME (if applicable)

Dragon Ball Z: Tenkaichi Tag Team (ULES00789)

### Steps to reproduce

After running the european Dragon Ball Z Budokai Tenkaichi 3 PSP ISO, the narrator pronounces the title very fast (unlike the physical PSP game). Also, ARK installed via FastARK sometimes freezes and I have to turn off the PS Vita (including by pressing its power button because thr PS Vita doesn't respond to the PS button).

### USB Charge

Disabled

### Overclock

Disabled

### Power Save

Disabled

### Balanced Energy Mode

Disabled

### Autoboot Launcher

Off

### Disable PSP Go Pause

Off

### Unlock Extra Memory

Disabled

### Memory Stick Speedup

Disabled

### Inferno Cache

Off

### Skip Sony Logos

Off

### Hide PIC0 and PIC1

Off

### Hide MAC Address

Off

### Hide DLC

Off

### Turn off LEDs

Disabled

JoseAaronLopezGarcia, 2023-09-05:
Try using another format like CSO or PBP to force it to be slower. There aren't that many games on PSP that are sensitive to timing but I guess we've found one.

Goldtuga, 2023-09-05:
> Try using another format like CSO or PBP to force it to be slower. There aren't that many games on PSP that are sensitive to timing but I guess we've found one.

Already did and it still has this issue. I didn't know it should be slower if on those formats. Anyway, what I told was just an example and the audio is rushed in other parts of the game and even the gameplay seems rushed.

Also, ARK installed via FastARK sometimes freezes.

JoseAaronLopezGarcia, 2023-09-05:
Perhaps it can be fixed by forcing the system to use PSP clock speeds instead of stock. Will have to debug this further.

Goldtuga, 2023-09-05:
Also, ARK installed via FastARK sometimes freezes.

Goldtuga, 2023-09-06:
> Tested using FastARK-4 with our new settings and works fine. I don't see any slowdowns or speed ups played a few matches just to see if would or not. Also the main menu from what I have found sounds the same on other CFW's as well.

What do you mean by "new settings"? Did you use a pre-release? I don't have this problem with the physical PSP version and on Pro-C and Adrenaline. Did you test the european ISO?

> @Goldtuga can you try the latest pre-release? We've added options to change CPU speed, perhaps the "PSP CPU Clock" or "Balanced Energy Mode" will fix this issue. If it does, I advise to add a custom line to SETTINGS.TXT so you only enable it on that specific game. Something like this:
> 
> ULES00789, overclock, on
> 
> Note: on Vita, the "PSP CPU Clock" corresponds with real PSP's "Overclock", but on Vita it's not called "Overclock" because Vita has a different Overclock that can go to 444MHz or even 555MHz while the PSP's Overclock is 333MHz, so to avoid any confusion, we decided to name the PSP's "Overclock" as "PSP CPU Clock" on Vita.

OK, I'll try it, but wouldn't it be better if ARK knew the games which need those settings turned on instead of needing the user intervention?

> @Goldtuga btw ULES00789 is not the correct GAME ID for Dragon Ball Z: Tenkaichi Tag Team. When you are in the Custom Launcher you can press L on the game's icon and the "view info" it will tell you the correct Game ID.

You're right, my mistake. It's actually ULES01456.

JoseAaronLopezGarcia, 2023-09-06:
Might be possible to have it inside the CFW.

Goldtuga, 2023-09-06:
> @Goldtuga if you can confirm that "PSP CPU Clock" setting fixes this game on Standalone then I can make it so it gets applied automatically.

So, I enabled that setting and I still have this issue.

Goldtuga, 2023-09-07:
So, I saw the difference between the physical PSP version, the ISO played on Adrenaline and the ISO played on ARK and the latter two have the same speed and are after all just very slightly faster than the physical version so maybe it's normal, after all. I even tested the non-pre-release version and the game has the same speed as the pre-release one. 

Anyway, ARK installed via FastARK sometimes freezes and I have to turn off the PS TV (including being forced to press its power button because the PS TV doesn't respond to the controller).

Goldtuga, 2023-09-07:
> @Goldtuga btw ULES00789 is not the correct GAME ID for Dragon Ball Z: Tenkaichi Tag Team. When you are in the Custom Launcher you can press L on the game's icon and the "view info" it will tell you the correct Game ID.

> @Goldtuga if you can confirm that "PSP CPU Clock" setting fixes this game on Standalone then I can make it so it gets applied automatically.

Also, why does ARK have new commits every day but ARK's version number stays the same?

JoseAaronLopezGarcia, 2023-09-07:
> Also, why does ARK have new commits every day but ARK's version number stays the same?

Sometimes we make revisions when we do changes, but other times we prefer to keep the changes in pre-release and test them well to make a new proper release.

Goldtuga, 2023-09-08:
> Sometimes we make revisions when we do changes, but other times we prefer to keep the changes in pre-release and test them well to make a new proper release.

Does that go for the ARK installed by FastARK as well or just for the "normal" ARK?

JoseAaronLopezGarcia, 2023-09-08:
> Does the ARK installed by FastARK always have the latest ARK commits or just the "normal" ARK (the one downloaded from ARK's GitHub repo)?

What do you mean by "normal"? There's only one version of ARK.
FastARK is usually not up to date with new releases, but it can be easily updated using Network Tools in Custom Launcher.

Goldtuga, 2023-09-08:
> What do you mean by "normal"? There's only one version of ARK. FastARK is usually not up to date with new releases, but it can be easily updated using Network Tools in Custom Launcher.

OK, so I guess the version installed by FastARK will have the latest commits because it's always the same version, only the install method changes.

I know it can. Should I keep the "Force update" setting disabled or may ARK actually downgrade or not inatall the latest version likw happened to me?

krazynez, 2023-09-08:
> > > Does the ARK installed by FastARK always have the latest ARK commits or just the "normal" ARK (the one downloaded from ARK's GitHub repo)?
> > 
> > 
> > What do you mean by "normal"? There's only one version of ARK. FastARK is usually not up to date with new releases, but it can be easily updated using Network Tools in Custom Launcher.
> 
> OK, so I guess the version installed by FastARK will have the latest commits because it's always the same version, only the install method changes.
> 
> I know it can. Should I keep the "Force update" setting disabled or may ARK actually downgrade or not inatall the latest version like happened to me?

FastARK-4 is only updated when I update it.

Just keep Force update off.

Goldtuga, 2023-09-08:
> > > > Does the ARK installed by FastARK always have the latest ARK commits or just the "normal" ARK (the one downloaded from ARK's GitHub repo)?
> > > 
> > > 
> > > What do you mean by "normal"? There's only one version of ARK. FastARK is usually not up to date with new releases, but it can be easily updated using Network Tools in Custom Launcher.
> > 
> > 
> > OK, so I guess the version installed by FastARK will have the latest commits because it's always the same version, only the install method changes.
> > I know it can. Should I keep the "Force update" setting disabled or may ARK actually downgrade or not inatall the latest version like happened to me?
> 
> FastARK-4 is only updated when I update it. It will **only** be updated to stable releases not unstable releases.
> 
> Just keep Force update off.

I was talking about ARK installed by FastARK, not FastARK itself. It's good to know it will only install the version with the latest commits because there's only one ARK version. I know it will.

OK, but will it always update to the latest version, including  the latest Rev version or the version with the latest commits?

closed 2023-09-08 by Goldtuga